coolsnowwolf/lede #10873: WCN6855无法在光影猫中工作

ntbowen, 2023-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
WCN6855无法在光影猫中工作，固件已含驱动，没有无线设置
### (2) 路由器型号和固件版本

A：
型号 | Ariaboard Photonicat
架构 | ARMv8 Processor x 4 (1416MHz, 27.2°C)
固件版本 | OpenWrt R23.2.14 / LuCI Master (git-23.025.27564-bfa4697)
内核版本 | 5.15.91

### (3) 详细日志

A：
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.184501] ath11k_pci 0002:21:00.0: BAR 0: assigned [mem 0x380000000-0x3801fffff 64bit]
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.185307] ath11k_pci 0002:21:00.0: enabling device (0000 -> 0002)
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.186565] ath11k_pci 0002:21:00.0: MSI vectors: 32
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.187018] ath11k_pci 0002:21:00.0: wcn6855 hw2.1
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.367568] mhi mhi0: Requested to power ON
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.368205] mhi mhi0: Power on setup success
Sun Feb  5 16:17:16 2023 kern.info kernel: [   15.856651] mhi mhi0: Wait for device to enter SBL or Mission mode
Sun Feb  5 16:17:16 2023 user.info kernel: [   16.466766] kmodloader: done loading kernel modules from /etc/modules.d/*
Sun Feb  5 16:17:16 2023 kern.info kernel: [   16.479879] ath11k_pci 0002:21:00.0: chip_id 0x12 chip_family 0xb board_id 0xff soc_id 0x400c1211
Sun Feb  5 16:17:16 2023 kern.info kernel: [   16.480699] ath11k_pci 0002:21:00.0: fw_version 0x11090c35 fw_build_timestamp 2022-04-18 20:23 fw_build_id WLAN.HSP.1.1-03125-QCAHSPSWPL_V1_V2_SILICONZ_LITE-3.6510.9
Sun Feb  5 16:17:16 2023 kern.warn kernel: [   18.000774] EXT4-fs (mmcblk1p1): warning: mounting unchecked fs, running e2fsck is recommended
Sun Feb  5 16:17:16 2023 kern.info kernel: [   18.003270] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Opts: . Quota mode: none.
Sun Feb  5 16:17:16 2023 daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Sun Feb  5 16:17:16 2023 daemon.notice procd: /etc/rc.d/S13openssl: Enabling engine devcrypto
Sun Feb  5 16:17:17 2023 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Feb  5 16:17:17 2023 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Feb  5 16:17:18 2023 daemon.notice wpa_supplicant[995]: Successfully initialized wpa_supplicant
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: 8021ad
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: 8021q
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: macvlan
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: veth
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: bridge
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: Network device
Sun Feb  5 16:17:18 2023 user.notice : Added device handler type: tunnel
Sun Feb  5 16:17:19 2023 daemon.notice procd: /etc/rc.d/S45pcat-manager:
Sun Feb  5 16:17:19 2023 daemon.notice procd: /etc/rc.d/S45pcat-manager: ** (pcat-manager:1155): WARNING **: 16:17:19.314: Failed to load keyfile /etc/pcat-manager-userdata.conf: No such file or directory!
Sun Feb  5 16:17:19 2023 daemon.notice procd: /etc/rc.d/S45pcat-manager:
Sun Feb  5 16:17:19 2023 daemon.notice procd: /etc/rc.d/S45pcat-manager: ** (pcat-manager:1155): WARNING **: 16:17:19.314: Failed to load user config data, use default one!
Sun Feb  5 16:17:19 2023 kern.err kernel: [   20.913324] of_dma_request_slave_channel: dma-names property of node '/serial@fe680000' missing or empty
Sun Feb  5 16:17:19 2023 kern.warn kernel: [   20.914180] dw-apb-uart fe680000.serial: failed to request DMA
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.020946] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.026427] rk_gmac-dwmac fe2a0000.ethernet eth1: PHY [stmmac-0:00] driver [YT8521 Gigabit Ethernet] (irq=POLL)
Sun Feb  5 16:17:19 2023 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: ubus call service delete { "name": "cron" } (Not found)
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.036593] dwmac4: Master AXI performs any burst length
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.037129] rk_gmac-dwmac fe2a0000.ethernet eth1: No Safety Features support found
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.037853] rk_gmac-dwmac fe2a0000.ethernet eth1: IEEE 1588-2008 Advanced Timestamp supported
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.038654] rk_gmac-dwmac fe2a0000.ethernet eth1: configuring for phy/sgmii link mode
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.042063] br-lan: port 1(eth1) entered blocking state
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.042537] br-lan: port 1(eth1) entered disabled state
Sun Feb  5 16:17:19 2023 kern.info kernel: [   21.043235] device eth1 entered promiscuous mode
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'lan' is enabled
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'lan' is setting up now
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'lan' is now up
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'loopback' is enabled
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Sun Feb  5 16:17:19 2023 daemon.notice netifd: Interface 'loopback' is now up
[... 71 lines cut ...]
Sun Feb  5 16:17:34 2023 daemon.err nmbd[3737]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Sun Feb  5 16:17:34 2023 daemon.err smbd[3736]: [2023/02/05 16:17:34.333874,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun Feb  5 16:17:34 2023 daemon.err smbd[3736]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sun Feb  5 16:17:41 2023 cron.err crond[4500]: crond (busybox 1.36.0) started, log level 5
Sun Feb  5 16:17:41 2023 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Stop!
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6109] N: ttyd 1.6.3 (libwebsockets 4.3.2-20221001-586-g60178fd03)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6110] N: tty configuration:
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6111] N:   start command: /bin/login
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6111] N:   close signal: SIGHUP (1)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6111] N:   terminal type: xterm-256color
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6155] N:    /usr/lib/libwebsockets-evlib_uv.so
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6156] N: lws_create_context: LWS: 4.3.2-20221001-586-g60178fd03, NET CLI SRV H1 H2 WS ConMon IPV6-off
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6158] N: elops_init_pt_uv:  Using foreign event loop...
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6159] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6160] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6161] N: __lws_lc_tag:  ++ [vh|1|default|br-lan|br-lan|7681] (2)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6166] N: [vh|1|default|br-lan|br-lan|7681]: lws_socket_bind: source ads 172.16.0.1
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6167] N: __lws_lc_tag:  ++ [wsi|1|listen|default|br-lan|7681] (2)
Sun Feb  5 16:17:41 2023 daemon.notice ttyd[4618]: [2023/02/05 16:17:41:6167] N:  Listening on port: 7681
Sun Feb  5 16:17:46 2023 daemon.info procd: - init complete -
Sun Feb  5 16:17:47 2023 daemon.err procd: Could not find firmware /lib/firmware/ath11k/WCN6855/hw2.1/board.bin: No such file or directory
Sun Feb  5 16:17:47 2023 kern.err kernel: [   49.792303] ath11k_pci 0002:21:00.0: failed to fetch board data for bus=pci,vendor=17cb,device=1103,subsystem-vendor=11ad,subsystem-device=a80d,qmi-chip-id=18,qmi-board-id=255 from ath11k/WCN6855/hw2.1/board-2.bin
Sun Feb  5 16:17:47 2023 kern.err kernel: [   49.794027] ath11k_pci 0002:21:00.0: failed to fetch board.bin from WCN6855/hw2.1
Sun Feb  5 16:17:47 2023 kern.warn kernel: [   49.794692] ath11k_pci 0002:21:00.0: qmi failed to fetch board file: -12
Sun Feb  5 16:17:47 2023 kern.warn kernel: [   49.795279] ath11k_pci 0002:21:00.0: failed to load board data file: -12

LxDebug, 2023-02-05:
请问你的固件是写入emmc的吗？

ntbowen, 2023-02-05:
> 请问你的固件是写入emmc的吗？

是的，分别写入emmc和tf，从tf引导

zhouzhiwen2000, 2023-02-05:
请问你的USB有供电吗？

ntbowen, 2023-02-05:
> 请问你的USB有供电吗？

我也不确定，但我怀疑没有供电，在你提的issue#10859找不到wwan设备我这里也是的

zhouzhiwen2000, 2023-02-05:
> > 请问你的USB有供电吗？
> 
> 我也不确定，但我怀疑没有供电，在你提的issue#10859找不到wwan设备我这里也是的

OK

1582130940, 2023-02-05:
Could not find firmware /lib/firmware/ath11k/WCN6855/hw2.1/board.bin: No such file or directory
你是不是漏编译东西了
CONFIG_PACKAGE_ath11k-firmware-wcn6855
这个选了吗

ntbowen, 2023-02-05:
> Could not find firmware /lib/firmware/ath11k/WCN6855/hw2.1/board.bin: No such file or directory 你是不是漏编译东西了 CONFIG_PACKAGE_ath11k-firmware-wcn6855 这个选了吗

当然选了，这个问题是指定文件不存在，路径是存在的，但实际的文件是board-2.bin

1582130940, 2023-02-07:
https://github.com/coolsnowwolf/lede/pull/10696
你把那个改动revert掉试试

ntbowen, 2023-02-10:
> 看了一下，回退那個比較困難了，是否可以嘗試 [1054009064@b3bae07](https://github.com/1054009064/lede/commit/b3bae07a674785f56b5480936e0e5d1009b49f81) 這個補丁？
> 
> 然後只選擇 `ath11k-firmware-wcn6855-kalle` 這個 firmware
> 
> 參考命令
> 
> ```
> wget https://github.com/1054009064/lede/commit/b3bae07a674785f56b5480936e0e5d1009b49f81.patch
> git am b3bae07*.patch
> rm b3bae07*.patch
> ```
> 
> [image]

非常感谢你的建议，设备似乎能正常工作了。多问一句，这个模块理论上支持双频，如何在openwrt上启用？

closed 2023-02-11 by ntbowen