coolsnowwolf/lede #10877: 懂的兄弟帮看下，反复卡这里，就是过不去

guochan2019, 2023-02-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
X86机器，编译反复卡同一处
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
X86机器，最后一次编译6.18成功是1月26日，前两天升级，修改了默认版本6.1，但是卡住，后反复测试，执行过make dirclean不行，又make distclean，但很奇怪，每次都卡在同一个地方，实在不解，求大神帮忙
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
X86 6.1
A：

### (3) 详细日志

A：
gcry_twofish.marker gcry_whirlpool.marker; do \
  b=`basename $pp .marker`; \
  if grep 'VIDEO_LIST_MARKER' $pp >/dev/null 2>&1; then \
    echo $b; \
  fi; \
done) | sort -u > video.lst
cp lib/libgcrypt-grub/cipher/crypto.lst crypto.lst
cat kernel_syms.lst > syminfo.lst.new
for m in disk.module trig.module       cs5536.module        usb.module         usbserial_common.module         usbserial_pl2303.module         usbserial_ftdi.module         usbserial_usbdebug.module         uhci.module        ohci.module        ehci.module             nativedisk.module                               iorw.module      cbtable.module       cbtime.module       cbls.module       cbmemc.module regexp.module     acpi.module          lsacpi.module       lsefisystab.module       lssal.module       lsefimmap.module       lsefi.module       efifwsetup.module      blocklist.module        boot.module           cat.module cmp.module configfile.module       cpuid.module    date.module  echo.module eval.module      extcmd.module        fixvideo.module gptsync.module        halt.module                       reboot.module                hashsum.module pgp.module       hdparm.module   help.module hexdump.module keystatus.module  loadbios.module loadenv.module ls.module lsmmap.module       lspci.module  memrw.module minicmd.module parttool.module password.module password_pbkdf2.module       play.module       spkmodem.module       morse.module probe.module read.module search.module search_fs_file.module search_fs_uuid.module search_label.module       setpci.module        pcidump.module  sleep.module     smbios.module        terminal.module test.module true.module       usbtest.module   videoinfo.module videotest.module xnu_uuid.module dm_nv.module loopback.module cryptodisk.module json.module afsplitter.module luks.module luks2.module geli.module lvm.module ldm.module mdraid09.module mdraid09_be.module mdraid1x.module diskfilter.module raid5rec.module raid6rec.module scsi.module memdisk.module       ata.module         ahci.module        pata.module         usbms.module               font.module              procfs.module affs.module afs.module bfs.module zstd.module btrfs.module archelp.module cbfs.module cpio.module cpio_be.module newc.module odc.module ext2.module fat.module exfat.module f2fs.module fshelp.module hfs.module hfsplus.module hfspluscomp.module iso9660.module jfs.module minix.module minix2.module minix3.module minix_be.module minix2_be.module minix3_be.module nilfs2.module ntfs.module ntfscomp.module reiserfs.module romfs.module sfs.module squash4.module tar.module udf.module ufs1.module ufs1_be.module ufs2.module xfs.module zfs.module zfscrypt.module zfsinfo.module macbless.module  gettext.module gfxmenu.module hello.module gzio.module offsetio.module       bufio.module              elf.module crypto.module pbkdf2.module       relocator.module               datetime.module                setjmp.module       aout.module       bsd.module       linux16.module       multiboot2.module           multiboot.module         linux.module                         xnu.module     random.module macho.module  appleldr.module    chain.module            mmap.module          normal.module part_acorn.module part_amiga.module part_apple.module part_gpt.module part_msdos.module part_sun.module part_plan.module part_dvh.module part_bsd.module part_sunpc.module part_dfly.module msdospart.module       at_keyboard.module       gfxterm.module              gfxterm_background.module       serial.module               terminfo.module             usb_keyboard.module             video_cirrus.module       video_bochs.module functional_test.module exfctest.module strtoull_test.module setjmp_test.module signature_test.module sleep_test.module xnu_uuid_test.module pbkdf2_test.module    legacy_password_test.module           div.module           div_test.module mul_test.module shift_test.module cmp_test.module ctz_test.module bswap_test.module videotest_checksum.module gfxterm_menu.module cmdline_cat_test.module bitmap.module bitmap_scale.module  efi_gop.module       efi_uga.module jpeg.module png.module tga.module        video_fb.module                    video.module              video_colors.module   datehook.module net.module tftp.module http.module      efinet.module         legacycfg.module    syslinuxcfg.module test_blockarg.module xzio.module lzopio.module testload.module       backtrace.module          keylayouts.module priority_queue.module time.module   adler32.module crc64.module mpi.module all_video.module       testspeed.module  tpm.module      tr.module progress.module file.module       rdmsr.module       wrmsr.module gcry_arcfour.module gcry_blowfish.module gcry_camellia.module gcry_cast5.module gcry_crc.module gcry_des.module gcry_dsa.module gcry_idea.module gcry_md4.module gcry_md5.module gcry_rfc2268.module gcry_rijndael.module gcry_rmd160.module gcry_rsa.module gcry_seed.module gcry_serpent.module gcry_sha1.module gcry_sha256.module gcry_sha512.module gcry_tiger.module gcry_twofish.module gcry_whirlpool.module; do \
  sh gensyminfo.sh $m >> syminfo.lst.new || exit 1; \
done
mv syminfo.lst.new syminfo.lst
cat syminfo.lst | sort | gawk -f ./genmoddep.awk > moddep.lst || (rm -f moddep.lst; exit 1)
gcc -o build-grub-module-verifier -I../include -std=gnu99  -DGRUB_FILE=\".util/grub-module-verifier.c\" -I. -I. -I.. -I.. -I../include -I../include -I../grub-core/lib/libgcrypt-grub/src/  -DGRUB_BUILD=1 -DGRUB_UTIL=1 -DGRUB_BUILD_PROGRAM_NAME=\"build-grub-module-verifier\" ../util/grub-module-verifier.c ../util/grub-module-verifier32.c ../util/grub-module-verifier64.c ../grub-core/kern/emu/misc.c ../util/misc.c
TARGET_OBJ2ELF= sh genmod.sh moddep.lst disk.module build-grub-module-verifier disk.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst trig.module build-grub-module-verifier trig.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cs5536.module build-grub-module-verifier cs5536.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst usb.module build-grub-module-verifier usb.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst usbserial_common.module build-grub-module-verifier usbserial_common.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst usbserial_pl2303.module build-grub-module-verifier usbserial_pl2303.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst usbserial_ftdi.module build-grub-module-verifier usbserial_ftdi.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst usbserial_usbdebug.module build-grub-module-verifier usbserial_usbdebug.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst uhci.module build-grub-module-verifier uhci.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst ohci.module build-grub-module-verifier ohci.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst ehci.module build-grub-module-verifier ehci.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst nativedisk.module build-grub-module-verifier nativedisk.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst iorw.module build-grub-module-verifier iorw.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cbtable.module build-grub-module-verifier cbtable.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cbtime.module build-grub-module-verifier cbtime.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cbls.module build-grub-module-verifier cbls.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cbmemc.module build-grub-module-verifier cbmemc.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst regexp.module build-grub-module-verifier regexp.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst acpi.module build-grub-module-verifier acpi.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lsacpi.module build-grub-module-verifier lsacpi.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lsefisystab.module build-grub-module-verifier lsefisystab.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lssal.module build-grub-module-verifier lssal.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lsefimmap.module build-grub-module-verifier lsefimmap.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lsefi.module build-grub-module-verifier lsefi.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst efifwsetup.module build-grub-module-verifier efifwsetup.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst blocklist.module build-grub-module-verifier blocklist.mod
[... cut ...]
echo "grub2-efi" >> /home/hellomax/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/hellomax/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#35.44#14.45#62.48
make[2]: Leaving directory '/home/hellomax/lede'
make[1]: *** [package/Makefile:110: /home/hellomax/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hellomax/lede'
make: *** [/home/hellomax/lede/include/toplevel.mk:231: world] Error 2
hellomax@hellomax:~/lede$

chenzhitong, 2023-02-06:
我也遇到过，我的原因是硬盘空间满了，报错是一样的。

guochan2019, 2023-02-06:
> 我也遇到过，我的原因是硬盘空间满了，报错是一样的。

应该不是硬盘空间，我的虚拟机空间才用了50%，刚才又重新拉了代码，用上次成功的配置文件重新编译，还是卡在这儿，难道是配置文件出问题了？上次明明编译成功了，费解

closed 2023-02-10 by guochan2019