coolsnowwolf/lede #10880: x86云编译失败

hznanami, 2023-02-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

### (1) 具体问题

A：make menuconfig中没有更改任何设置，直接编译会出错

### (2) 路由器型号和固件版本

A：j4125

### (3) 详细日志

make[3]: Entering directory '/home/ubuntu/lede/toolchain/gcc/final'
export SHELL="bash"; make CFLAGS="-O2 -I/home/ubuntu/lede/staging_dir/host/include " CFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CXXFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" GOCFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" gcc_cv_libc_provides_ssp=yes --jobserver-auth=3,4  -C /home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final all
make[4]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libiberty'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libiberty'
make[7]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libiberty/testsuite'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libiberty'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/intl'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/intl'
make[7]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libiberty'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libbacktrace'
make  all-am
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "CXXFLAGS=-g -O2" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "INSTALL=/home/ubuntu/lede/staging_dir/host/bin/install -c" "INSTALL_DATA=/home/ubuntu/lede/staging_dir/host/bin/install -c -m 644" "INSTALL_PROGRAM=/home/ubuntu/lede/staging_dir/host/bin/install -c" "INSTALL_SCRIPT=/home/ubuntu/lede/staging_dir/host/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "LIBCFLAGS_FOR_TARGET=-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "MAKE=make" "MAKEINFO=/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/missing makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl" "infodir=/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/share/info" "libdir=/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib" "prefix=/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl" "tooldir=/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/x86_64-openwrt-linux-musl" "AR=ar" "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libcody'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libcody'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/zlib'
make[7]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libbacktrace'
true  DO=all multi-do # make
make[7]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libbacktrace'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libbacktrace'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libdecnumber'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libdecnumber'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/fixincludes'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/lto-plugin'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/fixincludes'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/fixincludes'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/fixincludes'
make  all-am
make[7]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/lto-plugin'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[7]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/lto-plugin'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/lto-plugin'
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/build-x86_64-pc-linux-gnu/libcpp'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/libcpp'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/gcc'
/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/./gcc/xgcc -B/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/./gcc/ -xc++ -nostdinc /dev/null -S -o /dev/null -fself-test=/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/testsuite/selftests
xgcc: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
make[6]: *** [/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cp/Make-lang.in:196: s-selftest-c++] Error 1
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final/gcc'
make[5]: *** [Makefile:4448: all-gcc] Error 2
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
make[4]: *** [Makefile:976: all] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-final'
[... cut ...]
make[3]: Leaving directory '/home/ubuntu/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#1.15#0.16#1.11
    ERROR: toolchain/gcc/final failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [toolchain/Makefile:93: /home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2023-02-20:
https://github.com/coolsnowwolf/lede/issues/10895
这个问题？

coolsnowwolf, 2026-09-05:
日志来自旧 GCC 11.3 工具链，错误为构建目录内 `xgcc` 找不到 `cc1plus`，属于不完整/损坏的本地工具链构建状态；没有干净环境复现或当前版本错误证据，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf