coolsnowwolf/lede #10882: Frp 内网穿透 - 基本设置- 服务器 只能填ip地址，不能域名了。

82kg, 2023-02-07:
X86软路由 OpenWrt R23.2.14 / LuCI Master (git-23.025.27564-bfa4697) 

Frp 内网穿透 - 基本设置- 服务器  之前版本都是能填域名的。
现在填域名后 
报错[service.go:132] login to server failed: dial tcp: lookup xx.xxx.top: i/o timeout

去frp查了以后 有如下发现 ：

以linux为例，未指定dns_server参数的话，它会去系统的/etc/resolv.conf查找dns配置，利用配置的dns地址去解析域名。如果没这个配置文件，所以需要手动指定一下dns_server

但是 frpc.ini 没法添加 dns_server  =8.8.8.8 参数。会报错。

/etc/resolv.conf 看着也没问题。。。

求大佬帮助

mitsukileung, 2023-02-07:
FRP的版本也没有动过，再说就算是0.46.1也没有这样的问题，都可以用域名

82kg, 2023-02-07:
> FRP的版本也没有动过，再说就算是0.46.1也没有这样的问题，都可以用域名 要不升到0.46.1试试 自己改下文件就可以了 [coolsnowwolf/packages#639](https://github.com/coolsnowwolf/packages/pull/639)

升级到0.46.1 就可以了。 哈哈
@coolsnowwolf  希望大佬空了 合并下这个pr。

closed 2023-02-07 by 82kg

82kg, 2023-02-08:
> FRP的版本也没有动过，再说就算是0.46.1也没有这样的问题，都可以用域名 要不升到0.46.1试试 自己改下文件就可以了 [coolsnowwolf/packages#639](https://github.com/coolsnowwolf/packages/pull/639)

今天发现，编译出来的 frpc 只有 3.7MB 版本也是0.46.1。报错 。。。
我直接用frp编译好的frpc 有12MB  替换 /usr/bin/frpc 就可以了