coolsnowwolf/lede #10887: RM2100wifi无法启动

UR-l, 2023-02-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新最新版的lede，编译完刷进红米ac2100之后wifi无法启动，提示Wireless is disabled or not associated

### (2) 路由器型号和固件版本

A：红米AC2100，R23.2.14

### (3) 详细日志

A：
[    0.000000] Linux version 5.4.231 (lf@lf-virtual-machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r5726-91bc6bfd3)) #0 SMP Tue Feb 7 03:40:27 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Redmi Router AC2100
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
[    0.000000] pcpu-alloc: s26640 r8192 d22512 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0001cc08
[    0.000000] Readback ErrCtl register=0001cc08
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 113288K/131072K available (6159K kernel code, 271K rwdata, 1292K rodata, 8264K init, 235K bss, 17784K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000008] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007869] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.074033] pid_max: default: 32768 minimum: 301
[    0.078815] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086015] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096601] rcu: Hierarchical SRCU implementation.
[    0.102031] smp: Bringing up secondary CPUs ...
[    3.286849] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.286858] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    3.286870] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    3.286970] CPU1 revision is: 0001992f (MIPS 1004Kc)
[... 198 lines cut ...]
[   23.512649] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[   23.519746] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[   23.524779] AP Driver version-5.0.4.0
[   23.528423] RtmpChipOpsHook(223): Not support for HIF_MT yet! MACVersion=0x0
[   23.535445] mt7615_init()-->
[   23.538309] Use the default ePAeLNA bin image!
[   23.542738] Use the default /etc_ro/wlan/MT7615E_EEPROM1.bin bin image!
[   23.549332] <--mt7615_init()
[   23.556211] <-- RTMPAllocTxRxRingMemory, Status=0
[   23.595857] xt_time: kernel timezone is -0000
[   23.658936] PPP generic driver version 2.4.2
[   23.665061] PPP MPPE Compression module registered
[   23.671318] NET: Registered protocol family 24
[   23.689224] kmodloader: done loading kernel modules from /etc/modules.d/*
[   48.777897] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   48.786219] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   48.795183] br-lan: port 1(eth0) entered blocking state
[   48.800466] br-lan: port 1(eth0) entered disabled state
[   48.806489] device eth0 entered promiscuous mode
[   48.815513] br-lan: port 1(eth0) entered blocking state
[   48.820859] br-lan: port 1(eth0) entered forwarding state
[   48.859237] mtk_soc_eth 1e100000.ethernet eth1: configuring for fixed/rgmii link mode
[   48.868031] mtk_soc_eth 1e100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   48.877407] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   49.790938] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

hwliu11, 2023-02-17:
手动可以开启不？可以的话是闭源驱动问题https://github.com/coolsnowwolf/lede/issues/10662#issue-1506218750

coolsnowwolf, 2026-09-05:
报告使用 2023 年 5.4.231 旧固件及厂商 MT7615 驱动，日志显示的是旧无线栈初始化，缺少当前 mt76/设备配置的复现信息；关闭为 not planned。

closed 2026-09-05 by coolsnowwolf