coolsnowwolf/lede #10892: newifi 到

346673164, 2023-02-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败，一个个插件排查。发现是passwall里v2ray造成

### (2) 路由器型号和固件版本

A：newifi d2 最新固件

### (3) 详细日志

A：scripts/kconfig/conf  --syncconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:478:warning: multi-line strings not supported
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      lib/decompress.o
  AR      lib/lib.a
  EXPORTS lib/lib-ksyms.o
  AR      lib/built-in.a
  GEN     usr/initramfs_data.cpio.lzma
v2ray ../init.d/v2ray 777 0 0' line 375
make[6]: *** [usr/Makefile:63: usr/initramfs_data.cpio.lzma] Error 255
make[5]: *** [Makefile:1739: usr] Error 2
make[5]: Leaving directory '/data/workspace/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.231'
make[4]: *** [Makefile:45: /data/workspace/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.231/.image] Error 2
make[4]: Leaving directory '/data/workspace/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/data/workspace/lede/target/linux'
time: target/linux/install#12.28#4.84#16.00
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/data/workspace/lede'
make[1]: *** [target/Makefile:24: /data/workspace/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/data/workspace/lede'
make: *** [/data/workspace/lede/include/toplevel.mk:231: world] Error 2

tangyl2000, 2023-02-11:
我也遇到同样的问题了

Jonnyan404, 2023-02-12:
我也同样问题,补充一下日志.

```bash
Configuring luci-i18n-nlbwmon-zh-cn.
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libatomic1 to hold.
Setting flags for package libstdcpp6 to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
cp -fpR /home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x /home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root.orig-ipq806x
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/upgrade/asrock.sh: line 1: /lib/functions.sh: No such file or directory
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/v2ray: line 2: $'\r': command not found
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/v2ray: line 5: $'\r': command not found
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/v2ray: line 8: $'\r': command not found
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/v2ray: line 9: syntax error near unexpected token `$'{\r''
'home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/v2ray: line 9: `start_service() {
Enabling arpbind
Enabling autoreboot
Enabling avahi-daemon
Enabling boot
/home/ubuntu/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/upgrade/asrock.sh: line 1: /lib/functions.sh: No such file or directory
```

Jonnyan404, 2023-02-12:
我暂时通过把v2ray改成xray,解决了.

KevinMX, 2023-02-18:
> 

同样的问题，在 ipq40xx 也复现了。

对 v2ray init 脚本的处理有问题？

Jonnyan404, 2023-02-18:
> > 
> 
> 同样的问题，在 ipq40xx 也复现了。
> 
> 对 v2ray init 脚本的处理有问题？

用官方的passwall源,不要用三方就行了.