coolsnowwolf/lede #10895: lede源码编译报gcc找不到

jepson66, 2023-02-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：源码编译报gcc找不到
[image]

## 2. 详细叙述
环境：macbook  pro parallels虚拟机 ubuntu20.04
1.报错如上面所述
2.本地已经安装了gcc，g++
[image]
[image]
3.另外请问一下，编译源码是不是不能用最新版本的gcc/g++,报了一个Checking 'working-gcc'... failed.我怀疑是这个问题，所以把gcc/g++降级了，降级后，就报了现在的错误
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

hcym, 2023-02-12:
没有吧，最近两个月风平浪静，没什么事

WYC-2020, 2023-02-13:
这个自己排查一下，要是真有问题早就炸了

zheshifandian, 2023-02-13:
升级你的gcc

jepson66, 2023-02-13:
@zheshifandian请问，升级多少版本的gcc，我升级gcc之后报了其他错误

dfbucy090, 2023-02-16:
同问，小白看不懂就问，大佬都是怎么解决

N1和X86编译都不行了，开始都出现
mkdir -p /root/lede/dl
SHELL= flock /root/lede/tmp/.fortify-headers-1.1.tar.gz.flock -c ' /root/lede/scripts/download.pl "/root/lede/dl" "fortify-headers-1.1.tar.gz" "6ba5d860a2d2ba4c3346924b93930c34856eafe148bdbdf271ecab8065201fb6" "" "https://dl.2f30.org/releases" '
mkdir -p /root/lede/dl
SHELL= flock /root/lede/tmp/.linux-5.15.94.tar.xz.flock -c ' /root/lede/scripts/download.pl "/root/lede/dl" "linux-5.15.94.tar.xz" "da9270dbe64ddf1db13c70470957ff6796eb996d867bb4aed7d14a70e1c65a72" "" "@KERNEL/linux/kernel/v5.x" '
bash: line 1: aarch64-openwrt-linux-musl-gcc: command not found
bash: line 1: aarch64-openwrt-linux-musl-gcc: command not found

curl -f --connect-timeout 20 --retry 5 --location https://mirrors.aliyun.com/gnu/gdb/gdb-12.1.tar.xz
bash: line 1: aarch64-openwrt-linux-musl-gcc: command not found
curl -f --connect-timeout 20 --retry 5 --location https://mirrors.aliyun.com/gnu/binutils/binutils-2.39.tar.xz
mkdir -p /home/fangqin094/lede/dl
SHELL= flock /home/fangqin094/lede/tmp/.linux-5.15.94.tar.xz.flock -c ' /home/fangqin094/lede/scripts/download.pl "/home/fangqin094/lede/dl" "linux-5.15.94.tar.xz" "da9270dbe64ddf1db13c70470957ff6796eb996d867bb4aed7d14a70e1c65a72" "" "@KERNEL/linux/kernel/v5.x" '
bash: line 1: x86_64-openwrt-linux-musl-gcc: command not found

curl -f --connect-timeout 20 --retry 5 --location https://mirrors.aliyun.com/gnu/binutils/binutils-2.39.tar.xz
curl -f --connect-timeout 20 --retry 5 --location https://mirrors.aliyun.com/gnu/gdb/gdb-12.1.tar.xz
bash: line 1: x86_64-openwrt-linux-musl-gcc: command not found
bash: line 1: x86_64-openwrt-linux-musl-gcc: command not found
结果是

touch /home/fangqin094/lede/staging_dir/hostpkg/stamp/.ruby_installed
make[3]: Leaving directory '/home/fangqin094/lede/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#343.05#58.31#775.89
make[2]: Leaving directory '/home/fangqin094/lede'
make[1]: *** [package/Makefile:110: /home/fangqin094/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fangqin094/lede'
make: *** [/home/fangqin094/lede/include/toplevel.mk:231: world] Error 2

closed 2023-02-20 by coolsnowwolf