coolsnowwolf/lede #10897: IPv6中继模式失败

rainchensh, 2023-02-13:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
二级路由使用中继模式，其下设备可以获得IPV6地址（24xx:xxxx:5e45:2600...），但http://www.test-ipv6.com/测试失败显示无IPV6地址
### (1) 具体问题
类似https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/753；和https://github.com/openwrt/openwrt/issues/9881

### (2) 路由器型号和固件版本

A：HiWiFi HC5962 B70；OpenWrt R23.2.14 (2023-02-13) / LuCI Master (git-23.040.60681-9fc2e34)

### (3) 详细日志

A：

Mon Feb 13 20:23:39 2023 daemon.info hostapd: wlan1: STA 60:ab:67:f2:14:08 RADIUS: starting accounting session CD0BCA74A9515318
Mon Feb 13 20:23:39 2023 daemon.info hostapd: wlan1: STA 60:ab:67:f2:14:08 WPA: pairwise key handshake completed (RSN)
Mon Feb 13 20:23:39 2023 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 60:ab:67:f2:14:08
Mon Feb 13 20:23:42 2023 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 60:ab:67:f2:14:08
Mon Feb 13 20:23:43 2023 daemon.info hostapd: wlan0: STA 32:9c:41:d3:bb:54 IEEE 802.11: authenticated
Mon Feb 13 20:23:43 2023 daemon.info hostapd: wlan0: STA 32:9c:41:d3:bb:54 IEEE 802.11: associated (aid 7)
Mon Feb 13 20:23:43 2023 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 32:9c:41:d3:bb:54
Mon Feb 13 20:23:43 2023 daemon.info hostapd: wlan0: STA 32:9c:41:d3:bb:54 RADIUS: starting accounting session F52FDF00109D5EAB
Mon Feb 13 20:23:43 2023 daemon.info hostapd: wlan0: STA 32:9c:41:d3:bb:54 WPA: pairwise key handshake completed (RSN)
Mon Feb 13 20:23:43 2023 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 32:9c:41:d3:bb:54
Mon Feb 13 20:31:14 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 52:be:20:f8:35:d6
Mon Feb 13 20:31:14 2023 daemon.info hostapd: wlan0: STA 52:be:20:f8:35:d6 IEEE 802.11: disassociated
Mon Feb 13 20:31:15 2023 daemon.info hostapd: wlan1: STA 0a:b5:f9:02:db:0f IEEE 802.11: authenticated
Mon Feb 13 20:31:15 2023 daemon.info hostapd: wlan1: STA 0a:b5:f9:02:db:0f IEEE 802.11: associated (aid 5)
Mon Feb 13 20:31:15 2023 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 0a:b5:f9:02:db:0f
Mon Feb 13 20:31:15 2023 daemon.info hostapd: wlan1: STA 0a:b5:f9:02:db:0f RADIUS: starting accounting session 1D97D7999B28A775
Mon Feb 13 20:31:15 2023 daemon.info hostapd: wlan1: STA 0a:b5:f9:02:db:0f WPA: pairwise key handshake completed (RSN)
Mon Feb 13 20:31:15 2023 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 0a:b5:f9:02:db:0f
Mon Feb 13 20:31:15 2023 daemon.info hostapd: wlan0: STA 52:be:20:f8:35:d6 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

lqzhj, 2023-02-15:
同问

rainchensh, 2023-02-19:
电信换了设备就行了。代码应该没问题。

rainchensh, 2023-02-19:
是电信设备的问题

closed 2023-02-19 by rainchensh