coolsnowwolf/lede #10898: > FRP的版本也没有动过，再说就算是0.46.1也没有这样的问题，都可以用域名 要不升到0.46.1试试 自己改下文件就可以了 [coolsnowwolf/packages#639](https://github.com/coolsnowwolf/packages/pull/639)

lee29, 2023-02-14:
> FRP的版本也没有动过，再说就算是0.46.1也没有这样的问题，都可以用域名 要不升到0.46.1试试 自己改下文件就可以了 [coolsnowwolf/packages#639](https://github.com/coolsnowwolf/packages/pull/639)

今天发现，编译出来的 frpc 只有 3.7MB 版本也是0.46.1。报错 。。。
我直接用frp编译好的frpc 有12MB  替换 /usr/bin/frpc 就可以了

_Originally posted by @82kg in https://github.com/coolsnowwolf/lede/issues/10882#issuecomment-1422110514_

lee29, 2023-02-14:
mt7621 hc5962同样启动报错，也一样需要替换才能启动

82kg, 2023-02-15:
@mitsukileung frp 可否升级到0.47再试下

mitsukileung, 2023-02-15:
> mt7621 hc5962同样启动报错，也一样需要替换才能启动

我是编译到固件里面的，之前使用0.46.1也是可以正常使用，现在升到0.47也没有问题。我的X86、K3是云编译的固件，AX6是本地编译都没有遇到你的问题

[image]

[image]

[image]

[image]

我用X86做服务端，AX6、K3做客户端可以正常使用

mitsukileung, 2023-02-15:
> @mitsukileung frp 可否升级到0.47再试下

大雕已经更新了，更新下源码

closed 2023-02-15 by coolsnowwolf