coolsnowwolf/lede #10899: 寻找历史版本的Lean源码

freessrnet3vnet, 2023-02-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：因当前版本编译的固件中samba4,运行时死机，手动安装samba3也有问题，可能是newwifi3-d2的配置低，现找旧版 R22.3.13 源码编译，在本站没找到在哪儿git历史版本，请大佬指点。

### (2) 路由器型号和固件版本

A：Lenovo NewWifi3-D2

### (3) 详细日志

A：  仅找历史版本源代码，不需日志。

weilai-all, 2023-02-15:
当个嘴替：要不先弄清楚啥叫git

freessrnet3vnet, 2023-02-15:
fork行吧，找到一个2022年1月份的bandle，git clone的解开git push上传的，所以就说git了，新手勿怪

lxwq, 2023-02-16:
修改package/lean/autosamba/makefile文件中samba4为samba
修改package/lean/autosamba/files/20-smb文件中samba4为samba
试试看,

deplives, 2023-02-16:
要不您用学学用 git？
https://learngitbranching.js.org/?locale=zh_CN

freessrnet3vnet, 2023-02-16:
git教程特别棒
我在用git啊，一边学一边用，下载了一个coolsnowwolf-lede_-_2022-01-17_09-55-02.bundle，用git clone解开了，然后git push 它的master分支到我新建的仓库上了，我是说还有没有其他版本的lede，如果只有这一版，那就算了。我现在正在把这个版本fork到我的仓库，然后改代码，修改samba36-server的MakeFile，让它编译3.6.25旧版本。

freessrnet3vnet, 2023-02-16:
> 修改package/lean/autosamba/makefile文件中samba4为samba 修改package/lean/autosamba/files/20-smb文件中samba4为samba 试试看,

非常感谢，samba36的4.x版本 和 samba4应该是两回事。在make menuconfig的时候 安装的是luci-app-samba不是 luci-app-samba4。所以，我应该是编译samba36的3.x版本。

zzy19868, 2023-02-16:
试试 git checkout  <commit id>

hwliu11, 2023-02-17:
完整库clone到本地，查看tag/branch/log随意checkout

freessrnet3vnet, 2023-02-20:
> 完整库clone到本地，查看tag/branch/log随意checkout
感谢感谢

freessrnet3vnet, 2023-02-20:
> 

感谢，解决了大问题