coolsnowwolf/lede #10903: M2 mac 原生系统编译问题

ShaneGa0, 2023-02-16:
已经安装java，
java -version
java version "19.0.2" 2023-01-17
Java(TM) SE Runtime Environment (build 19.0.2+7-44)
Java HotSpot(TM) 64-Bit Server VM (build 19.0.2+7-44, mixed mode, sharing)

依然报错
bash-3.2$ sudo apt update -y
Password:
The operation couldn’t be completed. Unable to locate a Java Runtime that supports apt.
Please visit http://www.java.com for information on installing Java.

coolsnowwolf, 2023-02-16:
? 首页不是有 mac 编译说明吗？

ShaneGa0, 2023-02-16:
> ? 首页不是有 mac 编译说明吗？
在运行./scripts/feeds update -a这一句的时候一直跳
[image]
已经重复安装7次了还是跳

ecrasy, 2023-02-16:
当然跳了
这是mac os的开发者工具链
新手还是老老实实建个Ubuntu虚拟机
照着L大的Readme一步一步来

babyfish21, 2023-04-17:
> 已经安装java， java -version java version "19.0.2" 2023-01-17 Java(TM) SE Runtime Environment (build 19.0.2+7-44) Java HotSpot(TM) 64-Bit Server VM (build 19.0.2+7-44, mixed mode, sharing)
>
> 依然报错 [...]

从第3步开始，“3.下载源代码，更新 feeds 并选择配置”

coolsnowwolf, 2026-09-05:
该 issue 的报错发生在 macOS 主机环境，且尝试使用 macOS 不存在的 `apt`；评论已确认应使用 Linux 虚拟机/容器编译。当前没有可泛化的 LEDE 源码修复，关闭。

closed 2026-09-05 by coolsnowwolf