coolsnowwolf/lede #10906: 最新编译固件不分配IP，dns也不解析

zhg-em, 2023-02-17:
x86 内核版本 5.15.94 不分配IP，dns也不解析！不知大家有没有遇到相同的问题？求解决方法。
系统日志如下：

Fri Feb 17 17:35:24 2023 user.info mwan3track[7704]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Feb 17 17:35:24 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:24 2023 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-mwan3.d: No such file or directory
Fri Feb 17 17:35:24 2023 daemon.crit dnsmasq[1]: FAILED to start up
Fri Feb 17 17:35:25 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:26 2023 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Fri Feb 17 17:35:26 2023 daemon.info procd: - init complete -
Fri Feb 17 17:35:26 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:27 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:27 2023 cron.err crond[14462]: crond (busybox 1.36.0) started, log level 8
Fri Feb 17 17:35:28 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Feb 17 17:35:29 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:29 2023 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-mwan3.d: No such file or directory
Fri Feb 17 17:35:29 2023 daemon.crit dnsmasq[1]: FAILED to start up
Fri Feb 17 17:35:29 2023 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Fri Feb 17 17:35:30 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:31 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:32 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:33 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:34 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:35 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:36 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:37 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:38 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Feb 17 17:35:39 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:40 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:41 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:42 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:43 2023 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Fri Feb 17 17:35:43 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:44 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:45 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:46 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:47 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:48 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Fri Feb 17 17:35:49 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:50 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:51 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:52 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:53 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:54 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:55 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:56 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:57 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock
Fri Feb 17 17:35:58 2023 daemon.notice dawn: run_server()=tcpsocket.c@258 Address in use - usock

dzhuqi, 2023-02-18:
你在    启动项 重新启动  dnsmasq 试试

sxx1314, 2023-02-18:
dnsmasq 确实无法启动。手动关闭再重启也一直失效。只能通过修改 resolv临时获取联网能力，但是这玩意就是个临时工。等大佬修复。

yanxin152133, 2023-02-20:
有mwan3？

zhg-em, 2023-02-21:
> 有mwan3？

确实是mwan3的问题，把mwan3去掉就可以了。但负载均衡就没有了呀！

yanxin152133, 2023-02-21:
> > 有mwan3？
> 
> 确实是mwan3的问题，把mwan3去掉就可以了。但负载均衡就没有了呀！

负载均衡你们都用来干啥的，多条宽带？

zhg-em, 2023-02-21:
> > > 有mwan3？
> > 
> > 
> > 确实是mwan3的问题，把mwan3去掉就可以了。但负载均衡就没有了呀！
> 
> 负载均衡你们都用来干啥的，多条宽带？

当然，mwan3就是用来干这个的呀！

yanxin152133, 2023-02-21:
> mwan3

https://github.com/coolsnowwolf/lede/issues/10624
看看这个，要不你回退旧版本试试？

zhg-em, 2023-02-21:
> > mwan3
> 
> #10624 看看这个，要不你回退旧版本试试？

谢谢回复！看来只能退回旧版本了。

haohaoget, 2023-05-10:
这个问题也困扰我好久了，一直以为是我上面插件选错了，希望能尽快解决，不然连最基本的网络连通性都没有办法保证了

haohaoget, 2023-05-10:
> 这个问题也困扰我好久了，一直以为是我上面插件选错了，希望能尽快解决，不然连最基本的网络连通性都没有办法保证了

我编译删除了mwan3还是无法分配ip、解析dns，重启dnsmasq也一样，发下53端口没有占用，nslookup也无法响应
nslookup baidu.com
nslookup: write to '127.0.0.1': Connection refused
nslookup: write to '::1': Connection refused
;; connection timed out; no servers could be reached

netstat -lntup
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:19999           0.0.0.0:*               LISTEN      6404/netdata
tcp        0      0 0.0.0.0:3702            0.0.0.0:*               LISTEN      7822/wsdd2
tcp        0      0 127.0.0.1:8125          0.0.0.0:*               LISTEN      6404/netdata
tcp        0      0 192.168.2.3:7681        0.0.0.0:*               LISTEN      7443/ttyd
tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN      7822/wsdd2
tcp        0      0 192.168.2.3:445         0.0.0.0:*               LISTEN      5344/smbd
tcp        0      0 192.168.2.3:22          0.0.0.0:*               LISTEN      2499/dropbear
tcp        0      0 192.168.2.3:139         0.0.0.0:*               LISTEN      5344/smbd
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      1314/vsftpd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      3305/uhttpd
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      3305/uhttpd
tcp        0      0 ::1:8125                :::*                    LISTEN      6404/netdata
tcp        0      0 :::19999                :::*                    LISTEN      6404/netdata
tcp        0      0 :::3702                 :::*                    LISTEN      7822/wsdd2
tcp        0      0 :::5355                 :::*                    LISTEN      7822/wsdd2
tcp        0      0 :::80                   :::*                    LISTEN      3305/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      3305/uhttpd
udp        0      0 127.0.0.1:8125          0.0.0.0:*                           6404/netdata
udp        0      0 192.168.2.255:137       0.0.0.0:*                           5345/nmbd
udp        0      0 192.168.2.3:137         0.0.0.0:*                           5345/nmbd
udp        0      0 0.0.0.0:137             0.0.0.0:*                           5345/nmbd
udp        0      0 192.168.2.255:138       0.0.0.0:*                           5345/nmbd
udp        0      0 192.168.2.3:138         0.0.0.0:*                           5345/nmbd
udp        0      0 0.0.0.0:138             0.0.0.0:*                           5345/nmbd
udp        0      0 0.0.0.0:161             0.0.0.0:*                           7331/snmpd
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           -
udp        0      0 0.0.0.0:5355            0.0.0.0:*                           7822/wsdd2
udp        0      0 0.0.0.0:3702            0.0.0.0:*                           7822/wsdd2
udp        0      0 ::1:8125                :::*                                6404/netdata
udp        0      0 :::161                  :::*                                7331/snmpd
udp        0      0 :::546                  :::*                                4085/odhcp6c
udp        0      0 :::5353                 :::*                                -
udp        0      0 :::5355                 :::*                                7822/wsdd2
udp        0      0 :::3702                 :::*                                7822/wsdd2

xayane, 2023-10-10:
> 我编译删除了mwan3还是无法分配ip、解析dns，重启dnsmasq也一样，发下53端口没有占用，nslookup也无法响应 nslookup baidu.com nslookup: write to '127.0.0.1': Connection refused nslookup: write to '::1': Connection refused ;; connection timed [...]

这个后面解决了，我今天也突然就这样了。像是dnsmasq无法启动，还不知道怎么解决。

xayane, 2023-10-10:
> 我编译删除了mwan3还是无法分配ip、解析dns，重启dnsmasq也一样，发下53端口没有占用，nslookup也无法响应 nslookup baidu.com nslookup: write to '127.0.0.1': Connection refused nslookup: write to '::1': Connection refused ;; connection timed [...]

这个后面解决没有？我也出现了同样的情况，看上去dnsmasq没有运行。

yanxin152133, 2023-10-10:
> 这个后面解决没有？我也出现了同样的情况，看上去dnsmasq没有运行。

有没有可能是你的广告拦截出问题了

xayane, 2023-10-11:
> 有没有可能是你的广告拦截出问题了

后面我发现是昨天的anti-AD的源好像出现了问题，导致生产的ad.conf出现乱码。
手动替换后，dnsmasq终于可以启动了。

carmineyiu, 2023-10-14:
https://forum.openwrt.org/t/fixed-luci-app-dawn-config/159013/2 
改 /etc/config/dawn  "option tcp_port"

haohaoget, 2023-10-15:
> 这个后面解决没有？我也出现了同样的情况，看上去dnsmasq没有运行。

躺平了，不自己编译arm了，坑太多了

closed 2026-09-05 by coolsnowwolf