coolsnowwolf/lede #10909: mt76的Makefile没有更新到最新，导致编译失败

resnou, 2023-02-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在这个提交中，[1823afb](https://github.com/coolsnowwolf/lede/commit/1823afb3f039b73719b847e9681498d3ea234670)，没有提交配套的最新的mt76的Makefile，导致编译报错，
`lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2022-12-22-5b509e80/dma.c:666:70: error: macro "mtk_wed_device_rx_ring_setup" requires 4 arguments, but only 3 given`
官方的Makefile和这个库的有点不同，我不太懂这些，希望能有人更新一下。
https://github.com/openwrt/openwrt/blob/77d061167cf8a7edb3ea7cbc9926f25c8ac8fe1e/package/kernel/mt76/Makefile#L44

resnou, 2023-02-17:
自己尝试更新PKG的信息后，又出现提示
`Patch failed!  Please fix ./patches/001-Revert-sync-with-upstream.patch!`
解决不了了。。。

resnou, 2023-02-18:
刚才看了一下官方主线，又有新的修复5.15了
`kernel: 5.15: refresh patches`
https://github.com/openwrt/openwrt/commit/c0683ec4586d03c9a30efb3c7497fc30db33fc60

coolsnowwolf, 2023-02-18:
已修复

closed 2023-02-18 by coolsnowwolf