coolsnowwolf/lede #10911: 求助大神们：关于绿联8156usb网卡 插在r4s 只能半双工

JeiWang, 2023-02-20:
各位大神：

8156 2.5g  usb网卡插上后，只能是2500m 半双工，通过ethtool查看，是cdc-ncm在工作，rtl8152驱动并未工作
但删除cdc-ncm驱动后，又不能识别接口。

请问大神们该如何处理

WYC-2020, 2023-02-20:
编译5.10内核试试，我看patch 只有5.10有加入8156 @JeiWang

WYC-2020, 2023-02-20:
l大默认用的 usb-net-rtl8152-vendor这个，这个是外挂的，所以应该和内核无关，用的这个地方的 驱动，https://github.com/wget/realtek-r8152-linux/，但是我看这里用的版本是2.16.3.20220914 ，你试下升级这个驱动 然后在编译 看下 ，我看最新版本已经到了v2.16.3.20221209，如果还不行 试下编译

WYC-2020, 2023-02-20:
如果还是不行 实现编译 usb-net-rtl8152 吧usb-net-rtl8152-vendor 取消

JeiWang, 2023-02-20:
@WYC-2020 多谢，我去试试，我现在应该是用的usb-net-rtl8152-vendor，只能把网卡识别成cdc-ncm，没法正确加载，我现在内核是5.15

WYC-2020, 2023-02-20:
那用用 usb-net-rtl8152 这个驱动试下 ，记得吧usb-net-rtl8152-vendor取消，如果要升级usb-net-rtl8152-vendor这个，看下图
[image]
@JeiWang

JeiWang, 2023-02-20:
@WYC-2020 多谢多谢

JeiWang, 2023-02-21:
昨天试了一下，usb-net-rtl8152(-vendor) 都还是不行，只能2500半双工，启动的还是cdc_ncm驱动，8152驱动应该是安上了，因为log里有  “cdc 注册了一个 r8152 driver”类似的一条。
搜了linux关于这个问题的相关情况，说50-usb-realtek-net.rules能够阻止用cdc_ncm驱动，但是openwrt 18之后就没有 udev了，意思是放在 hotplug.d/usb里，准备试试。

coolsnowwolf, 2026-09-05:
当前仓库同时提供内核 `cdc_ncm` 与 Realtek 专用 `kmod-usb-net-rtl8152`/`r8152` 驱动；RTL8156 被 cdc_ncm 绑定时，半双工表现来自所选 USB 网络协议/设备固件，而不是缺失的 LEDE 补丁。应按设备选择专用 r8152 驱动并避免同时加载 cdc_ncm；该报告没有可复现的当前代码错误，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf