coolsnowwolf/lede #10912: frpc不工作

windybird, 2023-02-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
新三路由编译，frpc不工作
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：2023/02/20 15:55:41: frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！
frpc -v不显示版本，卡着不动，貌似不能启动！

### (2) 路由器型号和固件版本

A：new3-d2

### (3) 详细日志

A：2023-02-20T04:29:30.3624480Z ##[group]Run cd openwrt
2023-02-20T04:29:30.3624905Z [36;1mcd openwrt[0m
2023-02-20T04:29:30.3625196Z [36;1mecho -e "$(nproc) thread compile"[0m
2023-02-20T04:29:30.3625523Z [36;1mmake -j$(nproc) || make -j1 || make -j1 V=s[0m
2023-02-20T04:29:30.3625858Z [36;1mecho "::set-output name=status::success"[0m
2023-02-20T04:29:30.3626245Z [36;1mgrep '^CONFIG_TARGET.*DEVICE.*=y' .config | sed -r 's/.*DEVICE_(.*)=y/\1/' > DEVICE_NAME[0m
2023-02-20T04:29:30.3626691Z [36;1m[ -s DEVICE_NAME ] && echo "DEVICE_NAME=_$(cat DEVICE_NAME)" >> $GITHUB_ENV[0m
2023-02-20T04:29:30.3627178Z [36;1mecho "FILE_DATE=_$(date +"%Y%m%d%H%M")" >> $GITHUB_ENV[0m
2023-02-20T04:29:30.3696027Z shell: /usr/bin/bash -e {0}
2023-02-20T04:29:30.3696297Z env:
2023-02-20T04:29:30.3697627Z   REPO_URL: https://github.com/coolsnowwolf/lede
2023-02-20T04:29:30.3697948Z   REPO_BRANCH: master
2023-02-20T04:29:30.3698320Z   FEEDS_CONF: feeds.conf.default
2023-02-20T04:29:30.3698583Z   CONFIG_FILE: .config
2023-02-20T04:29:30.3698832Z   DIY_P1_SH: diy-part1.sh
2023-02-20T04:29:30.3699115Z   DIY_P2_SH: diy-part2.sh
2023-02-20T04:29:30.3699358Z   UPLOAD_BIN_DIR: false
2023-02-20T04:29:30.3699584Z   UPLOAD_FIRMWARE: true
2023-02-20T04:29:30.3699839Z   UPLOAD_COWTRANSFER: false
2023-02-20T04:29:30.3700097Z   UPLOAD_WETRANSFER: false
2023-02-20T04:29:30.3700346Z   UPLOAD_RELEASE: true
2023-02-20T04:29:30.3700559Z   TZ: Asia/Shanghai
2023-02-20T04:29:30.3700783Z ##[endgroup]
2023-02-20T04:29:30.3806923Z 2 thread compile
2023-02-20T04:29:38.0263357Z make[2]: Entering directory '/workdir/openwrt/scripts/config'
2023-02-20T04:29:38.0264010Z make[2]: 'conf' is up to date.
2023-02-20T04:29:38.0265165Z make[2]: Leaving directory '/workdir/openwrt/scripts/config'
2023-02-20T04:29:39.9224211Z  make[1] world
2023-02-20T04:29:42.3892736Z  make[2] tools/compile
2023-02-20T04:29:42.4406141Z  make[2] package/cleanup
2023-02-20T04:29:45.0078529Z  make[3] -C tools/flock compile
2023-02-20T04:29:45.0802155Z  make[3] -C tools/xz compile
2023-02-20T04:30:04.1764383Z  make[3] -C tools/sed compile
2023-02-20T04:30:35.6114940Z  make[3] -C tools/libdeflate compile
2023-02-20T04:30:35.6119775Z  make[3] -C tools/patch compile
2023-02-20T04:30:39.0287921Z  make[3] -C tools/tar compile
2023-02-20T04:31:21.4421039Z  make[3] -C tools/m4 compile
2023-02-20T04:31:21.4432083Z  make[3] -C tools/autoconf-archive compile
2023-02-20T04:31:23.2501642Z  make[3] -C tools/ninja compile
2023-02-20T04:32:10.9465619Z  make[3] -C tools/expat compile
2023-02-20T04:32:20.6317139Z  make[3] -C tools/zlib compile
2023-02-20T04:32:22.8951461Z  make[3] -C tools/cpio compile
2023-02-20T04:32:24.4083397Z  make[3] -C tools/lzma compile
2023-02-20T04:32:31.1501962Z  make[3] -C tools/make-ext4fs compile
2023-02-20T04:32:33.4729697Z  make[3] -C tools/mtools compile
2023-02-20T04:32:48.9333158Z  make[3] -C tools/patch-image compile
2023-02-20T04:32:49.1607967Z  make[3] -C tools/squashfskit4 compile
2023-02-20T04:32:55.5024701Z  make[3] -C tools/sstrip compile
2023-02-20T04:32:57.3368200Z  make[3] -C tools/zip compile
2023-02-20T04:33:01.0117493Z  make[3] -C tools/kernel2minor compile
[... 199 lines cut ...]
2023-02-20T06:04:39.5224367Z  make[3] -C package/lean/ddns-scripts_dnspod compile
2023-02-20T06:04:39.9905576Z  make[3] -C package/network/ipv6/6in4 compile
2023-02-20T06:04:40.4467284Z  make[3] -C package/system/opkg compile
2023-02-20T06:04:48.7267467Z  make[3] -C feeds/luci/modules/luci-compat compile
2023-02-20T06:04:49.3131956Z  make[3] -C feeds/luci/modules/luci-mod-admin-full compile
2023-02-20T06:04:51.1728838Z  make[3] -C feeds/luci/applications/luci-app-adbyby-plus compile
2023-02-20T06:04:52.1133734Z  make[3] -C feeds/luci/applications/luci-app-ddns compile
2023-02-20T06:04:53.0652328Z  make[3] -C feeds/luci/applications/luci-app-mwan3 compile
2023-02-20T06:04:54.3784042Z  make[3] -C package/lean/ipv6-helper compile
2023-02-20T06:04:54.8314121Z  make[3] -C package/OpenAppFilter/luci-app-oaf compile
2023-02-20T06:04:55.7866408Z  make[3] -C feeds/luci/collections/luci compile
2023-02-20T06:04:56.3985985Z  make[3] -C feeds/luci/applications/luci-app-syncdial compile
2023-02-20T06:04:56.9310803Z  make[3] -C package/lean/default-settings compile
2023-02-20T06:04:57.1827512Z  make[3] -C feeds/luci/applications/luci-app-autoreboot compile
2023-02-20T06:04:57.4318827Z  make[3] -C feeds/packages/utils/ttyd compile
2023-02-20T06:04:59.1741616Z  make[3] -C feeds/luci/applications/luci-app-ttyd compile
2023-02-20T06:04:59.8447120Z  make[2] package/install
2023-02-20T06:05:18.4024184Z  make[2] target/install
2023-02-20T06:05:21.1719984Z  make[3] -C target/linux install
2023-02-20T06:07:51.8517259Z  make[6] -C target/linux/ramips/image/lzma-loader compile loader.bin
2023-02-20T06:07:59.4331679Z  make[6] -C target/linux/ramips/image/lzma-loader compile loader.bin
2023-02-20T06:08:01.1168470Z  make[2] package/index
2023-02-20T06:08:07.2898142Z  make[2] json_overview_image_info
2023-02-20T06:08:10.1779050Z  make[2] checksum
2023-02-20T06:08:14.5957746Z ##[warning]The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

yanxin152133, 2023-02-20:
你这是什么日志

WYC-2020, 2023-02-21:
前不久 不是在吼到升级么，不应该不能启动吧

windybird, 2023-02-21:
> [image] 发这个看看

日志为空，frpc -v不显示版本号

windybird, 2023-02-21:
感觉不能正常工作。。。编译这个时要注意啥，不能和哪个模块同时编译，一直都是云编译。。。也没动。。。。到了0.45.0以后的就出问题。。。

yanxin152133, 2023-02-21:
> 感觉不能正常工作。。。编译这个时要注意啥，不能和哪个模块同时编译，一直都是云编译。。。也没动。。。。到了0.45.0以后的就出问题。。。

https://github.com/yanxin152133/lede-x86_x64
试试我这个，我最近刚换了一个版本的，frpc可以使用

windybird, 2023-02-23:
> > 感觉不能正常工作。。。编译这个时要注意啥，不能和哪个模块同时编译，一直都是云编译。。。也没动。。。。到了0.45.0以后的就出问题。。。
> 
> https://github.com/yanxin152133/lede-x86_x64 试试我这个，我最近刚换了一个版本的，frpc可以

我是新三的(MIPS),能编个这个试下？反正我云编完后frpc不能用

yanxin152133, 2023-02-23:
> > > 感觉不能正常工作。。。编译这个时要注意啥，不能和哪个模块同时编译，一直都是云编译。。。也没动。。。。到了0.45.0以后的就出问题。。。
> > 
> > 
> > https://github.com/yanxin152133/lede-x86_x64 试试我这个，我最近刚换了一个版本的，frpc可以
> 
> 我是新三的(MIPS),能编个这个试下？反正我云编完后frpc不能用

有的好像也跟勾选的选项有关
[image]

我改成别的也无法连接server端

sidpixel, 2023-02-24:
+1 出现同样报错，frpc无法使用
另外就是cpu资源使用率很高，关掉frpc服务后cpu使用率降低

yanxin152133, 2023-02-24:
是R23.2.14版本吗？我的好像cpu利用率也很高

yanxin152133, 2023-02-24:
这几天我在编译R22.5.5这个版本，一直提示upx有问题。

> 感觉不能正常工作。。。编译这个时要注意啥，不能和哪个模块同时编译，一直都是云编译。。。也没动。。。。到了0.45.0以后的就出问题。。。

mitsukileung, 2023-02-24:
试了下，MTK7621的确启动不了，但X86和ip807x好像没有这个问题

82kg, 2023-02-27:
小米4，MTK7621 同样不行。

windybird, 2023-02-28:
是的。。。frpc应该是官方问题

 
 
 &gt; +1 出现同样报错，frpc无法使用 
 &gt; 另外就是cpu资源使用率很高，关掉frpc服务后cpu使用率降低 
 &gt; 
 &gt; — 
 &gt; <https://github.com/coolsnowwolf/lede/issues/10912#issuecomment-1442949233&gt;, 
 &gt; . 
 &gt; ***@***.***&gt; 
 &gt;

mitsukileung, 2023-03-04:
刚刚试了把 frp里的
PKG_BUILD_DEPENDS:=golang/host upx/host 改成 PKG_BUILD_DEPENDS:=golang/host
把 $(STAGING_DIR_HOST)/bin/upx --lzma --best $$(1)/usr/bin/$(1) || true 删掉
在 7620 上编译出来的 frpc 可以正常工作，但编译出来的文件比之前的大了1M左右。

[image]

[image]

WYC-2020, 2023-03-04:
那就是upx压缩后不匹配呗，不压缩就行了撒

mitsukileung, 2023-03-04:
> 那就是upx压缩后不匹配呗，不压缩就行了撒

不过好像就是 7620 7621 会这样，ipq807x X86好像不会
大佬这个能不能帮忙匹配，因为对于 flash 小的固件不太友好

yanxin152133, 2023-03-05:
> > 那就是upx压缩后不匹配呗，不压缩就行了撒
> 
> 不过好像就是 7620 7621 会这样，ipq807x X86好像不会 大佬这个能不能帮忙匹配，因为对于 flash 小的固件不太友好

自己编译的时候用旧版本呗

WYC-2020, 2023-03-08:
> > 那就是upx压缩后不匹配呗，不压缩就行了撒
> 
> 不过好像就是 7620 7621 会这样，ipq807x X86好像不会 大佬这个能不能帮忙匹配，因为对于 flash 小的固件不太友好

这玩意我记得以前有一个什么库 也是upx压缩后在你们这些机器上运行不了，是什么库来的忘记了，本身这种压缩壳匹配不好很正常，不是所有平台都能兼容，什么压缩壳 加密壳 都一样因为他们要写shellcode 加载到内存然后释放，自然这个shellcode适配就是关键了，你可以试试升级upx到最新 然后试下

mitsukileung, 2023-03-16:
> > > 那就是upx压缩后不匹配呗，不压缩就行了撒
> > 
> > 
> > 不过好像就是 7620 7621 会这样，ipq807x X86好像不会 大佬这个能不能帮忙匹配，因为对于 flash 小的固件不太友好
> 
> 这玩意我记得以前有一个什么库 也是upx压缩后在你们这些机器上运行不了，是什么库来的忘记了，本身这种压缩壳匹配不好很正常，不是所有平台都能兼容，什么压缩壳 加密壳 都一样因为他们要写shellcode 加载到内存然后释放，自然这个shellcode适配就是关键了，你可以试试升级upx到最新 然后试下

直接改这三项编译出错，如果直接替换成 immortalwrt 的 Makefile 可以编译成功，但编译出来的文件不会压缩。
PKG_VERSION:=3.96
PKG_SOURCE_VERSION:=d7ba31cab8ce8d95d2c10e88d2ec787ac52005ef
PKG_MIRROR_HASH:=15b4884dbe35d3b7bda27538429cab92aaeaca006d558deb49f2ef93e3c261b8

WYC-2020, 2023-03-17:
> > > > 那就是upx压缩后不匹配呗，不压缩就行了撒
> > > 
> > > 
> > > 不过好像就是 7620 7621 会这样，ipq807x X86好像不会 大佬这个能不能帮忙匹配，因为对于 flash 小的固件不太友好
> > 
> > 
> > 这玩意我记得以前有一个什么库 也是upx压缩后在你们这些机器上运行不了，是什么库来的忘记了，本身这种压缩壳匹配不好很正常，不是所有平台都能兼容，什么压缩壳 加密壳 都一样因为他们要写shellcode 加载到内存然后释放，自然这个shellcode适配就是关键了，你可以试试升级upx到最新 然后试下
> 
> 直接改这三项成最新版本编译出错，如果直接替换成 immortalwrt 的 Makefile 可以编译成功，但编译出来的文件不会压缩。 PKG_VERSION:=3.96 PKG_SOURCE_VERSION:=d7ba31cab8ce8d95d2c10e88d2ec787ac52005ef PKG_MIRROR_HASH:=15b4884dbe35d3b7bda27538429cab92aaeaca006d558deb49f2ef93e3c261b8

那不清楚 我自己升级到了4.0.1，编译没出错，不过我都是x86 测试的，其他平台没测，基本不用

mitsukileung, 2023-03-22:
> 

升至 4.0.2 编译失败

time: package/feeds/luci/luci-app-filetransfer/compile#0.11#0.09#0.19
make[3]: Entering directory '/home/koysuke/lede/package/lean/ucl'
make[3]: Leaving directory '/home/koysuke/lede/package/lean/ucl'
time: package/lean/ucl/host-compile#0.16#0.09#0.22
make[3]: Entering directory '/home/koysuke/lede/package/lean/upx'
cp -fpR /home/koysuke/lede/build_dir/hostpkg/upx-4.0.2/src/upx.out /home/koysuke/lede/staging_dir/host/bin/upx
cp: cannot stat '/home/koysuke/lede/build_dir/hostpkg/upx-4.0.2/src/upx.out': No such file or directory
make[3]: *** [Makefile:61: /home/koysuke/lede/staging_dir/hostpkg/stamp/.upx_installed] Error 1
make[3]: Leaving directory '/home/koysuke/lede/package/lean/upx'
time: package/lean/upx/host-compile#0.15#0.09#0.21
    ERROR: package/lean/upx [host] failed to build.
make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [package/Makefile:110: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koysuke/lede'
make: *** [/home/koysuke/lede/include/toplevel.mk:231：world] 错误 2

langliang3241, 2023-04-21:
用最新的源码【 OpenWrt R23.4.1 / LuCI Master (git-23.109.50273-11b9572) 】编译出的固件 7621的京东云和歌华链 frpc 都报错：‘“frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！”，frpc 版本是：0.47.0-1 服务器端 tcp_mux 设置是true ，frps 版本也是0.47 ，以前的老版本都可以正常启动，有解决办法么

yanxin152133, 2023-04-21:
> 用最新的源码【 OpenWrt R23.4.1 / LuCI Master (git-23.109.50273-11b9572) 】编译出的固件 7621的京东云和歌华链 frpc 都报错：‘“frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！”，frpc 版本是：0.47.0-1 服务器端 tcp_mux 设置是true ，frps 版本也是0.47 ，以前的老版本都可以正常启动，有解决办法么

最好的办法就是用旧版本的frp

langliang3241, 2023-04-22:
> > 用最新的源码【 OpenWrt R23.4.1 / LuCI Master (git-23.109.50273-11b9572) 】编译出的固件 7621的京东云和歌华链 frpc 都报错：‘“frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！”，frpc 版本是：0.47.0-1 服务器端 tcp_mux 设置是true ，frps 版本也是0.47 ，以前的老版本都可以正常启动，有解决办法么
> 
> 最好的办法就是用旧版本的frp

怎么在编译时选择旧版本frpc ？我电脑配置太差了，都是云编译的

yanxin152133, 2023-04-22:
> > > 用最新的源码【 OpenWrt R23.4.1 / LuCI Master (git-23.109.50273-11b9572) 】编译出的固件 7621的京东云和歌华链 frpc 都报错：‘“frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！”，frpc 版本是：0.47.0-1 服务器端 tcp_mux 设置是true ，frps 版本也是0.47 ，以前的老版本都可以正常启动，有解决办法么
> > 
> > 
> > 最好的办法就是用旧版本的frp
> 
> 怎么在编译时选择旧版本frpc ？我电脑配置太差了，都是云编译的

https://github.com/kuoruan/openwrt-frp

ikaroscn, 2023-08-12:
> > 用最新的源码【 OpenWrt R23.4.1 / LuCI Master (git-23.109.50273-11b9572) 】编译出的固件 7621的京东云和歌华链 frpc 都报错：‘“frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！”，frpc 版本是：0.47.0-1 服务器端 tcp_mux 设置是true ，frps 版本也是0.47 ，以前的老版本都可以正常启动，有解决办法么
> 
> 最好的办法就是用旧版本的frp

我试过3个旧版本，都是一样的问题，只能把frpc移除了，然后重新安装以前我编译的版本才能正常使用

closed 2026-09-05 by coolsnowwolf