coolsnowwolf/lede #10917: 6.1版本不支持 iwlwifi

caixin1218, 2023-02-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:250: /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.o] Error 1
make[12]: *** Waiting for unfinished jobs....

WYC-2020, 2023-02-22:
大师我算不出来

caixin1218, 2023-02-22:
如何不选择iwlwifi 就成功编译，但intel 7260网卡不能识别

caixin1218, 2023-02-22:
kmod-iwlwifi  5.4编译是没有问题的。

WYC-2020, 2023-02-22:
没人修，那我来修吧，像下面一样修：
[image]
[image]
[image]
[854-export-efivarfs-function.patch](https://github.com/coolsnowwolf/lede/files/10805295/854-export-efivarfs-function.patch)
[0001-fix-build-with-kernel-6.1.patch](https://github.com/coolsnowwolf/lede/files/10805296/0001-fix-build-with-kernel-6.1.patch)
别和我说编译不过什么，亲测

caixin1218, 2023-02-22:
lean 大佬为什么没有修复这个问题

WYC-2020, 2023-02-22:
开源免费项目哪来那么多为什么，别人又不是靠这个吃饭，为啥什么问题都修，那的看时间

caixin1218, 2023-02-23:
fi
Looking in links: /home/caixin/lede5.4.214/tmp/tmpz_comdyy
Processing /home/caixin/lede5.4.214/tmp/tmpz_comdyy/setuptools-63.2.0-py3-none-any.whl
Processing /home/caixin/lede5.4.214/tmp/tmpz_comdyy/pip-22.2.2-py3-none-any.whl
Installing collected packages: setuptools, pip
Successfully installed pip-22.2.2 setuptools-63.2.0
make[4]: Leaving directory '/home/caixin/lede5.4.214/build_dir/hostpkg/Python-3.10.7'
touch /home/caixin/lede5.4.214/staging_dir/hostpkg/lib/python3.10/site-packages/.setuptools_installed_63.2.0-1

Applying ./patches-pip/001-pep517-pyc-fix.patch using plaintext: 
patching file pip/_vendor/pep517/in_process/__init__.py
touch /home/caixin/lede5.4.214/staging_dir/hostpkg/lib/python3.10/site-packages/.pip_installed_22.2.2-1
mkdir -p /home/caixin/lede5.4.214/staging_dir/hostpkg/stamp
touch /home/caixin/lede5.4.214/build_dir/hostpkg/Python-3.10.7/.built
touch /home/caixin/lede5.4.214/staging_dir/hostpkg/stamp/.python3_installed
make[3]: Leaving directory '/home/caixin/lede5.4.214/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#273.77#22.56#155.43
make[2]: Leaving directory '/home/caixin/lede5.4.214'
make[1]: *** [package/Makefile:110: /home/caixin/lede5.4.214/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caixin/lede5.4.214'
make: *** [/home/caixin/lede5.4.214/include/toplevel.mk:230: world] Error 2
caixin@debian:~/lede5.4.214$ git pull
./scripts/feeds update -a   
./scripts/feeds install -a
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
Already up to date.
Updating feed 'routing' from 'https://github.com/coolsnowwolf/routing' ...
Already up to date.
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
WARNING: Makefile 'package/feeds/packages/midisport-firmware/Makefile' has a dependency on 'fxload', which does not exist
Installing all packages from feed packages.
WARNING: Not overriding core package 'n2n'; use -f to force
Installing all packages from feed luci.
WARNING: Not overriding core package 'luci-app-qbittorrent-simple'; use -f to force
Installing all packages from feed routing.
Installing all packages from feed telephony.
caixin@debian:~/lede5.4.214$ 

这几天5.4也有点问题，前几天还是好的。

WYC-2020, 2023-02-23:
自己排除 我5.4 测过没问题

arimitx, 2023-02-23:
iwlwifi先用5.15吧，要用6.1就去把官方的mac80211搬过来。

closed 2023-03-02 by coolsnowwolf