coolsnowwolf/lede #10918: target/linux failed to build.

DD-YPH, 2023-02-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译提示make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1

### (2) 路由器型号和固件版本

A：BPI-R3

### (3) 详细日志

A：
make[3] -C target/linux install
    WARNING: Image file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz is too big: 76414976 > 46137344
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1

WYC-2020, 2023-02-22:
是完全看不懂英语？这么明显的错误
[image]

closed 2023-02-22 by aiamadeus

dd-bpir3, 2023-02-25:
> 是完全看不懂英语？这么明显的错误 [image]

看错误提示，不是看警告