coolsnowwolf/lede #10924: 编译失败 UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 14: invalid continuation byte

Ljzd-PRO, 2023-02-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

#10504
#5110
没有找到解决方案。没有添加passwall插件，路径换成全英文的也不行。

## 2. 详细叙述

### (1) 具体问题

A：
编译失败，报错提示：
```
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 14: invalid continuation byte

../../../build_dir/target-mipsel_24kc_musl/util-linux-2.38.1/meson.build:214:0: ERROR: Unhandled python exception
```
系统是 Ubuntu 22.04.2 LTS

### (2) 路由器型号和固件版本

A：K2P 默认分支

### (3) 详细日志
```
WARNING: You should add the boolean check kwarg to the run_command call.
         It currently defaults to false,
         but it will default to true in future releases of meson.
         See also: https://github.com/mesonbuild/meson/issues/9300
Checking if "wchar_t support" compiles: YES 
Library m found: YES
Dependency tinfo skipped: feature tinfo disabled
Found pkg-config: /home/ljzd/文档/lede/staging_dir/host/bin/pkg-config (1.9.3)
Traceback (most recent call last):
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/mesonmain.py", line 148, in run
    return options.run_func(options)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 294, in run
    app.generate()
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 185, in generate
    self._generate(env)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 229, in _generate
    intr.run()
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 2554, in run
    super().run()
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 148, in run
    self.evaluate_codeblock(self.ast, start=1)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 173, in evaluate_codeblock
    raise e
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 166, in evaluate_codeblock
    self.evaluate_statement(cur)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 181, in evaluate_statement
    self.assignment(cur)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 566, in assignment
    value = self.evaluate_statement(node.value)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 179, in evaluate_statement
    return self.function_call(cur)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 454, in function_call
    res = func(node, func_args, kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  [Previous line repeated 5 more times]
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 96, in wrapped
    ret = f(*wrapped_args, **wrapped_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 115, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 276, in wrapper
    return f(*nargs, **wrapped_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 1577, in func_dependency
    d = df.lookup(kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 347, in lookup
    dep = func(kwargs, func_args, func_kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 90, in _do_dependency
    dep = dependencies.find_external_dependency(name, self.environment, kwargs)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/detect.py", line 114, in find_external_dependency
    d = c()
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 96, in __init__
    self._set_libs()
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 372, in _set_libs
    ret, out, err = self._call_pkgbin(libcmd, env=env)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 157, in _call_pkgbin
    cache[(self.pkgbin, targs, fenv)] = self._call_pkgbin_real(args, env)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 115, in _call_pkgbin_real
    p, out, err = Popen_safe(cmd, env=env)
  File "/home/ljzd/文档/lede/staging_dir/host/bin/meson.py/mesonbuild/mesonlib/universal.py", line 1397, in Popen_safe
    o, e = p.communicate(write)
  File "/usr/lib/python3.10/subprocess.py", line 1152, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/usr/lib/python3.10/subprocess.py", line 2041, in _communicate
    stdout = self._translate_newlines(stdout,
  File "/usr/lib/python3.10/subprocess.py", line 1029, in _translate_newlines
    data = data.decode(encoding, errors)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 14: invalid continuation byte

../../../build_dir/target-mipsel_24kc_musl/util-linux-2.38.1/meson.build:214:0: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
make[3]: *** [Makefile:878: /home/ljzd/文档/lede/build_dir/target-mipsel_24kc_musl/util-linux-2.38.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/home/ljzd/文档/lede/package/utils/util-linux'
time: package/utils/util-linux/compile#0.83#0.13#0.93
    ERROR: package/utils/util-linux failed to build.
make[2]: *** [package/Makefile:116: package/utils/util-linux/compile] Error 1
make[2]: Leaving directory '/home/ljzd/文档/lede'
make[1]: *** [package/Makefile:110: /home/ljzd/文档/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ljzd/文档/lede'
make: *** [/home/ljzd/文档/lede/include/toplevel.mk:231：world] 错误 2
```

WYC-2020, 2023-02-23:
搞什么中文路径

closed 2023-02-23 by aiamadeus

rongguoquan, 2024-02-20:
我的报错和你一样，是因为在中文目录下编译。去文件管理器把【文档】文件夹改成document，终端输入make -j1 V=s继续编译就好了