coolsnowwolf/lede #10926: x86-32位编译，Hysteria打不开网页

dfbucy090, 2023-02-23:
有几个月了，每次编译32位固件，Hysteria都不能科学上网，原来编译的可以，例如5.15.50版本。现在不论passwall,还是其他的app插件里的Hysteria，都不能正常工作，不知道哪里出了问题，更新到最新版本也不起作用。N1和x86-64位都正常，本人小白，本指望过段时间能正常，可是几个月了还是没有改变。希望得到大佬的帮助指点，谢谢。

WYC-2020, 2023-02-26:
这年头还用32位哈哈，这个很少编译32位了，没留意是否有这个问题

dfbucy090, 2023-02-26:
> 这年头还用32位哈哈，这个很少编译32位了，没留意是否有这个问题

谢谢大佬能够关注，我知道会引起笑话，我这纯属于抚摸党和强迫症，况且也有32位机器，每次编译都顺带编一次。过去曾得到大佬您的指点，也非常佩服您的学识，希望您能测试下。

WYC-2020, 2023-02-27:
> > 这年头还用32位哈哈，这个很少编译32位了，没留意是否有这个问题
> 
> 谢谢大佬能够关注，我知道会引起笑话，我这纯属于抚摸党和强迫症，况且也有32位机器，每次编译都顺带编一次。过去曾得到大佬您的指点，也非常佩服您的学识，希望您能测试下。

有时间再说，现在在搞ventoy定制

dfbucy090, 2023-02-27:
> > > 这年头还用32位哈哈，这个很少编译32位了，没留意是否有这个问题
> > 
> > 
> > 谢谢大佬能够关注，我知道会引起笑话，我这纯属于抚摸党和强迫症，况且也有32位机器，每次编译都顺带编一次。过去曾得到大佬您的指点，也非常佩服您的学识，希望您能测试下。
> 
> 有时间再说，现在在搞ventoy定制

谢谢，期盼您能解决。

WYC-2020, 2023-02-28:
> > > > 这年头还用32位哈哈，这个很少编译32位了，没留意是否有这个问题
> > > 
> > > 
> > > 谢谢大佬能够关注，我知道会引起笑话，我这纯属于抚摸党和强迫症，况且也有32位机器，每次编译都顺带编一次。过去曾得到大佬您的指点，也非常佩服您的学识，希望您能测试下。
> > 
> > 
> > 有时间再说，现在在搞ventoy定制
> 
> 谢谢，期盼您能解决。

我只能测试x86版本，看你描述 好像说的是 x86 正常，那无能为力了，没设备调试

dfbucy090, 2023-03-03:
看来是没大佬能解决了。

WYC-2020, 2023-03-06:
你这个是能运行，不会崩溃，看下是否配置问题吧

dfbucy090, 2023-03-06:
> 你这个是能运行，不会崩溃，看下是否配置问题吧

配置没问题，我都认真比对过。自己的节点直接导入，64位，N1编译的版本也是相同操作，而且再用过去编译的32位5.15.50旧版本，同样相同配置，均能科学上网。去年7月中旬以前编译的都正常。我怀疑现在32位编译哪里依赖或环境有问题，变化了什么。我自己看不懂，所以麻烦大佬们帮助，感谢您能回复。

dfbucy090, 2023-03-08:
今天回滚到2022.05.05版本编译，出来是5.15.38，结果还是不行。用我过去的5.15.50版刷没问题，看来是32位Hysteria插件问题。

dfbucy090, 2023-03-15:
昨天更新的Hysteria 1.3.4，修复了我报的BUG。马上试验，已经正常了。

closed 2023-03-15 by dfbucy090