coolsnowwolf/lede #10927: 更新源码后编译naiveproxy无法通过

bi4tms, 2023-02-23:
更新源码后无法编译通过，make clean后出现同样错误，具体日志如下：
ninja: Entering directory `/home/x86.5.4/lede/build_dir/target-x86_64_musl/naiveproxy-110.0.5481.100-1/src/out/Release'
[4/3495] CXX obj/base/third_party/double_conversion/double_conversion/bignum-dtoa.o
FAILED: obj/base/third_party/double_conversion/double_conversion/bignum-dtoa.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/third_party/double_conversion/double_conversion/bignum-dtoa.o.d -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -D__MUSL__ -DCR_CLANG_REVISION="llvmorg-16-init-12251-g87d0ff91-2" -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -DCR_LIBCXX_REVISION=2948540a20cbb5b2192119f791b04dd62ca7af1c -D_LIBCPP_ENABLE_ASSERTIONS_DEFAULT=1 -D_LIBCPP_AVAILABILITY_CUSTOM_VERBOSE_ABORT_PROVIDED=1 -DCR_SYSROOT_KEY=20221105T211506Z-0 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -I../../buildtools/third_party/libc++ -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers --target=x86_64-openwrt-linux-musl -m64 -msse3 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -gdwarf-4 -g1 -gdwarf-aranges -fprofile-use=../../chrome/build/pgo_profiles/chrome-linux-5481-1675162426-393d781cbc3753d814a299bfd43435032b5acfd9.profdata -Wno-profile-instr-unprofiled -Wno-profile-instr-out-of-date -Wno-backend-plugin -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang raw-ref-template-as-trivial-member -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wno-unused-const-variable -Wno-unused-function -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-11.3.0_musl -fvisibility-inlines-hidden -c ../../base/third_party/double_conversion/double-conversion/bignum-dtoa.cc -o obj/base/third_party/double_conversion/double_conversion/bignum-dtoa.o
/bin/sh: 1: ../../third_party/llvm-build/Release+Asserts/bin/clang++: Exec format error
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:131: /home/x86.5.4/lede/build_dir/target-x86_64_musl/naiveproxy-110.0.5481.100-1/.built] Error 1
make[3]: Leaving directory '/home/x86.5.4/lede/feeds/passwall_packages/naiveproxy'
time: package/feeds/passwall_packages/naiveproxy/compile#4.95#10.70#14.64
ERROR: package/feeds/passwall_packages/naiveproxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall_packages/naiveproxy/compile] Error 1
make[2]: Leaving directory '/home/x86.5.4/lede'
make[1]: *** [package/Makefile:110: /home/x86.5.4/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

如不应在此提issue烦请关闭本issue。

bi4tms, 2023-02-25:
和Naiveproxy作者沟通了一下，提示clang++: Exec format error，作者认为是编译环境问题。
请教如何修改@coolsnowwolf

WYC-2020, 2023-02-26:
那个仓库的去哪里提

firkerword, 2023-03-04:
正常编译啊  三个内核版本都没问题

honnbin, 2023-04-19:
wls2用ubuntu22.04，全新环境下面，默认编译一次过，加上naiveproxy就报上面相同的错误，放狗找了一圈，也没解决

honnbin, 2023-04-19:
试了wsl下面按ubuntu20和22都不行，好像是clang和wsl的兼容性问题，换个虚拟机装ubuntu再编译就好了

coolsnowwolf, 2026-09-05:
日志中的真实失败是 naiveproxy 自带 clang++ 在 WSL 环境返回 `Exec format error`，评论已确认换 Linux 虚拟机后可正常编译；这是主机/工具链架构环境问题，不是 LEDE 包源码可修复的问题。

closed 2026-09-05 by coolsnowwolf