coolsnowwolf/lede #10934: 执行opkg list-upgradable报错

helloyan, 2023-02-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：执行opkg list-upgradable报错

### (2) 路由器型号和固件版本

A：OpenWrt R23.2.14 / LuCI Master (git-23.055.70076-3ace903)
      P&W R619AC 128M

### (3) 详细日志
```
root@OpenWrt:~# opkg list-upgradable
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libatomic1 and libatomic1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libstdcpp6 and libstdcpp6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libatomic1 and libatomic1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libstdcpp6 and libstdcpp6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libstdcpp6 and libstdcpp6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libatomic1 and libatomic1) providing same name marked HOLD or PREFER. Using latest.
iwinfo - 2021-06-28-c9b1672f-5 - 2023-02-06-c7eb8ebe-1
openssh-sftp-server - 9.1p1-1 - 9.2p1-1
kmod-usb-core - 5.4.232-1 - 5.15.94-1
ubus - 2022-06-01-2bebf93c-1 - 2022-06-15-9913aa61-1
iw - 5.16-1 - 5.19-1
kmod-crypto-manager - 5.4.232-1 - 5.15.94-1
rpcd - 2022-09-21-8c852b65-1 - 2023-01-21-c0df2a7a-2
kmod-nfnetlink - 5.4.232-1 - 5.15.94-1
libubus-lua - 2022-06-01-2bebf93c-1 - 2022-06-15-9913aa61-1
kmod-crypto-hash - 5.4.232-1 - 5.15.94-1
kmod-tun - 5.4.232-1 - 5.15.94-1
kmod-crypto-aead - 5.4.232-1 - 5.15.94-1
libiwinfo-lua - 2021-06-28-c9b1672f-5 - 2023-02-06-c7eb8ebe-1
strongswan-mod-sha1 - 5.9.5-1 - 5.9.8-5
libdbus - 1.13.18-1 - 1.13.18-12
kmod-crypto-md5 - 5.4.232-1 - 5.15.94-1
openssl-util - 1.1.1t-1 - 3.0.8-2
kmod-lib-crc-ccitt - 5.4.232-1 - 5.15.94-1
openssh-server - 9.1p1-1 - 9.2p1-1
kmod-ipsec - 5.4.232-1 - 5.15.94-1
libustream-openssl20201210 - 2022-12-08-9217ab46-1 - 2022-12-08-9217ab46-2
kmod-gre - 5.4.232-1 - 5.15.94-1
libprotobuf-c - 1.4.0-1 - 1.4.0-2
kmod-pppox - 5.4.232-1 - 5.15.94-1
strongswan-mod-openssl - 5.9.5-1 - 5.9.8-5
kmod-ipt-conntrack - 5.4.232-1 - 5.15.94-1
kmod-nf-reject - 5.4.232-1 - 5.15.94-1
base-files - 63-r5761-efb924226 - 1526-r22141-dd8a4a8c34
kmod-nf-nat - 5.4.232-1 - 5.15.94-1
netifd - 2021-06-04-50381d0a-1 - 2022-12-30-81c1fbcb-2
ddns-scripts - 2.7.8-5 - 2.8.2-35
zerotier - 1.10.2-1 - 1.10.3-1
kmod-crypto-hmac - 5.4.232-1 - 5.15.94-1
kmod-crypto-sha256 - 5.4.232-1 - 5.15.94-1
ubusd - 2022-06-01-2bebf93c-1 - 2022-06-15-9913aa61-1
kmod-crypto-acompress - 5.4.232-1 - 5.15.94-1
strongswan - 5.9.5-1 - 5.9.8-5
iptables-mod-ipsec - 1.8.7-2 - 1.8.8-1
kmod-iptunnel - 5.4.232-1 - 5.15.94-1
[... 178 lines cut ...]
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-ip6tables
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ip6tables found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-lib-crc32c
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-crc32c found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-usb3
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-crypto-ghash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-pppol2tp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-hwmon-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-crypto-ccm
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-ipt-ipset
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-leds-gpio
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-gpio-button-hotplug
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-cfg80211
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-crypto-cmac
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-crypto-gcm
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-mac80211
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-nf-nathelper-extra
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-ath
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-ath10k-ct
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.94-1-e1cf528cf8266f179489605beea15a22) for kmod-lib-lzo
```

Her0R0cket, 2023-02-27:
更新源的在线内核版本和本地版本不一样,很正常的.内核版本不一样是不能更新的,这个版本不只是大小版本,还有那串类似hash的字符.
      不是稳定版本,内核不会固定,基本不能在线安装内核模块.

helloyan, 2023-02-27:
我如果想要使用github action编译且可以使用在线源，应该选择哪个分支？

closed 2023-02-28 by aiamadeus