coolsnowwolf/lede #10940: 建议默认启用sha512模块

fjzg, 2023-02-28:
## 1.
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译lede时，系统默认没启用sha512支持，导致依赖此模块的程序无法启动
### (1) 具体问题

A：编译arm64和x86架构lede时，由于系统默认没有启用sha512支持，导致依赖此模块的程序无法启动。
sha512作为常用的摘要算法，很多linux发行版都默认编译此模块，建议lede默认启用 Cryptographic API modules>>kmod-crypto-sha512

### (2) 路由器型号和固件版本

A：x86_86、R2S（Arm65）

### (3) 详细日志

A：
Mon Feb 27 18:37:26 2023 daemon.info : 15[CFG] selected proposal: ESP:AES_CBC_128/HMAC_SHA1_96/NO_EXT_SEQ
Mon Feb 27 18:37:26 2023 daemon.info : 15[IKE] received 3600s lifetime, configured 0s
Mon Feb 27 18:37:26 2023 daemon.info : 15[IKE] received 250000000 lifebytes, configured 0
Mon Feb 27 18:37:26 2023 daemon.info : 15[ENC] generating QUICK_MODE response 1 [ HASH SA No ID ID ]
Mon Feb 27 18:37:26 2023 daemon.info : 14[ENC] parsed QUICK_MODE request 1 [ HASH ]
Mon Feb 27 18:37:26 2023 daemon.info : 14[KNL] received netlink error: No such file or directory (2)
Mon Feb 27 18:37:26 2023 daemon.info : 14[KNL] unable to add SAD entry with SPI c29143b0 (FAILED)
Mon Feb 27 18:37:26 2023 kern.info kernel: [ 2135.622522] DRBG: could not allocate digest TFM handle: hmac(sha512)
Mon Feb 27 18:37:26 2023 daemon.info : 14[KNL] received netlink error: No such file or directory (2)
Mon Feb 27 18:37:26 2023 daemon.info : 14[KNL] unable to add SAD entry with SPI 17a44d62 (FAILED)
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Mon Feb 27 18:37:26 2023 daemon.info : 14[KNL] deleting policy 192.168.1.175/32[udp/l2f] === 192.168.1.1/32[udp/l2f] in failed, not found
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] queueing QUICK_DELETE task
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] activating new tasks
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE]   activating QUICK_DELETE task
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] sending DELETE for ESP CHILD_SA with SPI 17a44d62
Mon Feb 27 18:37:26 2023 daemon.info : 14[ENC] generating INFORMATIONAL_V1 request 2961201030 [ HASH D ]
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] activating new tasks
Mon Feb 27 18:37:26 2023 daemon.info : 14[IKE] nothing to initiate
Mon Feb 27 18:37:26 2023 kern.info kernel: [ 2135.705839] DRBG: could not allocate digest TFM handle: hmac(sha512)
Mon Feb 27 18:38:01 2023 daemon.info : 04[ENC] parsed INFORMATIONAL_V1 request 3064858232 [ HASH D ]
Mon Feb 27 18:38:01 2023 daemon.info : 04[IKE] received DELETE for ESP CHILD_SA with SPI 17a44d62
Mon Feb 27 18:38:01 2023 daemon.info : 04[IKE] CHILD_SA not found, ignored
Mon Feb 27 18:38:01 2023 daemon.info : 07[ENC] parsed INFORMATIONAL_V1 request 3412929471 [ HASH D ]
Mon Feb 27 18:38:01 2023 daemon.info : 07[IKE] received DELETE for IKE_SA L2TP-IKEv1-PSK[1]
Mon Feb 27 18:38:01 2023 daemon.info : 07[IKE] deleting IKE_SA L2TP-IKEv1-PSK[1] between 192.168.1.1[192.168.1.1]...192.168.1.175[192.168.1.175]
Mon Feb 27 18:38:01 2023 authpriv.info : 07[IKE] deleting IKE_SA L2TP-IKEv1-PSK[1] between 192.168.1.1[192.168.1.1]...192.168.1.175[192.168.1.175]
Mon Feb 27 18:38:01 2023 daemon.info : 07[IKE] IKE_SA L2TP-IKEv1-PSK[1] state change: ESTABLISHED => DELETING
Mon Feb 27 18:38:01 2023 daemon.info : 07[IKE] IKE_SA L2TP-IKEv1-PSK[1] state change: DELETING => DELETING
Mon Feb 27 18:38:01 2023 daemon.info : 07[IKE] IKE_SA L2TP-IKEv1-PSK[1] state change: DELETING => DESTROYING

fjzg, 2023-02-28:
缺少该模块会导致依赖此模块的程序异常，常见的如ipsec在交换摘要算法阶段时失败

WYC-2020, 2023-02-28:
与其加入默认，倒不如更新ipsec的依赖，在确定ipsec没有512不能运行的情况下

fjzg, 2023-02-28:
> 与其加入默认，倒不如更新ipsec的依赖，在确定ipsec没有512不能运行的情况下

sha512作为一个公共算法，不只ipsec会用到，有些docker容器默认直接调用系统内的模块，缺少这个docker容器也会跟着出问题

WYC-2020, 2023-02-28:
> > 与其加入默认，倒不如更新ipsec的依赖，在确定ipsec没有512不能运行的情况下
> 
> sha512作为一个公共算法，不只ipsec会用到，有些docker容器默认直接调用系统内的模块，缺少这个docker容器也会跟着出问题

我没编译这个docker也没问题，看l会不会加吧，反正我clone的，到时候我可以revert哈哈

coolsnowwolf, 2026-09-05:
核验当前 master：`kmod-crypto-sha512` 已作为独立可选内核包提供（`package/kernel/linux/modules/crypto.mk`），并被需要它的内核功能显式依赖；各 target 配置按硬件/镜像需求选择，通用配置并未默认把所有摘要算法打入每个固件。原日志是 strongSwan/IPsec 使用 HMAC-SHA512 时未选择该可选模块，属于固件配置取舍而非当前代码缺陷。请在目标配置中启用 `Kernel modules -> Cryptographic API modules -> kmod-crypto-sha512`。本 issue 不作全局默认启用，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf