coolsnowwolf/lede #10941: R2S内核出现NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out

chobitsos, 2023-02-28:
反馈bug

## 1.关于你要提交的问题

R2S内核出现transmit queue 0 timed out

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

还没有类似issue

## 2. 详细叙述

内核日志出现NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out

### (1) 具体问题

[A：最新编译的](https://github.com/coolsnowwolf/lede/commit/a2c61871bc438e4293eb9f99489557301f9cd913)commit中出现
NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out，具体出现原因未知，无法复现，固件跑了24小时出现

### (2) 路由器型号和固件版本

NanoPi R2S 
OpenWrt R23.2.14 / LuCI Master (git-23.054.10809-0a8d99a)
Kernel Version | 5.15.94

### (3) 详细日志

`Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Tue Feb 28 05:11:38 2023 kern.notice kernel: [    0.000000] Linux version 5.15.94 (seed@seed-virtual-machine) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5760-53b83ba81) 11.3.0, GNU ld (GNU Binutils) 2.39) #0 SMP PREEMPT Thu Feb 23 03:02:26 2023
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Machine model: FriendlyElec NanoPi R2S
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Zone ranges:
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000]   DMA32    empty
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000]   Normal   empty
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Early memory node ranges
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] cma: Reserved 64 MiB at 0x0000000039c00000
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s33368 r8192 d32168 u73728
Tue Feb 28 05:11:38 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s33368 r8192 d32168 u73728 alloc=18*4096
Tue Feb 28 05:11:38 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Tue Feb 28 05:11:38 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Memory: 944480K/1046528K available (8576K kernel code, 2372K rwdata, 2452K rodata, 1664K init, 721K bss, 36512K reserved, 65536K cma-reserved)
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] 	Trampoline variant of Tasks RCU enabled.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
Tue Feb 28 05:11:38 2023 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
[... 368 lines cut ...]
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.673830]  el1h_64_irq+0x74/0x78
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.674150]  cpuidle_enter_state+0x134/0x31c
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.674556]  cpuidle_enter+0x38/0x50
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.674896]  do_idle+0x1d0/0x224
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.675206]  cpu_startup_entry+0x24/0x60
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.675574]  rest_init+0xe0/0xf0
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.675882]  arch_call_rest_init+0x10/0x1c
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.676270]  start_kernel+0x51c/0x540
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.676616]  __primary_switched+0xbc/0xc4
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.676996] ---[ end trace 8471701bc2c0c7f7 ]---
Tue Feb 28 10:08:36 2023 kern.warn kernel: [17832.677456] r8152 4-1:1.0 eth1: Tx timeout
Tue Feb 28 10:08:38 2023 kern.warn kernel: [17834.723499] r8152 4-1:1.0 eth1: Tx status -2
Tue Feb 28 10:08:38 2023 kern.warn kernel: [17834.723983] r8152 4-1:1.0 eth1: Tx status -2
Tue Feb 28 10:08:38 2023 kern.warn kernel: [17834.724479] r8152 4-1:1.0 eth1: Tx status -2
Tue Feb 28 10:08:38 2023 kern.warn kernel: [17834.725066] r8152 4-1:1.0 eth1: Tx status -2
Tue Feb 28 10:08:39 2023 kern.info kernel: [17835.542475] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Tue Feb 28 10:08:39 2023 kern.err kernel: [17835.575379] r8152 4-1:1.0 eth1: Invalid ether addr 00:00:00:00:00:00
Tue Feb 28 10:08:39 2023 kern.info kernel: [17835.576010] r8152 4-1:1.0 eth1: Random ether addr 26:fa:12:0f:0a:6d
Tue Feb 28 10:08:39 2023 kern.notice kernel: [17835.583350] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Tue Feb 28 10:57:40 2023 kern.warn kernel: [20776.630350] hrtimer: interrupt took 55710 ns
Tue Feb 28 12:18:06 2023 authpriv.info dropbear[18137]: Child connection from 192.168.1.22:59665
Tue Feb 28 12:18:08 2023 authpriv.notice dropbear[18137]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:1wcZPOEBXjo9U88OXQdD6aP85sJYFecj+tk2iRn5VYg from 192.168.1.22:59665
Tue Feb 28 12:30:49 2023 authpriv.info dropbear[18137]: Exit (root) from <192.168.1.22:59665>: Exited normally
Tue Feb 28 12:42:11 2023 daemon.err uhttpd[1100]: cat: can't open '/etc/bench.log': No such file or directory
`

XiaoBinin, 2023-03-30:
换成 kmod-usb-net-rtl8152-vendor 应该能解决

NuoFang6, 2024-09-05:
> 换成 kmod-usb-net-rtl8152-vendor 应该能解决

我也发现了这个问题，驱动是 kmod-usb-net-rtl8152-vendor v2.18.1.20240701。

[image]

出现后立马断网，需要重新插拔网线