coolsnowwolf/lede #10942: 网络诊断 选择了 ipv4，结果都是 ipv6

tanst, 2023-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无论 ping 还是 traceroute，
域名输入：www.qq.com
诊断 选择了 ipv4，结果都是 ipv6 的结果
但是默认的 openwrt.org 选择 ipv4 正常

### (2) 路由器型号和固件版本

A：X86 git-23.050.56566-a8351ec

### (3) 详细日志

A：

yanxin152133, 2023-03-05:
[image]
你连截图都没怎么让人给你分析问题，没有日志最起码把你网络信息以及问题现象发下吧。

tanst, 2023-03-06:
> [image]
> 
> 你连截图都没怎么让人给你分析问题，没有日志最起码把你网络信息以及问题现象发下吧。

[image]
就是 测试 ipv4 的除了默认的 openwrt.org 是正常的，其余的域名都显示 ipv6 的测试结果

yanxin152133, 2023-03-06:
把ipv6关了不就好了

tanst, 2023-03-06:
> 把ipv6关了不就好了

那不行，我上传主要靠 ipv6 呢

yanxin152133, 2023-03-06:
https://github.com/coolsnowwolf/lede/issues/10693
这个优先级的问题？

coolsnowwolf, 2026-09-05:
没有提供 DNS 配置、`ping -4/-6`、解析结果或诊断脚本日志；域名 A/AAAA 解析优先级和本地网络环境均可能导致该现象，当前无法形成 LEDE 代码修复。

closed 2026-09-05 by coolsnowwolf