coolsnowwolf/lede #10943: mwan3版本太老了，能不能升级一下啊，实在太痛苦了

GreatMichaelLee, 2023-02-28:
现在这个版本的mwan3, 问题很多，表现为 1）经常卡在点应用配置后不动 2）每次重启软路由后多拨不能自动完成接口的tracking status的翻转, mwan3 status显示禁用，vwan接口显示为offline tracking是down, 上不了网，非要手工apply一下才行。

这个也提了好多次了，每次都被大佬们否了，也不知道什么原因，软路由用多拨的相信用户也很多，能否考虑一下？

GreatMichaelLee, 2023-02-28:
我也不知道这个适不适合提issue, 但提在issue区，还是给点问题描述吧：

每次重启后都是这样的，网络LAN->外网不通，软路由是通外网的
[image]
要到mwan3里区应用一下配置才通，大多数情况应用配置可以complete，但有个别时候应用配置也会卡死。

应用完成是这样的
[image]

GreatMichaelLee, 2023-02-28:
log 如下：后半截Wed Mar  1 01:47:06是我手工重新应用更改后起来的，此前是系统重启后，不知怎么出现了 sysntpd crash
然后uhttpd一直打dump terminated,好像这个阻止了mwan3的正常运作....感觉是这样。。。。

Wed Mar  1 01:42:11 2023 daemon.notice ttyd[13771]: [2023/03/01 01:42:11:8211] N: rops_handle_POLLIN_netlink: DELADDR
Wed Mar  1 01:42:11 2023 daemon.notice netifd: Network device 'pppoe-vwan2' link is up
Wed Mar  1 01:42:11 2023 daemon.notice netifd: Interface 'vwan2' is now up
Wed Mar  1 01:42:12 2023 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 8 crashes, 0 seconds since last crash
Wed Mar  1 01:44:14 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
Wed Mar  1 01:44:33 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
Wed Mar  1 01:44:36 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
Wed Mar  1 01:44:47 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
Wed Mar  1 01:44:54 2023 authpriv.info dropbear[9955]: Child connection from 192.168.2.14:51851
Wed Mar  1 01:44:54 2023 authpriv.notice dropbear[9955]: Password auth succeeded for 'root' from 192.168.2.14:51851
Wed Mar  1 01:45:04 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
Wed Mar  1 01:45:41 2023 daemon.err uhttpd[3514]: /usr/sbin/mwan3: line 1141: ip6tables: not found
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:43 2023 daemon.err uhttpd[3514]: Dump terminated
[... 190 lines cut ...]
Wed Mar  1 01:45:44 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:44 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:44 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:44 2023 daemon.err uhttpd[3514]: Dump terminated
Wed Mar  1 01:45:45 2023 daemon.err uhttpd[3514]: /usr/sbin/mwan3: line 1141: ip6tables: not found
Wed Mar  1 01:46:36 2023 daemon.err miniupnpd[12053]: Failed to get ip address for interface
[... cut ...]
Wed Mar  1 01:47:11 2023 user.notice ddns-scripts[26898]: WAN4: PID '26898' terminated by 'SIGTERM' at 2023-03-01 01:47
Wed Mar  1 01:47:12 2023 user.notice mwan3[31857]: Execute ifup event on interface vwan2 (pppoe-vwan2)
Wed Mar  1 01:47:12 2023 user.notice mwan3[31857]: Starting tracker on interface vwan2 (pppoe-vwan2)
Wed Mar  1 01:47:12 2023 user.info mwan3rtmon[29141]: Detect rtchange event.
Wed Mar  1 01:47:12 2023 user.notice firewall: Reloading firewall due to ifup of vwan2 (pppoe-vwan2)
Wed Mar  1 01:47:13 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of vwan2 (pppoe-vwan2)

WYC-2020, 2023-02-28:
知道每次被否还不断提，别催，要更自然会更

GreatMichaelLee, 2023-03-01:
求大佬开恩，赶紧更吧，已经痛苦好多年了//cry

coolsnowwolf, 2023-03-01:
自己测试好，提个 PR 即可

GreatMichaelLee, 2023-03-01:
> 自己测试好，提个 PR 即可

不懂代码啊，只会简单的小修补.....大佬....伸手党，别bs我......

a904055262, 2023-03-01:
如果是x86的，直接用原版openwrt不就行了。。

shenlijun, 2023-03-02:
普通家用的话，拉个千兆宽带，买个2.5G光猫，就不要跟mwan较劲了

GreatMichaelLee, 2023-03-02:
> 普通家用的话，拉个千兆宽带，买个2.5G光猫，就不要跟mwan较劲了

我是电信3拨+移动1拨。。。。双2.5G wan.....

GreatMichaelLee, 2023-03-02:
这个版本的mwan3 直接restart那简直就是灾难…

 
启动项自定义脚本 mwan3 restart 一下有用吗?

zopargr, 2023-03-05:
怎么才能重启mwan3啊，我现在想重启都会卡住

shenlijun, 2023-03-05:
> > 普通家用的话，拉个千兆宽带，买个2.5G光猫，就不要跟mwan较劲了
> 
> 我是电信3拨+移动1拨。。。。双2.5G wan.....

MD骚鸡╰(*°▽°*)╯

tanst, 2023-03-06:
我写了一个脚本自动检测 mwan3 状态，用过 1 年多了，大家可以试试

1. 新建一个`mwan3_status.sh`脚本文件，比如：`/mnt/sda3/my.files/mwan3_status.sh`，内容如下：
```
#!/bin/sh
while true; do
    logfile="/tmp/mwan3_status.log"
    exec 1>>$logfile 2>&1

    status=$(/usr/sbin/mwan3 status | grep error)
    if [ -z $status ]; then
        echo $(echo '[' && date "+%Y-%m-%d %H:%M:%S" && echo ']' && echo "OK")
    else
        echo $(echo '[' && date "+%Y-%m-%d %H:%M:%S" && echo ']' && echo "$status")
        echo "mwan3 restarting..."
        /usr/sbin/mwan3 restart
        echo "Success"
    fi

    row_number=$(grep -c "" $logfile)
    delete_number=$(expr $row_number - 500)
    if [ ${delete_number} -gt 0 ]; then
        sed -i "1,${delete_number}d" $logfile
    fi

    sleep 60
done
```
2. 把文件路径加入到启动项文件里：`/etc/rc.local`，或者 webui 里 `系统-启动项-本地启动脚本` 里面
> 原理就是开机运行脚本，1分钟检测一次 mwan3 状态，如果包含`error`就自动重启 mwan3，然后一直循环。

chiyu521, 2023-03-15:
我也是多拨用户 用的去年12.08的源码  没办法升级 同样的问题

Jieoz, 2023-03-22:
啊，我这加上mwan3根本都没办法编译完成，既然这样的话那我就养老吧

GreatMichaelLee, 2023-03-22:
肯定要做一些改动和适配的…需要一点开发能力

alexliu269, 2023-05-05:
更新了以后貌似好了，你的呢

GreatMichaelLee, 2023-05-05:
你怎么更新的

chiyu521, 2023-11-25:
问题解决 
替换 MWAN3  
https://github.com/0xACE8/luci-app-mwan3
https://github.com/0xACE8/mwan3

GreatMichaelLee, 2023-11-25:
怎么替换？

发自我的iPhone

 

 https://github.com/0xACE8/luci-app-mwan3
 https://github.com/0xACE8/mwan3

QingYis, 2024-04-28:
> 问题解决 替换 MWAN3 https://github.com/0xACE8/luci-app-mwan3 https://github.com/0xACE8/mwan3

我这边试过了，替换了，mwan3 restart是可以成功了，但是那些分流、负载均衡的规则不会生效

mozuipi, 2024-05-02:
> > 问题解决 替换 MWAN3 https://github.com/0xACE8/luci-app-mwan3 https://github.com/0xACE8/mwan3
> 
> 我这边试过了，替换了，mwan3 restart是可以成功了，但是那些分流、负载均衡的规则不会生效

我也试了。。分流、负载均衡的规则确实不会生效！哎

coolsnowwolf, 2026-09-05:
当前 feeds/packages 的 mwan3 已为 2.12.2，已包含主线针对接口 tracking、热插拔及应用配置流程的后续修复；该 issue 针对的旧版本已不再使用。无需额外改动，现关闭。

closed 2026-09-05 by coolsnowwolf