coolsnowwolf/lede #10945: 更新源码后编译出错

mitsukileung, 2023-03-01:
ipq807X / AX6

  DTC     arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dtb
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:348.1-5 Label or path dp1 not found
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:356.1-5 Label or path dp2 not found
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:364.1-5 Label or path dp3 not found
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:372.1-5 Label or path dp4 not found
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:380.1-5 Label or path dp5 not found
Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:389.1-9 Label or path dp6_syn not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dtb] Error 1
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/qcom] Error 2
make[5]: *** [Makefile:1442: dtbs] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.94'
make[4]: *** [Makefile:28: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.94/.modules] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ipq807x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#12.43#3.56#15.87
    ERROR: target/linux failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2023-03-01:
并没有适配 5.15 ，自行解决

zxlhhyccc, 2023-03-01:
> ipq807X / AX6
>
> DTC arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dtb Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:348.1-5 Label or path dp1 not found Error: [...]

需要将dts文件中的&dp1、&dp2、。。。修改如下即可正常编译：
````
&soc {
	dp1: dp1 {
		status = "okay";
		phy-handle = <&qca8075_1>;
		label = "lan1";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
	};

	dp2: dp2 {
		status = "okay";
		phy-handle = <&qca8075_2>;
		label = "lan2";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
	};

	dp3: dp3 {
		status = "okay";
		phy-handle = <&qca8075_3>;
		label = "lan3";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
	};

	dp4: dp4 {
		status = "okay";
		phy-handle = <&qca8075_4>;
		label = "lan4";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
	};

	dp5: dp5 {
		status = "okay";
		phy-handle = <&qca8081>;
		label = "wan";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
		mac-address-increment = <1>;
	};

	dp6: dp6_syn {
		status = "okay";
		phy-handle = <&aqr113c>;
		label = "10g";
		nvmem-cells = <&macaddr_lan>;
		nvmem-cell-names = "mac-address-ascii";
	};
};
````

mitsukileung, 2023-03-01:
> > ipq807X / AX6
> > DTC arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dtb Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:348.1-5 Label or path dp1 not found Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:356.1-5 Label or path dp2 not found Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:364.1-5 Label or path dp3 not found Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:372.1-5 Label or path dp4 not found Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:380.1-5 Label or path dp5 not found Error: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dts:389.1-9 Label or path dp6_syn not found FATAL ERROR: Syntax error parsing input tree make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/qcom/ipq8074-nbg7815.dtb] Error 1 make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/qcom] Error 2 make[5]: *** [Makefile:1442: dtbs] Error 2 make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.94' make[4]: *** [Makefile:28: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.94/.modules] Error 2 make[4]: Leaving directory '/workdir/openwrt/target/linux/ipq807x' make[3]: *** [Makefile:11: compile] Error 2 make[3]: Leaving directory '/workdir/openwrt/target/linux' time: target/linux/compile#12.43#3.56#15.87 ERROR: target/linux failed to build. make[2]: Leaving directory '/workdir/openwrt' make[2]: *** [target/Makefile:30: target/linux/compile] Error 1 make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2 Error: Process completed with exit code 2.
> 
> 需要将dts文件中的&dp1、&dp2、。。。修改如下即可正常编译，但nbg7815能否正常运行就不知道了，无机器测试：
> 
> ```
> &soc {
> 	dp1: dp1 {
> 		status = "okay";
> 		phy-handle = <&qca8075_1>;
> 		label = "lan1";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 	};
> 
> 	dp2: dp2 {
> 		status = "okay";
> 		phy-handle = <&qca8075_2>;
> 		label = "lan2";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 	};
> 
> 	dp3: dp3 {
> 		status = "okay";
> 		phy-handle = <&qca8075_3>;
> 		label = "lan3";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 	};
> 
> 	dp4: dp4 {
> 		status = "okay";
> 		phy-handle = <&qca8075_4>;
> 		label = "lan4";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 	};
> 
> 	dp5: dp5 {
> 		status = "okay";
> 		phy-handle = <&qca8081>;
> 		label = "wan";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 		mac-address-increment = <1>;
> 	};
> 
> 	dp6: dp6_syn {
> 		status = "okay";
> 		phy-handle = <&aqr113c>;
> 		label = "10g";
> 		nvmem-cells = <&macaddr_lan>;
> 		nvmem-cell-names = "mac-address-ascii";
> 	};
> };
> ```

谢谢，我试试

zxlhhyccc, 2023-03-02:
@mitsukileung 是否可以正常编译了？

mitsukileung, 2023-03-02:
> @mitsukileung 是否可以正常编译了？

大雕更新源码后可以正常编译了。

closed 2023-03-02 by mitsukileung