coolsnowwolf/lede #10951: 编译报错

Lile1, 2023-03-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

编译报错

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
time: package/feeds/packages/libnatpmp/compile#0.27#0.28#0.48
make[3]: Entering directory '/home/lilei/lede/feeds/packages/net/miniupnpc'
mkdir -p /home/lilei/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/lilei/lede/tmp/.root-copy.flock -c 'cp -fpR /home/lilei/lede/build_dir/target-x86_64_musl/miniupnpc-2.2.3/.pkgdir/libminiupnpc/. /home/lilei/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/lilei/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libminiupnpc_installed
echo "libminiupnpc" >> /home/lilei/lede/staging_dir/target-x86_64_musl/pkginfo/miniupnpc.default.install
make[3]: Leaving directory '/home/lilei/lede/feeds/packages/net/miniupnpc'
time: package/feeds/packages/miniupnpc/compile#0.25#0.33#0.50
make[3]: Entering directory '/home/lilei/lede/feeds/packages/net/zerotier'
rm -f /home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2/.built
touch /home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2=ZeroTierOne-1.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2=ZeroTierOne-1.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed " make  -C /home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2=ZeroTierOne-1.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ZT_EMBEDDED=1 ZT_SSO_SUPPORTED=0 DEFS="" OSTYPE="Linux"  one;
make[4]: Entering directory '/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2'
x86_64-openwrt-linux-musl-g++ -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2=ZeroTierOne-1.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -Wall -Wno-deprecated -std=c++17 -pthread -Izeroidc/target -isystem ext -DNDEBUG -DZT_USE_MINIUPNPC -DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=2 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H -DZT_USE_X64_ASM_SALSA2012 -DZT_USE_FAST_X64_ED25519 -msse -msse2 -Wl,-z,relro,-z,now -o zerotier-one node/AES.o node/AES_aesni.o node/AES_armcrypto.o node/C25519.o node/Capability.o node/CertificateOfMembership.o node/CertificateOfOwnership.o node/Identity.o node/IncomingPacket.o node/InetAddress.o node/Membership.o node/Multicaster.o node/Network.o node/NetworkConfig.o node/Node.o node/OutboundMulticast.o node/Packet.o node/Path.o node/Peer.o node/Poly1305.o node/Revocation.o node/Salsa20.o node/SelfAwareness.o node/SHA512.o node/Switch.o node/Tag.o node/Topology.o node/Trace.o node/Utils.o node/Bond.o ext/x64-salsa2012-asm/salsa2012.o ext/ed25519-amd64-asm/choose_t.o ext/ed25519-amd64-asm/consts.o ext/ed25519-amd64-asm/fe25519_add.o ext/ed25519-amd64-asm/fe25519_freeze.o ext/ed25519-amd64-asm/fe25519_mul.o ext/ed25519-amd64-asm/fe25519_square.o ext/ed25519-amd64-asm/fe25519_sub.o ext/ed25519-amd64-asm/ge25519_add_p1p1.o ext/ed25519-amd64-asm/ge25519_dbl_p1p1.o ext/ed25519-amd64-asm/ge25519_nielsadd2.o ext/ed25519-amd64-asm/ge25519_nielsadd_p1p1.o ext/ed25519-amd64-asm/ge25519_p1p1_to_p2.o ext/ed25519-amd64-asm/ge25519_p1p1_to_p3.o ext/ed25519-amd64-asm/ge25519_pnielsadd_p1p1.o ext/ed25519-amd64-asm/heap_rootreplaced.o ext/ed25519-amd64-asm/heap_rootreplaced_1limb.o ext/ed25519-amd64-asm/heap_rootreplaced_2limbs.o ext/ed25519-amd64-asm/heap_rootreplaced_3limbs.o ext/ed25519-amd64-asm/sc25519_add.o ext/ed25519-amd64-asm/sc25519_barrett.o ext/ed25519-amd64-asm/sc25519_lt.o ext/ed25519-amd64-asm/sc25519_sub_nored.o ext/ed25519-amd64-asm/ull4_mul.o ext/ed25519-amd64-asm/fe25519_getparity.o ext/ed25519-amd64-asm/fe25519_invert.o ext/ed25519-amd64-asm/fe25519_iseq.o ext/ed25519-amd64-asm/fe25519_iszero.o ext/ed25519-amd64-asm/fe25519_neg.o ext/ed25519-amd64-asm/fe25519_pack.o ext/ed25519-amd64-asm/fe25519_pow2523.o ext/ed25519-amd64-asm/fe25519_setint.o ext/ed25519-amd64-asm/fe25519_unpack.o ext/ed25519-amd64-asm/ge25519_add.o ext/ed25519-amd64-asm/ge25519_base.o ext/ed25519-amd64-asm/ge25519_double.o ext/ed25519-amd64-asm/ge25519_double_scalarmult.o ext/ed25519-amd64-asm/ge25519_isneutral.o ext/ed25519-amd64-asm/ge25519_multi_scalarmult.o ext/ed25519-amd64-asm/ge25519_pack.o ext/ed25519-amd64-asm/ge25519_scalarmult_base.o ext/ed25519-amd64-asm/ge25519_unpackneg.o ext/ed25519-amd64-asm/hram.o ext/ed25519-amd64-asm/index_heap.o ext/ed25519-amd64-asm/sc25519_from32bytes.o ext/ed25519-amd64-asm/sc25519_from64bytes.o ext/ed25519-amd64-asm/sc25519_from_shortsc.o ext/ed25519-amd64-asm/sc25519_iszero.o ext/ed25519-amd64-asm/sc25519_mul.o ext/ed25519-amd64-asm/sc25519_mul_shortsc.o ext/ed25519-amd64-asm/sc25519_slide.o ext/ed25519-amd64-asm/sc25519_to32bytes.o ext/ed25519-amd64-asm/sc25519_window4.o ext/ed25519-amd64-asm/sign.o controller/EmbeddedNetworkController.o controller/DBMirrorSet.o controller/DB.o controller/FileDB.o controller/LFDB.o controller/PostgreSQL.o osdep/EthernetTap.o osdep/ManagedRoute.o osdep/Http.o osdep/OSUtils.o service/SoftwareUpdater.o service/OneService.o osdep/LinuxEthernetTap.o osdep/LinuxNetLink.o osdep/PortMapper.o ext/http-parser/http_parser.o one.o -lminiupnpc -lnatpmp
/home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lilei/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/lib/crt1.o: in function `_start_c':
/home/lilei/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/musl-1.2.3/crt/crt1.c:18: undefined reference to `main'
collect2: error: ld returned 1 exit status
make[4]: *** [make-linux.mk:352: zerotier-one] Error 1
make[4]: Leaving directory '/home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2'
make[3]: *** [Makefile:84: /home/lilei/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.10.2/.built] Error 2
make[3]: Leaving directory '/home/lilei/lede/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#0.84#0.45#1.15
    ERROR: package/feeds/packages/zerotier failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/zerotier/compile] Error 1
make[2]: Leaving directory '/home/lilei/lede'
make[1]: *** [package/Makefile:110: /home/lilei/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lilei/lede'
make: *** [/home/lilei/lede/include/toplevel.mk:231: world] Error 2

zheshifandian, 2023-03-02:
先换一下编译的musl的版本

WYC-2020, 2023-03-03:
这种 基础库出问题的 自己多排查，如果这玩意都有问题早就炸了，看没什么反馈，自己make distclean 然后在编译，别动不动就提issue 提了也不关看到老火

Lile1, 2023-03-03:
> 这种 基础库出问题的 自己多排查，如果这玩意都有问题早就炸了，看没什么反馈，自己make distclean 然后在编译，别动不动就提issue 提了也不关看到老火

> 这种 基础库出问题的 自己多排查，如果这玩意都有问题早就炸了，看没什么反馈，自己make distclean 然后在编译，别动不动就提issue 提了也不关看到老火

粪吃多了吧，上来就开喷，你恼火个 j8 ,不知道还以为你是作者。 傻子一个。

closed 2023-03-03 by aiamadeus