coolsnowwolf/lede #10952: rtl8821cu无法驱动

zhangsd20, 2023-03-02:
您好,

我编译的固件为什么加载后插rtl8821cu出现下面打印.无法驱动起来?

[ 25.952736] usb 1-1.2: new high-speed USB device number 3 using ehci-platform
[ 26.415237] rtw_8821cu 1-1.2:1.2: Firmware version 24.11.0, H2C version 12
[ 26.657188] rtw_8821cu 1-1.2:1.2: rfe 38 isn't supported
[ 26.677542] rtw_8821cu 1-1.2:1.2: failed to setup chip efuse info
[ 26.683930] rtw_8821cu 1-1.2:1.2: failed to setup chip information
[ 26.902732] usb 1-1.3: new full-speed USB device number 4 using ehci-platform
[ 27.372732] usb 1-1.4: new full-speed USB device number 5 using ehci-platform

aiamadeus, 2023-03-02:
螃蟹的 firmware 限制，就算是 rtl8821ce 要开 ap 也不行。

zhangsd20, 2023-03-03:
我在openwrt之前版本(kernel-4.xx)上移植过rtl8821cu驱动,ap/sta都是可以起来的;移植的比较简陋;
最近发现内核5.10,之前移植的驱动没办法用,虽然修改后可以编译过,但是加载会崩溃掉;然后就看到很多rtw88里面有8821cu驱动
,尝试很多版本都是一样的结果;要移植新版本驱动我那种方式工作量太大了;

DHDAXCW, 2023-03-04:
[image]

zhangsd20, 2023-03-06:
谢谢,我已经重新移植了一份rtl8821cu驱动到openwrt的backport驱动里面,ap/sta都可以工作

WYC-2020, 2023-03-06:
这个 linux 6.2 已经在适配这个了，https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=aff5ffd718de23cb8603f2e229204670e2644334
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=fba119cee1412f49c472f3913241d00f830761fd
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=7e68dd7d07a28faa2e6574dd6b9dbd90cdeaae91

WYC-2020, 2023-03-06:
说不定那天 l突然内核就变成6.2了 也不是不可能哈哈

zhangsd20, 2023-03-06:
希望如此,每次移植8821cu太麻烦了,从驱动到openwrt一些脚本都需要适配(hostapd/mac80211)

glasscat, 2023-07-27:
> 谢谢,我已经重新移植了一份rtl8821cu驱动到openwrt的backport驱动里面,ap/sta都可以工作
能不能教教怎么移植的？

zhangsd20, 2023-07-31:
你在github找一个rtl8821cu的源码,尽量找kernel版本与你使用的接近的.成功率高很多,跨度太大会有很多数据结构不一样,
编译出来很多错误,即使编译成功驱动起来还容易崩溃;
默认的源码驱动是基于kernel的,需要重新移植到backport里面,无非就是一些makefile和头文件的处理,再就是很多宏定义要处理;
我之前移植是因为比较急所以用的比较笨的方法,基本需要的宏定义都全部重新定义了一遍,这样工作量大但是容易操作,不是最好的方式,
很多可以放在makefile里面.

closed 2026-09-05 by coolsnowwolf