coolsnowwolf/lede #10958: lede6.1编译出错

caixin1218, 2023-03-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：make[4]: [Makefile:1551: libinstall] Error 1 (ignored)
PYTHONPATH=/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/ipkg-install/usr/lib/python3.10  \
	_PYTHON_PROJECT_BASE=/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7 _PYTHON_HOST_PLATFORM=linux-x86_64 PYTHONPATH=/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/build/lib.linux-x86_64-3.10:./Lib _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata python3.10 -Wi -OO /home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/ipkg-install/usr/lib/python3.10/compileall.py \
	-j0 -d /usr/lib/python3.10 -f \
	-x 'bad_coding|badsyntax|site-packages|lib2to3/tests/data' \
	/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/ipkg-install/usr/lib/python3.10
Traceback (most recent call last):
  File "/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/ipkg-install/usr/lib/python3.10/compileall.py", line 17, in <module>
    import struct
  File "/home/caixin/lede6.1/build_dir/target-x86_64_musl/Python-3.10.7/ipkg-install/usr/lib/python3.10/struct.py", line 13, in <module>
    from _struct import *
ImportError: /lib/x86_64-linux-gnu/libc.so: invalid ELF header

caixin1218, 2023-03-03:
不知道是哪里问题，环境是debian 11

yanxin152133, 2023-03-05:
文件与运行环境不匹配

closed 2023-03-16 by aiamadeus