coolsnowwolf/lede #10962: 使用lede源码编译AX6的固件出错，用的是github云编译，求大佬看看

YuxiCN, 2023-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
Github云编译出错，可以在我的存储库里看到，编译了5-6次一直错误

### (2) 路由器型号和固件版本
AX6，内核5.15

### (3) 详细日志
Configuring uhttpd.
 * check_data_file_clashes: Package firewall4 wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/hotplug.d/iface/20-firewall
	But that file is already provided by package  * firewall
 * check_data_file_clashes: Package firewall4 wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/init.d/firewall
	But that file is already provided by package  * firewall
 * check_data_file_clashes: Package firewall4 wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/sbin/fw3
	But that file is already provided by package  * firewall
 * opkg_install_cmd: Cannot install package firewall4.
make[2]: *** [package/Makefile:70: package/install] Error 255
Configuring uhttpd-mod-ubus.
Configuring rpcd-mod-rpcsys.
Configuring attendedsysupgrade-common.
Configuring luci-app-attendedsysupgrade.
Configuring libcares.
Configuring node.
Configuring ipq-wifi-redmi_ax6.
Configuring nftables-json.
Configuring ca-certificates.
Configuring kmod-nft-offload.
Configuring luci-app-ramfree.
Configuring luci-i18n-ramfree-zh-cn.
Configuring wol.
Configuring luci-app-wol.
Configuring luci-i18n-wol-zh-cn.
Configuring rpcd-mod-rrdns.
Configuring luci.
Configuring default-settings.
Configuring luci-i18n-ttyd-zh-cn.
Configuring luci-i18n-base-zh-cn.
Configuring kmod-ath11k-ahb.
Configuring luci-app-pushbot.
Configuring kmod-lib-lzo.
Configuring wget-nossl.
Configuring ip6tables-extra.
Configuring rpcd-mod-iwinfo.
Configuring 6rd.
Configuring smartdns.
Configuring iperf3.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring 6to4.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring coreutils-nohup.
Configuring dnsmasq-full.
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

A：
[error_build_oldax6.txt](https://github.com/coolsnowwolf/lede/files/10887786/error_build_oldax6.txt)

firkerword, 2023-03-04:
firewall和firewall4冲突

YuxiCN, 2023-03-05:
> firewall和firewall4冲突

谢谢你的帮助，在修改后我已经成功编译了

closed 2023-03-05 by aiamadeus

catfishuang, 2023-04-05:
> > firewall和firewall4冲突
> 
> 谢谢你的帮助，在修改后我已经成功编译了

肿么修改，.config里吗