coolsnowwolf/lede #10967: github云编译X86固件出错

Wyatt323, 2023-03-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
之前编译都没问题的，昨天看到作者更新了，手动编译就出错了

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A： x86

### (3) 详细日志

A：
time: package/feeds/packages/sqlite3/compile#64.90#1.81#63.13
[1592](https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1593)
make[2]: Leaving directory '/workdir/openwrt'
[1593](https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1594)
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[1594](https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1595)
make[1]: Leaving directory '/workdir/openwrt'
[1595](https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1596)
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
[1596](https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1597)
Error: Process completed with exit code 2.

Wyatt323, 2023-03-05:
抱歉，我以为这部分就是报错内容了
action链接
https://github.com/Wyatt323/OPENWRT_x86_64_Action/actions/runs/4331876694/jobs/7563997436#step:11:1597

Wyatt323, 2023-03-05:
感谢大佬，我去删了这个插件再编译试试

WYC-2020, 2023-03-06:
确实跑路了，可以找到以前的clone一份 然后修改下源 或者直接用go版本，至于luci的话 看下是否匹配即可，我找到一份已经上次到我仓库，需要的话自己去克隆哈
[image]
像这样改即可
如果需要压缩那就加上下面这个
[image]

WYC-2020, 2023-03-06:
hash 自己算下，我忘记更新了，跳过也没啥

closed 2023-03-11 by aiamadeus