coolsnowwolf/lede #10971: openwrt从5.15降到5.4编译失败，希望大佬指点迷津

JackyLeeGM, 2023-03-06:
反馈问题：openwrt从5.15降到5.4编译失败，希望大佬指点迷津

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
nanopi r2s想用下L大5.4内核编译一下系统，改了lede/target/linux/rockchip/makefile中的KERNEL _PATCHVER：=5.4
KERNEL TESTING PATCHVER：=5.15，然后编译失败

A：

### (2) 路由器型号和固件版本
NanoPi R2S，固件版本5.4

A：

### (3) 详细日志
make[8]：*** No rule to make target 'drivers/net/phy/motorcomm.o'， needed by 'drivers/net/phy/built-in.a'. Stop.
make[7]：*** [scripts/Makefile.build：497：drivers/net/phy] Error 2
make[6]：*** [scripts/Makefile.build：497：drivers/net] Error 2
make[5]：*** [Makefile：1739：drivers] Error 2
make[5]：Leaving directory '/home/jackylee/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.234'
nake[4]：*** [Makefile：26：/home/jackylee/lede/build_dir/target-arch64_generic_musl/linux-rockchip_arnv8/linux-5.4.234/。modules] Error 2
make[4]：Leaving directory '/home/jackylee/lede/target/linux/rockchip'
make[3]：*** [Makefile：11：compile] Error 2
make[3]：Leaving directory '/home/jackylee/lede/target/linux'
time:target/linux/compile#276.94#28.31#308.91
ERROR:target/linux failed to build.
make[2]：*** [target/Makefile：30：target/linux/compile] Error 1
make[2]：Leaving directory '/home/jackylee/lede
nake[1]：***[target/Makefile：23：/home/jackylee/lede/staging_dir/target-aarch64_generic_musl/stamp/。target_compile] Error 2
make[1]：Leaving directory '/home/jackylee/lede'
make：** [/home/jackylee/lede/include/toplevel.mk：231：world]错误 2

A：

DHDAXCW, 2023-03-07:
No rule to make target 'drivers/net/phy/motorcomm.o'

JackyLeeGM, 2023-03-08:
> `No rule to make target 'drivers/net/phy/motorcomm.o'`
> 
> 5.4少了2个文件~ drivers/net/phy/motorcomm.c | 346 ++++++++++++++++++++++++++++++++++ include/linux/motorcomm_phy.h | 68 +++++++

骷髅头大佬🐂       0.0
我试试

coolsnowwolf, 2026-09-05:
Rockchip 当前目标已不再支持通过修改 Makefile 回退到 5.4；该版本文件集中没有 `motorcomm.c`，因此出现 `No rule to make target` 是内核版本文件集不匹配，不是当前目标的缺陷。

closed 2026-09-05 by coolsnowwolf