coolsnowwolf/lede #10972: acrh17为什么勾选了baidupcs就出错？

yuzhe1129, 2023-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
ACRH17， 一勾选了baidupcs-web就出错，是不是这个插件不兼容这个型号？

### (3) 详细日志
SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-slhc/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-slhc_installed
echo "kmod-slhc" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-ppp/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-ppp_installed
echo "kmod-ppp" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-pppox/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-pppox_installed
echo "kmod-pppox" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-pppoe/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-pppoe_installed
echo "kmod-pppoe" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-mppe/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-mppe_installed
echo "kmod-mppe" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-tcp-bbr/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-tcp-bbr_installed
echo "kmod-tcp-bbr" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-nls-base/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-nls-base_installed
echo "kmod-nls-base" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-nls-cp437/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-nls-cp437_installed
echo "kmod-nls-cp437" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-nls-iso8859-1/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-nls-iso8859-1_installed
echo "kmod-nls-iso8859-1" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-nls-utf8/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-nls-utf8_installed
echo "kmod-nls-utf8" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-usb-core/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-usb-core_installed
echo "kmod-usb-core" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
[... 26 lines cut ...]

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-usb-storage-extras/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-usb-storage-extras_installed
echo "kmod-usb-storage-extras" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-usb-storage-uas/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-usb-storage-uas_installed
echo "kmod-usb-storage-uas" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/c/lede/tmp/.root-copy.flock -c 'cp -fpR /home/c/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kmod-usb3/. /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kmod-usb3_installed
echo "kmod-usb3" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
mkdir -p /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

[... cut ...]
echo "kmod-usb-xhci-hcd" >> /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
make[3]: Leaving directory '/home/c/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.53#0.90#4.68
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [package/Makefile:110: /home/c/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2023-03-07:
自己看另外的帖子又说

WYC-2020, 2023-03-07:
https://github.com/coolsnowwolf/lede/issues/10967

closed 2023-03-07 by aiamadeus