coolsnowwolf/lede #10973: x86云编译内核6.1固件出错，求大佬指点

YuxiCN, 2023-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译出错

### (2) 路由器型号和固件版本
X86
[error-x86.txt](https://github.com/coolsnowwolf/lede/files/10905740/error-x86.txt)
存在error的信息：
2023-03-07T04:22:05.7899610Z     ERROR: target/linux failed to build.
2023-03-07T04:22:08.0155603Z make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2023-03-07T04:22:08.0194478Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
2023-03-07T04:55:36.9739734Z Write failed because No space left on device
2023-03-07T04:55:36.9739948Z 
2023-03-07T04:55:36.9740085Z FATAL ERROR:Failed to write to output filesystem
2023-03-07T04:55:36.9917845Z make[5]: *** [/workdir/openwrt/include/image.mk:328: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs] Error 1
2023-03-07T04:55:36.9918485Z [=======|                                                      ]  700/6027  11%make[5]: Leaving directory '/workdir/openwrt/target/linux/x86/image'
2023-03-07T04:55:36.9919591Z make[4]: *** [Makefile:27: install] Error 2
2023-03-07T04:55:36.9919961Z make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
2023-03-07T04:55:36.9990222Z make[3]: *** [Makefile:11: install] Error 2
2023-03-07T04:55:36.9990539Z make[2]: *** [target/Makefile:30: target/linux/install] Error 1
2023-03-07T04:55:36.9991060Z make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
2023-03-07T04:55:36.9991440Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2023-03-07T04:55:36.9991821Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2023-03-07T04:55:36.9992090Z time: target/linux/install#44.57#4.23#40.78
2023-03-07T04:55:36.9992336Z     ERROR: target/linux failed to build.
2023-03-07T04:55:36.9992653Z make[2]: Leaving directory '/workdir/openwrt'
2023-03-07T04:55:36.9992970Z make[1]: Leaving directory '/workdir/openwrt'
2023-03-07T04:55:37.0129592Z ##[error]Process completed with exit code 2.
action是我仓库里的x86-build，求大佬指点一下错误

aaronchin, 2023-03-07:
Write failed because No space left on device.........

firkerword, 2023-03-07:
今天云编译抽风了

WYC-2020, 2023-03-07:
明显提示

YuxiCN, 2023-03-07:
> 明显提示

我知道是Write failed because No space left on device这个错误，但是云编译不都是每开一次action清空上一次的临时文件再编译吗

WYC-2020, 2023-03-08:
这是因为云上的ubuntu 本身自带了很多库，我这边在测试一个脚本，直接能释放出百分之四十的空间，等我测试看编译又没问题，在告诉你

firkerword, 2023-03-08:
> 这是因为云上的ubuntu 本身自带了很多库，我这边在测试一个脚本，直接能释放出百分之四十的空间，等我测试看编译又没问题，在告诉你

我是把docker镜像给删了释放了大概3G左右 然后编译完还剩了3G多

WYC-2020, 2023-03-08:
一开始占用情况
[image]
清理完后
[image]
编译完成后
[image]
这才是王道

firkerword, 2023-03-08:
> 一开始占用情况 [image] 清理完后 [image] 编译完成后 [image] 这才是王道

你这占用咋这么多  我安装完依赖也就43G

YuxiCN, 2023-03-08:
现在云编译又没问题了，奇怪了

WYC-2020, 2023-03-08:
> > 一开始占用情况 [image] 清理完后 [image] 编译完成后 [image] 这才是王道
> 
> 你这占用咋这么多 我安装完依赖也就43G

不清楚，我云编译，用的20.4好像，反正系统一下来就这么多

WYC-2020, 2023-03-08:
至于编译后的话可能和我编译的库多少有关系，反正够用了现在，清理后

YuxiCN, 2023-03-09:
> 至于编译后的话可能和我编译的库多少有关系，反正够用了现在，清理后

能否分享一下清理脚本

WYC-2020, 2023-03-12:
> > 至于编译后的话可能和我编译的库多少有关系，反正够用了现在，清理后
> 
> 能否分享一下清理脚本

自己去我公开的仓库Actions-OpenWrt 看