coolsnowwolf/lede #10974: xdr6088 系统日志全是错误

jwkinko, 2023-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：xdr6088 系统日志全是错误，网口也错乱，有时候插拔后就连不上了，2.5g口也有问题

### (2) 路由器型号和固件版本

A：TP-Link TL-XDR6088 (CpuMark : 27946.266950 Scores)
固件版本 | OpenWrt R23.3.3 / LuCI Master (git-23.064.40202-5185e74)

### (3) 详细日志

A：Tue Mar  7 12:49:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:49:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:50:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:51:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:52:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:53:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:54:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:55:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:56:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 12:57:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
[... 424 lines cut ...]
Tue Mar  7 14:06:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:07:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:08:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:09:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:29 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:39 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:49 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Tue Mar  7 14:10:59 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns

yanxin152133, 2023-03-07:
[image]

jwkinko, 2023-03-07:
> [image]

谢谢

closed 2023-03-07 by jwkinko

xuexijiaoben, 2024-03-28:
请问，这个终极解决办法是啥呀，谢谢啦

Joey7802, 2025-02-08:
解决了吗？感谢