coolsnowwolf/lede #10976: 编译6.1时，只要选择kmod-iwlwifi 就报错，缺少依赖。

caixin1218, 2023-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：make[4]: Leaving directory '/home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1'
for subdir in .; do realdir=$(readlink -f /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1); grep -F /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1 /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/$subdir/Module.symvers >> /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/Module.symvers.tmp; [ "/home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1" = "$realdir" ] || grep -F $realdir /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/$subdir/Module.symvers >> /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/Module.symvers.tmp; done; sort -u /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/Module.symvers.tmp > /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/Module.symvers; mkdir -p /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/symvers; mv /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/Module.symvers /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/symvers/mac80211.symvers
touch /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/.built
rm -rf /home/caixin/lede6.1/tmp/stage-mac80211
mkdir -p /home/caixin/lede6.1/tmp/stage-mac80211/host /home/caixin/lede6.1/staging_dir/target-x86_64_musl/packages
mkdir -p /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211 /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211-backport /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211/ath /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/net/mac80211
cp -fpR /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/net/mac80211/*.h /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/include/* /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211/
cp -fpR /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/backport-include/* /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211-backport/
cp -fpR /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/net/mac80211/rate.h /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/net/mac80211/
cp -fpR /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/drivers/net/wireless/ath/*.h /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211/ath/
rm -f /home/caixin/lede6.1/tmp/stage-mac80211/usr/include/mac80211-backport/linux/module.h
find /home/caixin/lede6.1/tmp/stage-mac80211 -name '*.la' | xargs -r rm -f; 
if [ -f /home/caixin/lede6.1/staging_dir/target-x86_64_musl/packages/mac80211.regular.list ]; then /home/caixin/lede6.1/scripts/clean-package.sh "/home/caixin/lede6.1/staging_dir/target-x86_64_musl/packages/mac80211.regular.list" "/home/caixin/lede6.1/staging_dir/target-x86_64_musl"; fi
if [ -d /home/caixin/lede6.1/tmp/stage-mac80211 ]; then (cd /home/caixin/lede6.1/tmp/stage-mac80211; find ./ > /home/caixin/lede6.1/tmp/stage-mac80211.files); 	SHELL= flock /home/caixin/lede6.1/tmp/.staging-dir.flock -c ' mv /home/caixin/lede6.1/tmp/stage-mac80211.files /home/caixin/lede6.1/staging_dir/target-x86_64_musl/packages/mac80211.regular.list && cp -fpR /home/caixin/lede6.1/tmp/stage-mac80211/* /home/caixin/lede6.1/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/caixin/lede6.1/tmp/stage-mac80211
touch /home/caixin/lede6.1/staging_dir/target-x86_64_musl/stamp/.mac80211.regular_installed
removed '/home/caixin/lede6.1/bin/targets/x86/64/packages/kmod-cfg80211_6.1.15+5.15.33-1-2_x86_64.ipk'
mkdir -p /home/caixin/lede6.1/bin/targets/x86/64/packages /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/CONTROL /home/caixin/lede6.1/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/wifi /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/netifd/wireless
install -m0644 ./files/lib/wifi/mac80211.sh /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/wifi
install -m0755 ./files/lib/netifd/wireless/mac80211.sh /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/netifd/wireless
install -d -m0755 /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/etc/hotplug.d/ieee80211
install -m0644 ./files/mac80211.hotplug /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/etc/hotplug.d/ieee80211/10-wifi-detect
find /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/caixin/lede6.1/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/caixin/lede6.1/scripts/strip-kmod.sh" PATCHELF="/home/caixin/lede6.1/staging_dir/host/bin/patchelf" /home/caixin/lede6.1/scripts/rstrip.sh /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211
rstrip.sh: /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/modules/6.1.15/compat.ko: relocatable
rstrip.sh: /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/lib/modules/6.1.15/cfg80211.ko: relocatable
(cd /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/caixin/lede6.1/bin/targets/x86/64/packages
/home/caixin/lede6.1/staging_dir/host/bin/fakeroot /home/caixin/lede6.1/staging_dir/host/bin/bash /home/caixin/lede6.1/scripts/ipkg-build -m "" /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211 /home/caixin/lede6.1/bin/targets/x86/64/packages
Packaged contents of /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-cfg80211 into /home/caixin/lede6.1/bin/targets/x86/64/packages/kmod-cfg80211_6.1.15+5.15.33-1-2_x86_64.ipk
removed '/home/caixin/lede6.1/bin/targets/x86/64/packages/kmod-mac80211_6.1.15+5.15.33-1-2_x86_64.ipk'
mkdir -p /home/caixin/lede6.1/bin/targets/x86/64/packages /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/caixin/lede6.1/staging_dir/target-x86_64_musl/pkginfo
find /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-mac80211 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/caixin/lede6.1/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/caixin/lede6.1/scripts/strip-kmod.sh" PATCHELF="/home/caixin/lede6.1/staging_dir/host/bin/patchelf" /home/caixin/lede6.1/scripts/rstrip.sh /home/caixin/lede6.1/build_dir/target-x86_64_musl/linux-x86_64/backports-5.15.33-1/ipkg-x86_64/kmod-mac80211
[... cut ...]
time: package/kernel/mac80211/regular/compile#398.07#26.25#36.57
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/caixin/lede6.1'
make[1]: *** [package/Makefile:110: /home/caixin/lede6.1/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caixin/lede6.1'
make: *** [/home/caixin/lede6.1/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-03-08:
因为l大修补不完全
[image]

coolsnowwolf, 2026-09-05:
当前 `package/kernel/mac80211/intel.mk` 的 `kmod-iwlwifi` 已声明 `+kmod-mac80211` 及所需 11ac/11ax/11be 依赖，且历史提交已修复旧版本缺失依赖；该 issue 使用 2023 年 backports 构建，已不适用于当前源码。

closed 2026-09-05 by coolsnowwolf