coolsnowwolf/lede #10978: wwan   quectel_srpd_pcie编译失败，具体日志看图

Yofo-me, 2023-03-08:
[image]

WYC-2020, 2023-03-08:
怕是没匹配好吧，等我有时间看下

WYC-2020, 2023-03-08:
32位没看，应该会出错也可能不出错

Yofo-me, 2023-03-08:
好的，我试试

Yofo-me, 2023-03-08:
请问这个patch怎么用啊？直接复制到`package/wwan/driver/quectel_SRPD_PCIE/src/`下面吗？还是复制到`target/linux/x86/patches-5.15/`下面

Yofo-me, 2023-03-08:
[image]

之前还能编译出来固件（没有勾选quectel srpd pcie），我直接下载你的`https://github.com/coolsnowwolf/lede/files/10917655/5.15-6.1-872-export-some-functions-of-the-sched-module.patch `，放到`target/linux/x86/patches-5.15/`下面，再次编译固件就出现上面图片中的错误了

Yofo-me, 2023-03-08:
> [image] [5.15-6.1-872-export-some-functions-of-the-sched-module.patch](https://github.com/coolsnowwolf/lede/files/10917655/5.15-6.1-872-export-some-functions-of-the-sched-module.patch) [5.4-872-export-some-functions-of-the-sched-module.patch](https://github.com/coolsnowwolf/lede/files/10917657/5.4-872-export-some-functions-of-the-sched-module.patch) [image] [image] [image] 如此修复即可

之前还能编译出来固件（没有勾选quectel srpd pcie），我直接下载你的https://github.com/coolsnowwolf/lede/files/10917655/5.15-6.1-872-export-some-functions-of-the-sched-module.patch ，放到target/linux/x86/patches-5.15/下面，再次编译固件就出现上面图片中的错误了

[image]

把`https://github.com/coolsnowwolf/lede/files/10917655/5.15-6.1-872-export-some-functions-of-the-sched-module.patch`放到
`target/linux/x86/patches-6.1/`或者把`https://github.com/coolsnowwolf/lede/files/10917657/5.4-872-export-some-functions-of-the-sched-module.patch`放到`target/linux/x86/patches-5.4/`下还是不行。

YangCee, 2023-08-01:
我也遇到同样的问题了   @Yofo-me 老哥解决没 我看这个issue 还没关闭
[image]

kai0376, 2023-09-19:
一样，编译失败。修改三个文件也一样失败。

hzjnet, 2023-09-25:
问题确实还是没解决

closed 2026-09-05 by coolsnowwolf