coolsnowwolf/lede #10980: OrangePi R1 Plus LTS无法使用WAN口

BaichenXianchen, 2023-03-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：OrangePi R1 Plus LTS无法使用WAN口，貌似是YT8531C无法驱动

### (2) 路由器型号和固件版本

A：OrangePi R1 Plus LTS  6.1/5.15

### (3) 详细日志

A：

QYKing233, 2023-03-11:
刚才移植orangepi-linux里的官方驱动本地编译的内核6.1.5版本的固件双网卡正常工作，但是我不会打补丁，你可以看看官方提交的yt8531C支持，自己改下。或者你编译5.10内核的，lede里的patch是能打上的。

QYKing233, 2023-03-11:
[image]

BaichenXianchen, 2023-03-12:
那我先试一下

QYKing233, 2023-03-12:
> 那我先试一下

我的仓库里有编译好的固件

BaichenXianchen, 2023-03-12:
谢谢了

Gabrielxzx, 2023-03-21:
你好，问下现在问题解决了吗？我的编译后WAN口也无法使用

QYKing233, 2023-03-21:
> 你好，问下现在问题解决了吗？我的编译后WAN口也无法使用

我仓库有你可以下载用。

BaichenXianchen, 2023-03-23:
> > 那我先试一下
> 
> 我的库房里有编译好的固件

是否能给我一份添加修改YT8531 patch的教程，我需要自己编译固件

BaichenXianchen, 2023-04-06:
@coolsnowwolf YT8531C网卡有什么好的解决方案

BaichenXianchen, 2023-04-08:
问题已经解决

closed 2023-04-08 by BaichenXianchen