coolsnowwolf/lede #10982: 内核报错UBIFS error，概览界面bad gateway，重启砖

sunyink, 2023-03-09:
抱歉，自己排查了好久，实在无能为力。

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

#### 概况
自编译系统，克隆了LEDE仓库，feed了kenzo。

#### 主观记录
使用一段时间发现内核报错UBIFS error：`[   80.251158] UBIFS error (ubi0:1 pid 8476): ubifs_decompress: cannot decompress 512 bytes, compressor zstd, error -22
`同时伴随，登陆后的`[概览]`这个界面会白屏bad gateway，但从浏览器历史记录里，访问路由gui的其他地址可以直接进去。

第一次发现后，重启，路由变砖。尝试重编译减插件，刷入，开始几天使用正常无报错，当发现报错后，界面内升级模式刷固件（√保留配置），依旧如上；尝试重置设置，并导入未发生报错时的配置备份，内核记录依旧报错，但`[概览]`界面正常了，也能上网，记录里一直换着Pid刷报错。一共编译3次，都一样问题，最后一次留意了，开始报错是发生在一个pc设备初次尝试连接路由wifi后发生的。

#### 插件情况
feed了kenzo，*了smartdns、openclash并配置使用，M其他个别但均未实装，其余Lede自带。

#### 配置详细
当初把整个编译环境备份到github了，附带.config、.feed、/bin等，在：
`https://github.com/sunyink/Openwrt_lede-for-Netgear_R7800`分支v3

### (2) 路由器型号和固件版本

硬件路由器：netgear R7800
OS版本：OpenWrt R23.2.14 / LuCI Master (git-23.061.31098-da92698)

### (3) 详细日志

#### 内核日志
从`[   60.453898]`开始不对了
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.233 (sys@sys-ThinkPad) (gcc version 11.3.0 (OpenWrt GCC 11.3.0 r5783-be949a6fe)) #0 SMP Fri Mar 3 01:55:17 2023
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30348 r8192 d22900 u61440
[    0.000000] pcpu-alloc: s30348 r8192 d22900 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 474824K/491520K available (6450K kernel code, 198K rwdata, 1556K rodata, 1024K init, 231K bss, 16696K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000021] Switching to timer-based delay loop, resolution 160ns
[    0.000319] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000341] pid_max: default: 32768 minimum: 301
[    0.000528] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000545] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001588] CPU: Testing write buffer coherency: ok
[... 706 lines cut ...]
[  234.722460] 	00000200: 61 91 64 f0 b5 8b 0a 37 b1 98 b0 4e 2c 26 ca 1c f0 33 8e 70 57 a8 9c 95 c9 43 04 a4 ce af f9 2d
[  234.733146] 	00000220: 47 48 2e 22 5c 8e 95 17 16 05 45 b1 19 02 07 11 28 46 9e f4 3b 4a a0 02 08 fa 28 84 20 25 50 48
[  234.743800] 	00000240: 84 63 c7 bb 66 76 04 5e 2e d5 cc 8e 44 3c 02 2f 17 44 29 e5 6c 2b ee 1b af a8 df 71 0e 22 df ec
[  234.754484] 	00000260: 12 52 da 05 90 7f 0c 94 9a 14 66 30 ca cd 61 8f 8b 29 45 51 ea dc 50 db 70 a4 01 33 20 c5 a0 d3
[  234.765167] 	00000280: ef 9c c1 f7 20 20 80 ba 30 29 d4 05 03 a5 14 6a 2f 87 42 3d b1 03 48 33 2d f8 39 f2 8b f8 cd ca
[  234.775852] 	000002a0: 4b ee d5 22 0d 83 0c c7 d1 03 ad 34 f3 18 c9 31 72 cd ec c8 a3 42 1c 61 46 23 0d a8 6b 66 6b 8c
[  234.786505] 	000002c0: ab bd 1c 2f 1d 90 04 3c 42 8a 41 8e 97 7d e7 9b 2f c7 80 d0 a8 51 25 53 c8 d4 88 40 00 20 0c 03
[  234.797186] 	000002e0: 08 30 42 80 98 2b da 0e 12 60 a0 51 96 04 29 8c 31 46 31 46 1c 00 60 8c 08 1a 11 29 41 08 f0 c0
[  234.807871] 	00000300: 5b 81 93 10 9e 79 61 80 87 47 80 c9 3d 95 a3 cd d0 03 94 e6 68 d5 a8 89 b0 1b 88 5c 29 c8 e1 30
[  234.818555] 	00000320: cb 24 8c 18 38 7d e9 13 ce 6e c6 98 27 a8 c4 9c e3 57 02 c1 f2 ce 0b e6 81 fc c8 ab fe cb 2b 97
[  234.829209] 	00000340: e2 72 03 f9 ce a7 f4 d8 52 a5 2b 87 be 5d 26 0a 67 17 68 40 65 96 7e 25 51 ed 64 a0 04 42 cc d9
[  234.839892] 	00000360: 36 4f 04 9f 7c 80 8a 12 a0 5e b1 cc 85 d2 c9 50 11 c1 24 a0 50 2f c5 50 fe 94 92 01 4d 66 e4 2d
[  234.850576] 	00000380: 19 2b 64 c7 bb 17 59 f2 2b 76 03 7c 84 fb f7 be b5 fb 80 fd 85 6e 73 e9 a0 24 0a 6c 3a 9b 21 56
[  234.861261] 	000003a0: 88 e4 81 79 d9 d5 bb 61 3e 24 44 a5 20 ec 65 10 04 76 0f 38 15 a9 5d 8c a3 82 7d 0f ba b7 f6 4a
[  234.871947] 	000003c0: e0 c2 35 33 41 33 3f 64 8c 99 39 bd 24 f7 fc f9 7d e4 82 10 9a 19 5c 9c ca c0 89 20 c1 04 2d f7
[  234.882599] 	000003e0: e8 7c 5c 0b 1d 86 9a 4d d7 7e ff 02 9c 8b df 48 e0 ce a4 45 51 d4 57 ac 0c 7a c2 3c dd 33 73 1b
[  234.893281] 	00000400: ac 05 98 ae 0c 74 b5 b3 fc ff 0b 80 40 1f fa 74 20 bf 8f 9b c3 cb de 55 9a 79 0c 1c 04 59 ca 4b
[  234.903966] 	00000420: 1a 46 ca 66 34 3b 23 33 94 b0 8f 63 13 33 3a 40 d0 4e 94 52 91 3d 85 9f 94 6f 04 2e 13 26 fd e4
[  234.914657] 	00000440: 65 9c 9c 5e 8e 82 d8 e6 62 a8 cf 17 ee 51 4c e3 4b 24 8a 6c ea 11 74 7d 6d 17 b4 ae fa da 6f 11
[  234.925337] 	00000460: 73 cf 30 98 4a d3 66 74 c3 13 99 b1 5d 07 1f 4c de 3d f0 96 12 3d 8b b1 71 80 07 38 3d d2 a8 06
[  234.935991] 	00000480: 72 10 8a 66 31 aa 93 23 85 d5 da a9 e8 42 08 df 3e ef c6 53 40 6f 15 85 8c b4 74 d4 cc 65 b2 af
[  234.946675] 	000004a0: 19 0a 67 d7 f1 2e 8c 57 b2 de 8e 95 3d 9a 5b 5f cb 02 27 6a 01 ab 9e 6b c0 4e 83 ee 90 49 ec 23
[  234.957359] 	000004c0: db e4 48 06 6f d3 37 a7 2f 18 e7 dd b2 9f b3 53 d0 04
[  234.968058] UBIFS error (ubi0:1 pid 9381): do_readpage: cannot read page 2 of inode 149, error -22
```

KevinMX, 2023-03-11:
硬件/闪存故障可能？

minlang112, 2023-03-11:
闪存物理故障可能性较大

sunyink, 2023-03-20:
还在排查。我发现openclash的内核报错也会计入 系统的内核报错。现在换了老内核，持续观察一段时间。clash真的是反人类。

pk49800, 2023-06-23:
同R7800，以前刷某些固件也出现过这种错误，现在的版本，只保留了samba4和helloworld，开机持续1个月没有任何报错日志，版本是：OpenWrt R23.5.1 / LuCI Master (git-23.141.16773-28dd4b3)

closed 2026-09-05 by coolsnowwolf