coolsnowwolf/lede #10988: samba4.14访问速度慢

cuishitian77, 2023-03-11:
1.关于你要提交的问题
samba4.14访问速度慢

2. 详细叙述
windows10访问速度慢。速度只有350k/s。更换恩山论坛别人编译的固件最多速度只有30m/s。但是openwrt原版的samba4.17.5速度没有问题。内核更换6.1跟5.15都一样。

路由器型号和固件版本
3-12更新最新版本
J1900-OpenWrt R22.12.1

详细日志
没有找到关联的日志文件

WYC-2020, 2023-03-12:
那直接吧原版的samba4 文件夹整体替换了 ，然后在编译看下 是否正常了，正常了那就是samba4问题了 升级一下即可

cyclc77, 2023-03-12:
> 

已经替换过了，lede里带的是samba4.14。我用openwrt原版里的samba4.17.5直接替换，速度依旧。现在用的原版openwrt。没有问题。我也十分的纳闷。是我个人的原因吗？除了5.4的内核，内核换遍了。对了，手动勾选samba3.6根samba4两个编译出来的速度都差不多。

WYC-2020, 2023-03-12:
我这边用x86测试了一下正常的 5.4内核

cyclc77, 2023-03-12:
> 我这边用x86测试了一下正常的 5.4内核

好的，我有空试试用5.4内核编译一下，行嘛，最早auto samba选项的版本我记得是3.多的版本，那个版本用着一直没有问题，最新的samba3-server版本都升级到4.0了，就出现这个问题了。大佬也可以试试，因为openwrt官方内核5.15的samba4.17.5没有问题

WYC-2020, 2023-03-13:
> > 我这边用x86测试了一下正常的 5.4内核
> 
> 好的，我有空试试用5.4内核编译一下，行嘛，最早auto samba选项的版本我记得是3.多的版本，那个版本用着一直没有问题，最新的samba3-server版本都升级到4.0了，就出现这个问题了。大佬也可以试试，因为openwrt官方内核5.15的samba4.17.5没有问题

那是因为 这个切换到4了啊20550724175da39ef2909b504226d4495c756456 这个提交自己看吧，其他内核我记得x86 也是正常的，你安装虚拟机测试看吧

cyclc77, 2023-03-13:
> 

刚刚重新编译了5.4内核和6.1内核的最新版本，发现在VMware虚拟机中，samba都没有问题。都能达到106M/s。
我现在试试在物理机里安装，看看有没有问题。

cyclc77, 2023-03-13:
> 那是因为 这个切换到4了啊[20550724175da39ef2909b504226d4495c756456](https://github.com/coolsnowwolf/lede/commit/20550724175da39ef2909b504226d4495c756456) 这个提交自己看吧，其他内核我记得x86 也是正常的，你安装虚拟机测试看吧

刚刚重新编译了5.4内核和6.1内核的最新版本，发现在VMware虚拟机中，samba都没有问题。都能达到106M/s。
我现在试试在物理机里安装，看看有没有问题。

WYC-2020, 2023-03-13:
> 刚刚重新编译了5.4内核和6.1内核的最新版本，发现在VMware虚拟机中，samba都没有问题。都能达到106M/s。 我现在试试在物理机里安装，看看有没有问题。

嗯是的 你测试一下吧，反正我这边 基本没啥问题

cyclc77, 2023-03-13:
> 嗯是的 你测试一下吧，反正我这边 基本没啥问题

找到原因了，我在/etc/config/samba4中添加option allow_legacy_protocols '1'
这个对应着openwrt原版samba4Luci界面里的允许旧协议吧。
windows10的速度就只有350k了。
但是如果不添加这个，盒子就无法访问。
但是原版直接勾选，就没有这个问题。

WYC-2020, 2023-03-13:
> > > > > > 我这边用x86测试了一下正常的 5.4内核
> > > > > 
> > > > > 
> > > > > 好的，我有空试试用5.4内核编译一下，行嘛，最早auto samba选项的版本我记得是3.多的版本，那个版本用着一直没有问题，最新的samba3-server版本都升级到4.0了，就出现这个问题了。大佬也可以试试，因为openwrt官方内核5.15的samba4.17.5没有问题
> > > > 
> > > > 
> > > > 那是因为 这个切换到4了啊[20550724175da39ef2909b504226d4495c756456](https://github.com/coolsnowwolf/lede/commit/20550724175da39ef2909b504226d4495c756456) 这个提交自己看吧，其他内核我记得x86 也是正常的，你安装虚拟机测试看吧
> > > 
> > > 
> > > 刚刚重新编译了5.4内核和6.1内核的最新版本，发现在VMware虚拟机中，samba都没有问题。都能达到106M/s。 我现在试试在物理机里安装，看看有没有问题。
> > 
> > 
> > 嗯是的 你测试一下吧，反正我这边 基本没啥问题
> 
> 找到原因了，我在/etc/config/samba4中添加option allow_legacy_protocols '1' 这个对应着openwrt原版samba4Luci界面里的允许旧协议吧。 windows10的速度就只有350k了。 但是如果不添加这个，盒子就无法访问。 但是原版直接勾选，就没有这个问题。
> 
> LEDE版本- 固件版本 OpenWrt R23.3.3 LuCI Master (git-23.070.53221-d1e7a12) 内核版本 6.1.18 samba4版本 samba4-server 4.14.14

你把原版没问题的 samba4的config 和l这边产生的config 的对比看下 区别是那些，发一份 op 原版没问题的config 给我 我看下 我这边直接用这个配置 是不是也是这个速度

cyclc77, 2023-03-13:
> 你把原版没问题的 samba4的config 和l这边产生的config 的对比看下 区别是那些，发一份 op 原版没问题的config 给我 我看下 我这边直接用这个配置 是不是也是这个速度

/etc/config/samba4

config samba
        option workgroup 'WORKGROUP'
        option charset 'UTF-8'
        option description 'Samba on OpenWRT'
        option interface 'lan'
        option allow_legacy_protocols '1'

config sambashare
        option name 'sdb'
        option path '/mnt/sdb1'
        option read_only 'no'
        option force_root '1'
        option users 'root'
        option guest_ok 'no'
        option inherit_owner 'yes'
        option create_mask '0666'
        option dir_mask '0777

/etc/samba/smb.conf

[global]
	netbios name = |NAME| 
	interfaces = |INTERFACES|
	server string = |DESCRIPTION|
	unix charset = |CHARSET|
	workgroup = |WORKGROUP|

	## This global parameter allows the Samba admin to limit what interfaces on a machine will serve SMB requests.
	bind interfaces only = yes

	## time for inactive connections to-be closed in minutes
	deadtime = 15

	## disable core dumps
	enable core files = no

	## set security (auto, user, domain, ads)
	security = user

	## This parameter controls whether a remote client is allowed or required to use SMB encryption.
	## It has different effects depending on whether the connection uses SMB1 or SMB2 and newer:
    ## If the connection uses SMB1, then this option controls the use of a Samba-specific extension to the SMB protocol introduced in Samba 3.2 that makes use of the Unix extensions.
	## If the connection uses SMB2 or newer, then this option controls the use of the SMB-level encryption that is supported in SMB version 3.0 and above and available in Windows 8 and newer. 
	## (default/auto,desired,required,off)
	#smb encrypt = default

	## set invalid users
	#invalid users = root

	## map unknow users to guest
	map to guest = Bad User

	## allow client access to accounts that have null passwords. 
	null passwords = yes

	## The old plaintext passdb backend. Some Samba features will not work if this passdb backend is used. (NOTE: enabled for size reasons)
	## (tdbsam,smbpasswd,ldapsam)
	passdb backend = smbpasswd

	## Set location of smbpasswd ('smbd -b' will show default compiled location)
	#smb passwd file = /etc/samba/smbpasswd 

	## LAN (IPTOS_LOWDELAY TCP_NODELAY) WAN (IPTOS_THROUGHPUT) WiFi (SO_KEEPALIVE) try&error for buffer sizes (SO_RCVBUF=65536 SO_SNDBUF=65536)
	socket options = IPTOS_LOWDELAY TCP_NODELAY

	## If this integer parameter is set to a non-zero value, Samba will read from files asynchronously when the request size is bigger than this value.
[... 50 lines cut ...]
	## (445 139) Specifies which ports the server should listen on for SMB traffic.
	## 139 is netbios/nmbd
	#smb ports = 445 139

	## This is a list of files and directories that are neither visible nor accessible.
	## Each entry in the list must be separated by a '/', which allows spaces to be included in the entry. '*' and '?' can be used to specify multiple files or directories as in DOS wildcards.
	veto files = /Thumbs.db/.DS_Store/._.DS_Store/.apdisk/

	## If a directory that is to be deleted contains nothing but veto files this deletion will fail unless you also set the delete veto files parameter to yes.
	delete veto files = yes

################ Filesystem and creation rules ################
	## reported filesystem type (NTFS,Samba,FAT)
	#fstype = FAT

	## Allows a user who has write access to the file (by whatever means, including an ACL permission) to modify the permissions (including ACL) on it.
	#dos filemode = Yes

	## file/dir creating rules
	#create mask = 0666
	#directory mask = 0777
	#force group = root
	#force user = root
	#inherit owner = windows and unix
################################################################

cyclc77, 2023-03-13:
> 你把原版没问题的 samba4的config 和l这边产生的config 的对比看下 区别是那些，发一份 op 原版没问题的config 给我 我看下 我这边直接用这个配置 是不是也是这个速度

/etc/config/samba4

	config samba
	        option workgroup 'WORKGROUP'
	        option charset 'UTF-8'
	        option description 'Samba on OpenWRT'
	        option interface 'lan'
	        option allow_legacy_protocols '1'

	config sambashare
	        option name '......'
	        option path '/mnt/sdb1'
	        option read_only 'no'
	        option force_root '1'
	        option users 'root'
	        option guest_ok 'no'
	        option inherit_owner 'yes'
	        option create_mask '0666'
	        option dir_mask '0777'

/etc/samba/smb.conf

	[global]
		netbios name = |NAME| 
		interfaces = |INTERFACES|
		server string = |DESCRIPTION|
		unix charset = |CHARSET|
		workgroup = |WORKGROUP|

		## This global parameter allows the Samba admin to limit what interfaces on a machine will serve SMB requests.
		bind interfaces only = yes

		## time for inactive connections to-be closed in minutes
		deadtime = 15

		## disable core dumps
		enable core files = no

		## set security (auto, user, domain, ads)
		security = user

		## This parameter controls whether a remote client is allowed or required to use SMB encryption.
		## It has different effects depending on whether the connection uses SMB1 or SMB2 and newer:
		## If the connection uses SMB1, then this option controls the use of a Samba-specific extension to the SMB protocol introduced in Samba 3.2 that makes use of the Unix extensions.
		## If the connection uses SMB2 or newer, then this option controls the use of the SMB-level encryption that is supported in SMB version 3.0 and above and available in Windows 8 and newer. 
		## (default/auto,desired,required,off)
		#smb encrypt = default

		## set invalid users
		#invalid users = root

		## map unknow users to guest
		map to guest = Bad User

		## allow client access to accounts that have null passwords. 
		null passwords = yes

		## The old plaintext passdb backend. Some Samba features will not work if this passdb backend is used. (NOTE: enabled for size reasons)
		## (tdbsam,smbpasswd,ldapsam)
		passdb backend = smbpasswd

		## Set location of smbpasswd ('smbd -b' will show default compiled location)
		#smb passwd file = /etc/samba/smbpasswd 

		## LAN (IPTOS_LOWDELAY TCP_NODELAY) WAN (IPTOS_THROUGHPUT) WiFi (SO_KEEPALIVE) try&error for 	buffer sizes (SO_RCVBUF=65536 SO_SNDBUF=65536)
		socket options = IPTOS_LOWDELAY TCP_NODELAY

		## If this integer parameter is set to a non-zero value, Samba will read from files asynchronously when the request size is bigger than this value.
[... 50 lines cut ...]
		## (445 139) Specifies which ports the server should listen on for SMB traffic.
		## 139 is netbios/nmbd
		#smb ports = 445 139

		## This is a list of files and directories that are neither visible nor accessible.
		## Each entry in the list must be separated by a '/', which allows spaces to be included in the entry. '*' and '?' can be used to specify multiple files or directories as in DOS wildcards.
		veto files = /Thumbs.db/.DS_Store/._.DS_Store/.apdisk/

		## If a directory that is to be deleted contains nothing but veto files this deletion will fail unless you also set the delete veto files parameter to yes.
		delete veto files = yes

		########Filesystem and creation rules #########
		## reported filesystem type (NTFS,Samba,FAT)
		#fstype = FAT

		## Allows a user who has write access to the file (by whatever means, including an ACL permission) to modify the permissions (including ACL) on it.
		#dos filemode = Yes

		## file/dir creating rules
		#create mask = 0666
		#directory mask = 0777
		#force group = root
		#force user = root
		#inherit owner = windows and unix
	################################################################

WYC-2020, 2023-03-14:
> allow_legacy_protocols

我这边加了这玩意 感觉也没降速 哈哈

cyclc77, 2023-03-14:
> > allow_legacy_protocols
> 
> 我这边加了这玩意 感觉也没降速 哈哈

好吧，就先这么着吧，我也蒙了，只能归结于，计算机的玄学问题。我先用openwrt原版吧。

cyclc77, 2023-03-14:
> > allow_legacy_protocols
> 
> 我这边加了这玩意 感觉也没降速 哈哈

Tue Mar 14 12:01:58 2023 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Tue Mar 14 12:02:00 2023 user.info trm-ntp  [5749]: get ntp time sync
Tue Mar 14 12:02:00 2023 user.info trm-2.0.3[5850]: travelmate is currently disabled, please set 'trm_enabled' to '1' to use this service
Tue Mar 14 12:02:00 2023 user.info trm-2.0.3[5915]: travelmate instance stopped ::: action: stop, pid: 5850
Tue Mar 14 12:02:04 2023 daemon.info procd: Instance wsdd2::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
是不是这个原因。

WYC-2020, 2023-03-14:
> wsdd2

不清楚，l的wssd2 好像是有问题的，我的修复过的，你用官方的wssd 替换这个  再试下看看

cyclc77, 2023-03-14:
> > wsdd2
> 
> 不清楚，l的wssd2 好像是有问题的，我的修复过的，你用官方的wssd 替换这个 再试下看看

官方packages包里的wssd2版本号是2022-04-25 而lede中的是2021-10-22 是不是这个原因呢。我一会编译看看。然后反馈

cyclc77, 2023-03-14:
> > wsdd2
> 
> 不清楚，l的wssd2 好像是有问题的，我的修复过的，你用官方的wssd 替换这个 再试下看看

并不行，替换官方包的wsdd2系统日志还是出错。速度依旧70m/s。
wsdd2 跟haproxy 系统日志出错。都用了官方包的最新版本。问题依旧。
Tue Mar 14 16:54:44 2023 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Tue Mar 14 16:54:50 2023 daemon.info procd: Instance wsdd2::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
算了，我老老实实用官方包吧。lede这个估计是内核不兼容？我也不清楚，没能力解决了。

WYC-2020, 2023-03-14:
> 并不行，替换官方包的wsdd2系统日志还是出错。速度依旧70m/s。 wsdd2 跟haproxy 系统日志出错。都用了官方包的最新版本。问题依旧。 Tue Mar 14 16:54:44 2023 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last [...]

你是需要x86的吧，要不我发你一个我编译的固件，你试试看哈哈

cyclc77, 2023-03-14:
> 你是需要x86的吧，要不我发你一个我编译的固件，你试试看哈哈

可以啊。你发过来我试试吧。已经头大了。

WYC-2020, 2023-03-14:
> 可以啊。你发过来我试试吧。已经头大了。

https://drive.google.com/drive/folders/1dLZ_yGANrZJvXl3prDKO7R8aIGIuwdiM?usp=share_link

cyclc77, 2023-03-14:
> 

刚看到，有没有vmdk镜像。

cyclc77, 2023-03-14:
> https://drive.google.com/drive/folders/1dLZ_yGANrZJvXl3prDKO7R8aIGIuwdiM?usp=share_link

刚看到，有没有，vmdk镜像，我需要先到虚拟机里试试

WYC-2020, 2023-03-14:
> 刚看到，有没有，vmdk镜像，我需要先到虚拟机里试试

解压用工具转下就可以了，然后如果是esxi8.0  添加现有硬盘后，到最后一步的时候 返回一下 ，然后在继续就可以了，如果不是8.0 正常添加即可

cyclc77, 2023-03-14:
> 解压用工具转下就可以了，然后如果是esxi8.0 添加现有硬盘后，到最后一步的时候 返回一下 ，然后在继续就可以了，如果不是8.0 正常添加即可

你给的固件，在虚拟机里无法链接网络。算了，谢谢了。不弄了，弄了好几天了，也搞不明白。心力憔悴了。

WYC-2020, 2023-03-20:
> 你给的固件，在虚拟机里无法链接网络。算了，谢谢了。不弄了，弄了好几天了，也搞不明白。心力憔悴了。

可怜的孩子，那还是用官方的吧

closed 2023-03-25 by cyclc77

himho, 2023-04-06:
兄弟搞定了没有？我这里更惨 只有11M

cyclc77, 2023-04-06:
> 兄弟搞定了没有？我这里更惨 只有11M

并没有，最近没有试验，用的Openwrt官方的包

lemontealin, 2024-03-22:
我倒是解决了...一半...之前下载30M以内，上传10M，**关掉BBR加速**之后，现在OP的samba下载都正常，100M+的速度咔咔快，但是上传还是过不了10M

Fyred, 2024-10-28:
> 我倒是解决了...一半...之前下载30M以内，上传10M，**关掉BBR加速**之后，现在OP的samba下载都正常，100M+的速度咔咔快，但是上传还是过不了10M

确实BBR加速有问题，我这边是上传满千兆没问题，下载就只有20m，关了BBR立竿见影