coolsnowwolf/lede #10991: 更新feeds有警告，缺少依赖

rhwsoft, 2023-03-12:
Ruan hongwei, [12.03.23 15:26]
ruan@ruan-vm:~/lede$ ./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba4', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/qat/libs/eudev/Makefile' has a dependency on 'libkmod', which does not exist
WARNING: Makefile 'package/qat/libs/eudev/Makefile' has a build dependency on 'gperf/host', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-gobinetmodem/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-gobinetmodem/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/mt/luci-app-mtwifi/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-pcimodem/Makefile' has a dependency on 'pciutils', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-pcimodem/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-spdmodem/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-usbmodem/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/wwan/app/luci-app-usbmodem/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/mt/drivers/mt7615d/Makefile' has a dependency on 'maccalc', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libtins', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libyaml-cpp', which does not exist
WARNING: Makefile 'package/lean/pcat-manager/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/pcat-manager/Makefile' has a dependency on 'libgpiod', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist

Ruan hongwei, [12.03.23 15:26]
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a build dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/wwan/driver/quectel_MHI/Makefile' has a dependency on 'pciids', which does not exist
WARNING: Makefile 'package/wwan/driver/quectel_MHI/Makefile' has a dependency on 'pciutils', which does not exist
WARNING: Makefile 'package/network/services/samba36/Makefile' has a dependency on 'libaio', which does not exist

Ruan hongwei, [12.03.23 15:26]
还是老样子，已经删除lede目录重来了

WYC-2020, 2023-03-12:
你确定你是经常编译固件的？你这个怕是第一次吧第三方库拉下来 然后install 出现的告警吧，建议先去看编译教程

rhwsoft, 2023-03-12:
我不经常编译固件，但是也编译成功过好几次。这次我可是虚拟机重装ubuntu22.04了，我想应该环境干净了，然后按照lean大神教程一步一步下来，到升级feeds这步就出现警告了。

closed 2023-03-13 by aiamadeus