coolsnowwolf/lede #10994: XRAY 1.8.0 编译出错

Travel2Here, 2023-03-13:
lede日常更新多处，xray最近更新至1.8.0，
最近两日拉新后编译，过程发现使用xray的app在编译时会出错。是否平台和工具之间存在衔接问题；
会不会是golang的版本与1.8.0的xray core有冲突？
请指教~

Travel2Here, 2023-03-13:
手动把xray core的makefile退回1.7.5正常编译。。。看来确实是某个别的冲突了

WYC-2020, 2023-03-13:
错误也不贴，你让谁给你算命

Travel2Here, 2023-03-13:
> 

建议你好好说话；
就一句简单的error .V=s的详细内容又太长你会看？
如果有空方便帮人测试一下不是更好？也不是多大的包~

ERROR: package/feeds/xxx/xray-core failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/me/lede/include/toplevel.mk:231: world] Error 1

WYC-2020, 2023-03-13:
置顶帖请看下，明显就说了带日志，我即使不说话 像你这样的issue 直接就会被作者关闭
还有日志问题，请带上-j1 单线程日志，如果你能多线程找到错误信息也行，但是看你这个贴的日志明显就不会
还有别什么都指望别人帮你测试，有时间的自然会帮你测试，不是什么包多大的问题，而是很多人都不是靠这个吃饭，也要工作，即使靠这个吃饭的也不会这个时间来给你答复
至于说话我觉得没怎么，其他那些没日志我也是这么说的，没日志算不出很正常，让你贴下日志还让你觉得我吼你了一样哎，那你等其他人看有人帮你答复没吧

mitsukileung, 2023-03-13:
xray 1.8.0 编译没有问题

Travel2Here, 2023-03-13:
> xray 1.8.0 编译没有问题

兄弟可能是x86？我是rockchip

mitsukileung, 2023-03-13:
x86 ipq807x mt7621 K3 都没有问题

closed 2023-03-13 by Travel2Here

chjping, 2023-03-25:
> 置顶帖请看下，明显就说了带日志，我即使不说话 像你这样的issue 直接就会被作者关闭 还有日志问题，请带上-j1 单线程日志，如果你能多线程找到错误信息也行，但是看你这个贴的日志明显就不会 还有别什么都指望别人帮你测试，有时间的自然会帮你测试，不是什么包多大的问题，而是很多人都不是靠这个吃饭，也要工作，即使靠这个吃饭的也不会这个时间来给你答复 至于说话我觉得没怎么，其他那些没日志我也是这么说的，没日志算不出很正常，让你贴下日志还让你觉得我吼你了一样哎，那你等其他人看有人帮你答复没吧

`make[3]: *** [Makefile:52: /home/Disk/2Tsda/app/lede/build_dir/target-x86_64_musl/Xray-core-1.8.0/.built] Error 1
make[3]: Leaving directory '/home/Disk/2Tsda/app/lede/feeds/helloworld/xray-core'
time: package/feeds/helloworld/xray-core/compile#1.34#0.57#1261.54
    ERROR: package/feeds/helloworld/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/xray-core/compile] Error 1
make[2]: Leaving directory '/home/Disk/2Tsda/app/lede'
make[1]: *** [package/Makefile:110: /home/Disk/2Tsda/app/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/Disk/2Tsda/app/lede'
make: *** [/home/Disk/2Tsda/app/lede/include/toplevel.mk:231: world] Error 2`

shu0802, 2023-03-31:
> 手动把xray core的makefile退回1.7.5正常编译。。。看来确实是某个别的冲突了

怎么回退的？我编译时也出现错误，不知是网络原因还是有冲突。
[image]