coolsnowwolf/lede #10997: lean最新源码编译出错

Yofo-me, 2023-03-14:
lean最新源码编译出错，不管是action还是本地编译都是这个错误，日志如下
```shell
echo "rtl8822ce-firmware" >> /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.77#0.97#2.74
make[3]: Entering directory '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/package/kernel/linux'
mkdir -p /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.98/drivers/bluetooth/btbcm.ko' is missing.
make[3]: *** [modules/other.mk:74: /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/bin/targets/x86/64/packages/kmod-bluetooth_5.15.98-1_x86_64.ipk] Error 1
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[3]: Leaving directory '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#3.68#0.53#4.23
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:110: /home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt/include/toplevel.mk:231: world] Error 2
make[2]: Leaving directory '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt'
make[1]: Leaving directory '/home/code-runner/actions-runner/_work/Action-Openwrt/Action-Openwrt/workdir/openwrt'
Error: Process completed with exit code 2.
```

WYC-2020, 2023-03-14:
更新后 make menuconfig   然后保存，config  是需要更新了，config不是能一直通用的

WYC-2020, 2023-03-14:
你这个看 明显是https://github.com/coolsnowwolf/lede/commit/1ac276866a4f41289a510913e151b3bdf382f907 这个提交新增了东西

closed 2023-03-15 by coolsnowwolf