coolsnowwolf/lede #11001: 编译小米2100出错，是什么原因？

yuzhe1129, 2023-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

小米2100，这是什么错误啊

### (3) 详细日志
/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/ate/mt_mac/mt_testmode.c:2815:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
   for (IdBss = FIRST_MBSSID;
   ^~~
  mipsel-openwrt-linux-musl-gcc -Wp,-MD,/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/common/.cmm_tcprack.o.d  -nostdinc -isystem /home/c/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I./arch/mips/include -I./arch/mips/include/generated  -I./include -I./arch/mips/include/uapi -I./arch/mips/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80001000 -DLINKER_LOAD_ADDRESS=0x80001000 -DDATAOFFSET=0 -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -fno-stack-check -march=mips32r2 -mtune=34kc -Wa,--trap -DTOOLCHAIN_SUPPORTS_VIRT -DTOOLCHAIN_SUPPORTS_XPA -DTOOLCHAIN_SUPPORTS_CRC -DTOOLCHAIN_SUPPORTS_DSP -DTOOLCHAIN_SUPPORTS_GINV -I./arch/mips/include/asm/mach-ralink -I./arch/mips/include/asm/mach-ralink/mt7621 -I./arch/mips/include/asm/mach-generic -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -fmacro-prefix-map=./= -Wno-packed-not-aligned -fmacro-prefix-map=/home/c/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl -fno-caller-saves -I/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/include -I/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/include -I/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/ate/include -I/home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/include/mcu -DCONFIG_SUPPORT_OPENWRT -DCONFIG_RT_FIRST_CARD=7615 -DCONFIG_RT_FIRST_IF_RF_OFFSET=0xc0000 -DCONFIG_RT_FIRST_CARD_EEPROM="\"flash\"" -DCONFIG_FIRST_IF_IPAILNA -DWIFI_SYS_FW_V1 -DWIFI_SYS_FW_V2 -DMULTI_PROFILE -DUAPSD_SUPPORT -DUAPSD_DEBUG -DWSC_INCLUDED -DWSC_SINGLE_TRIGGER -DWSC_AP_SUPPORT -DWSC_V2_SUPPORT -DDOT11_VHT_AC -DDOT11W_PMF_SUPPORT -DTPC_SUPPORT -DBACKGROUND_SCAN_SUPPORT -DMT_DFS_SUPPORT -DCUSTOMISE_RDD_THRESHOLD_SUPPORT -DRDM_FALSE_ALARM_DEBUG_SUPPORT -DONDEMAND_DFS -DIGMP_SNOOP_SUPPORT -DSMART_CARRIER_SENSE_SUPPORT -DMBSS_SUPPORT -DWDS_SUPPORT -DAPCLI_SUPPORT -DMAT_SUPPORT -DMULTI_APCLI_SUPPORT -DAPCLI_AUTO_CONNECT_SUPPORT -DAPCLI_AUTO_BW_TMP -DDHCP_UC_SUPPORT -DFAST_EAPOL_WAR -DBT_APCLI_SUPPORT -DMAC_REPEATER_SUPPORT -DPCIE_ASPM_DYM_CTRL_SUPPORT -DCONFIG_ATE -DCONFIG_QA -DCONFIG_RT2880_ATE_CMD_NEW -DCONFIG_HW_HAL_OFFLOAD -DINTERNAL_CAPTURE_SUPPORT -DWIFI_SPECTRUM_SUPPORT -DRTMP_FLASH_SUPPORT -DPRE_CAL_TRX_SET1_SUPPORT -DRLM_CAL_CACHE_SUPPORT -DPRE_CAL_TRX_SET2_SUPPORT -DMCS_LUT_SUPPORT -DTXBF_SUPPORT -DVHT_TXBF_SUPPORT -DCFG_SUPPORT_MU_MIMO -DCFG_SUPPORT_MU_MIMO_RA -DMCAST_RATE_SPECIFIC -DBAND_STEERING -DVOW_SUPPORT -DRED_SUPPORT -DLINUX_NET_TXQ_SUPPORT -DMT7615 -DMT_BBP -DMT_RF -DRTMP_RF_RW_SUPPORT -DMT_MAC -DRTMP_MAC_PCI -DRTMP_PCI_SUPPORT -DMCS_LUT_SUPPORT -DA_BAND_SUPPORT -DRTMP_EFUSE_SUPPORT -DCAL_FREE_IC_SUPPORT -DBCN_OFFLOAD_SUPPORT -DERR_RECOVERY -DCONFIG_ANDES_SUPPORT -DCUT_THROUGH -DUNIFY_FW_CMD -DNEED_ROM_PATCH -DTXPWRMANUAL -DMAC_INIT_OFFLOAD -DENHANCED_STAT_DISPLAY -DCONFIG_FWOWN_SUPPORT -DNEWSEC -DSINGLE_SKU_V2 -DTX_POWER_CONTROL_SUPPORT -DTHERMAL_PROTECT_SUPPORT -DRESOURCE_PRE_ALLOC -DCONFIG_HW_HAL_OFFLOAD -DRX_SCATTER -DPKT_BUDGET_CTRL_SUPPORT -DRATE_ADAPTION -DRATE_ADAPT_AGBS_SUPPORT -DRACTRL_FW_OFFLOAD_SUPPORT -DCONFIG_TX_DELAY -DRELEASE_EXCLUDE -DPRE_CAL_TRX_SET1_SUPPORT -DRLM_CAL_CACHE_SUPPORT -DPRE_CAL_TRX_SET2_SUPPORT -DCONFIG_PROPRIETARY_DRIVER -DG_BAND_256QAM -DBRCM_256QAM -Wframe-larger-than=4096 -DHDR_TRANS_TX_SUPPORT -DCUT_THROUGH_FULL_OFFLOAD -DHDR_TRANS_RX_SUPPORT -DREDUCE_TCP_ACK_SUPPORT -DLED_CONTROL_SUPPORT -DCONFIG_WIFI_MTD -DLED_CONTROL_SUPPORT -DAGGREGATION_SUPPORT -DPIGGYBACK_SUPPORT -DWMM_SUPPORT -DLINUX -Wstrict-prototypes -Wno-trigraphs -Wall -Wno-unused-but-set-variable -DOLDSEC -DCONFIG_AP_SUPPORT -DSCAN_SUPPORT -DAP_SCAN_SUPPORT -DDOT11_N_SUPPORT -DDOT11N_DRAFT3 -DSTATS_COUNT_SUPPORT -DIAPP_SUPPORT -DDOT1X_SUPPORT -DDBG -DCONFIG_WIFI_PKT_FWD -DMTFWD -DIP_ASSEMBLY -DOUI_CHECK_SUPPORT -DHTC_DECRYPT_IOT -Inet/nat -DMT_WIFI_MODULE  -DMODULE -mno-long-calls  -DKBUILD_BASENAME='"cmm_tcprack"' -DKBUILD_MODNAME='"mt_wifi"' -c -o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/common/cmm_tcprack.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/common/cmm_tcprack.c
{   echo /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/mt_wifi.ko; :; } \
| awk '!x[$0]++' - > /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/modules.order
  mipsel-openwrt-linux-musl-ld  -m elf32ltsmip -z noexecstack --no-warn-rwx-segments    -r -o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/mt_wifi.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_assoc.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_auth.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_mlme.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_sanity.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_sync.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_wpa.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_sec.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_data.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_autoChSel.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_qload.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_cfg.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_nps.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_vow.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/ap_ioctl.o /home/c/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/embedded/ap/ap_mbss. [... cut ...]
echo "kmod-mt7615d" >> /home/c/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/mt7615d.default.install
make[3]: Leaving directory '/home/c/lede/package/lean/mt/drivers/mt7615d'
time: package/lean/mt/drivers/mt7615d/compile#139.06#26.91#98.58
make[2]: Leaving directory '/home/c/lede'
make[1]: *** [package/Makefile:110: /home/c/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/c/lede'
make: *** [/home/c/lede/include/toplevel.mk:231：world] 错误 2
c@ubuntu:~/lede$

yanxin152133, 2023-03-16:
https://github.com/coolsnowwolf/lede/issues/9358
time: package/lean/mt/drivers/mt7615d/compile#139.06#26.91#98.58

yanxin152133, 2023-03-16:
> 我这个默认就是5.4内核的啊，都没改内核

这都看不懂，你还是去找度娘吧

coolsnowwolf, 2026-09-05:
日志来自旧版 mt7615d vendor 驱动，且评论已指向关联 issue #9358；正文没有首个 `error:`、完整源码提交或可复现配置，无法形成当前 LEDE 的可靠修复。

closed 2026-09-05 by coolsnowwolf