coolsnowwolf/lede #11004: 更新源码后编译出错

mitsukileung, 2023-03-16:
ipq807x / Ax6

Applying /home/koysuke/lede/target/linux/ipq807x/patches-5.15/0005-v5.16-arm64-dts-qcom-Fix-IPQ8074-PCIe-PHY-nodes.patch using plaintext: 
patching file arch/arm64/boot/dts/qcom/ipq8074.dtsi
Hunk #1 FAILED at 174.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/qcom/ipq8074.dtsi.rej
Patch failed!  Please fix /home/koysuke/lede/target/linux/ipq807x/patches-5.15/0005-v5.16-arm64-dts-qcom-Fix-IPQ8074-PCIe-PHY-nodes.patch!
make[4]: *** [Makefile:36: /home/koysuke/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.102/.prepared_fd2ddd2231f59e974c14e0189c679fac] Error 1
make[4]: Leaving directory '/home/koysuke/lede/target/linux/ipq807x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/koysuke/lede/target/linux'
time: target/linux/compile#2.01#13.08#11.44
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/koysuke/lede'
make[1]: *** [target/Makefile:23: /home/koysuke/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/koysuke/lede'
make: *** [/home/koysuke/lede/include/toplevel.mk:231：world] 错误 2

DHDAXCW, 2023-03-16:
前六天上游更新导致的，不知道是不是已有了。。。

你可以手动删掉补丁 路径：`target/linux/ipq807x/patches-5.15` 找到下面两个补丁，删掉

```
0005-v5.16-arm64-dts-qcom-Fix-IPQ8074-PCIe-PHY-nodes.patch
0110-arm64-dts-ipq8074-fix-PCI-related-DT-nodes.patch
```

mitsukileung, 2023-03-16:
我试试，谢谢大佬

mitsukileung, 2023-03-16:
> 前六天上游更新导致的，不知道是不是已有了。。。
> 
> 你可以手动删掉补丁 路径：`target/linux/ipq807x/patches-5.15` 找到下面两个补丁，删掉
> 
> ```
> 0005-v5.16-arm64-dts-qcom-Fix-IPQ8074-PCIe-PHY-nodes.patch
> 0110-arm64-dts-ipq8074-fix-PCI-related-DT-nodes.patch
> ```

删除了哪个补丁后编译成功，谢谢

closed 2023-03-16 by mitsukileung