coolsnowwolf/lede #11006: 最近两个6.1.x的内核在N5105上老是崩

GreatMichaelLee, 2023-03-16:
主机ESXi 8.0, 最近两个6.1.18/19的内核在N5105上老是崩，不稳定，不建议升级，崩了也没日志，所以也没法提供debug....另外，最新出了6.2.x系列的内核，是不是切换到6.2？

Karmylr, 2023-03-16:
应该和内核版本无关。n5105虚拟化本身就有问题

GreatMichaelLee, 2023-03-16:
> 应该和内核版本无关。n5105虚拟化本身就有问题

我感觉就是和内核有关，6.1.x的版本我一直从6.1.1更新到6.1.19，5.1.x也用过很多，明显感觉有些中间版本能挺3-5天不崩（没有挺更长是因为我会频繁的更新build),但如果是有问题的内核版本（也不知道更新了啥引入了不稳定），一天之内或者最多一天多就会崩一次，看uptime就知道。我的menuconfig也从来不动的。

deplives, 2023-03-16:
5105的机器大多有问题

liuyanantj, 2023-03-17:
6.1的内核我直接物理机刷入  一直会维持在高频  不自动降频

GreatMichaelLee, 2023-03-17:
你是不是把C-State关了？

xiechangan123, 2023-03-17:
6.1的内核我直接物理机刷入3865U一直会维持在高频 不自动降频
需要修复

mmeiwenlong, 2023-03-18:
[image]

coolsnowwolf, 2023-03-23:
感觉在国产的小主机 BIOS 上面 C-STATES 问题比较多

fishg, 2023-04-08:
> 6.1.20的内核我直接物理机刷入3865U一直会维持在高频 不自动降频 需要修复

看看电源模式是不是performmance，好像新内核默认值变了

xiechangan123, 2023-04-08:
> > 6.1.20的内核我直接物理机刷入3865U一直会维持在高频 不自动降频 需要修复
> 
> 看看电源模式是不是performmance，好像新内核默认值变了

把bios重置了（默认开启c-state），可以降频了。。。

littoy, 2023-04-08:
物理机装的op，可以运行这以下命令看看电源设置，我的主机关c-state也可以睿频。
``` shell
cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors 
cat /sys/devices/system/cpu/intel_pstate/status
```

kxdn3, 2023-04-20:
我的4125不行，睿频只能1600到2500mhz，5.15能600到2500mhz，奇怪了

littoy, 2023-04-22:
> 主机ESXi 8.0, 最近两个6.1.18/19的内核在N5105上老是崩，不稳定，不建议升级，崩了也没日志，所以也没法提供debug....另外，最新出了6.2.x系列的内核，是不是切换到6.2？

试试最新的版本，应该是解决了

littoy, 2023-04-22:
> 我的4125不行，睿频只能1600到2500mhz，5.15能600到2500mhz，奇怪了
用6.1有段时间了，好像是6.1内核默认电源策略变了，可以装这个app手动调整电源策略试试。
[luci-app-cpufreq.zip](https://github.com/coolsnowwolf/lede/files/11299911/luci-app-cpufreq.zip)

kxdn3, 2023-12-16:
用cat /proc/cpuinfo | grep "cpu MHz" 和cpuinfo查询都正常睿频，应该就是luci页面显示不对

coolsnowwolf, 2026-09-05:
该 issue 仅描述 ESXi 8.0 虚拟机上 N5105 使用 6.1.18/19 偶发崩溃，没有 panic/oops、宿主机日志或复现条件，无法定位内核缺陷。相关版本已过时，当前 LEDE 已使用更新的 6.12/6.18 内核系列，不能通过切换到当年的 6.2.x 推断修复。若仍能复现，请提供完整 guest `pstore`/串口日志及 ESXi vmkernel 日志后重新提交；本 issue 关闭。

closed 2026-09-05 by coolsnowwolf