coolsnowwolf/lede #11007: 使用tc设置限速会导致断网

zzzzzhy, 2023-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用tc设置限速会导致断网,官方op无这种情况,是否是内核有哪里改动导致?

### (2) 路由器型号和固件版本

A：mt7621

### (3) 详细日志

A：        tc qdisc add dev $dev root handle 1: htb
        tc class add dev $dev parent 1: classid 1:1 htb rate ${dl}kbit

        ip link add dev ${dev}-ifb name ${dev}-ifb type ifb
        ip link set dev ${dev}-ifb up

        tc qdisc add dev ${dev}-ifb root handle 1: htb
        tc class add dev ${dev}-ifb parent 1: classid 1:1 htb rate ${up}kbit

        tc qdisc add dev $dev ingress
        tc filter add dev $dev parent ffff: protocol ip u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ${dev}-ifb

yanxin152133, 2023-03-20:
试试路由器限速 ：）

coolsnowwolf, 2026-09-05:
issue 只有 tc 命令片段，没有实际接口、qdisc 状态、内核版本/配置或断网时的内核日志；命令本身还涉及 ingress/IFB 绑定和 classid，无法判断失败点，当前没有可验证的 LEDE 代码修复。

closed 2026-09-05 by coolsnowwolf