coolsnowwolf/lede #11018: mt-wifi闭源驱动或者luci界面bug 型号识别错误

Starktomy, 2023-03-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这个路由器的wifi芯片应该是MediaTek MT7603E, MediaTek MT7612E
在menuconfig选择了kmod-mt7603e，kmod-mt76x2e，luci-app-mtwifi，mt_wifi，MTK_CHIP_MT7603E_MT7612E=y

但是luci中会显示7615.1和7615.2，5g可以开启但连不上，2.4g根本无法开启，如图
看了一下log似乎是设备名不对
[image]

### (2) 路由器型号和固件版本

A：zbt-3526-32m

### (3) 详细日志

A：config和系统日志作为文件上传了
config：[lede.txt](https://github.com/coolsnowwolf/lede/files/11010306/lede.txt)

系统日志：[syslog.txt](https://github.com/coolsnowwolf/lede/files/11010309/syslog.txt)

gforce1206, 2024-02-04:
#11861 情况跟我的一样

coolsnowwolf, 2026-09-05:
该 issue 使用的是旧版闭源 mt_wifi 驱动和旧设备识别逻辑。当前 LEDE 已提供 ZBT-WG3526 16M/32M 的独立 DTS、镜像定义及 MT7621 PCIe/Wi-Fi 初始化；同时已有 mtwifi 与 mt76 冲突防护，避免同时加载错误的 7615/7603 驱动。报告中的 zbt-3526-32m 应对应现行 `zbtlink,zbt-wg3526-32m`，需使用匹配镜像和驱动组合，不能把 MT7603E/MT7612E 与 mt7615d 闭源配置混用。当前代码无需新增修复，issue 关闭。

closed 2026-09-05 by coolsnowwolf