coolsnowwolf/lede #11019: btbcm.ko' is missing

vipboddy, 2023-03-20:
ERROR: module '/home/lucky/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.103/drivers/bluetooth/btbcm.ko' is missing.
make[2]: *** [modules/other.mk:73: /home/lucky/lede/bin/targets/x86/64/packages/kmod-bluetooth_5.15.103-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/home/lucky/lede/package/kernel/linux'
time: package/kernel/linux/compile#54.67#15.96#269.13
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1

coolsnowwolf, 2023-03-21:
fixed

catsimple, 2023-03-21:
git pull [https://github.com/coolsnowwolf/lede/commit/756cf526a691bc95ce43d91e8d7b5ee6a8cc126a](url)
make clean后问题依旧

mahakalaya, 2023-03-22:
> git pull make clean后问题依旧

+1

WYC-2020, 2023-03-22:
那样修复是不行的，亲测所有内核 5.4 5.10 5.15 6.1
[image]
[image]
[image]

mahakalaya, 2023-03-22:
> 那样修复是不行的，亲测所有内核 5.4 5.10 5.15 6.1 [image] [image] [image]

WCY老大修复一下嘛

744287383, 2023-03-24:
`SHELL= flock /home/cattle/Downloads/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cattle/Downloads/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/packages/.pkgdir/kernel/. /home/cattle/Downloads/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /home/cattle/Downloads/lede/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.kernel_installed
Package kmod-bluetooth is missing dependencies for the following libraries:
btbcm.ko
make[3]: *** [modules/other.mk:74: /home/cattle/Downloads/lede/bin/targets/armvirt/64/packages/kmod-bluetooth_5.15.104-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/cattle/Downloads/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.61#0.25#2.84
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/cattle/Downloads/lede'
make[1]: *** [package/Makefile:110: /home/cattle/Downloads/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cattle/Downloads/lede'
make: *** [/home/cattle/Downloads/lede/include/toplevel.mk:231：world] 错误 2
`
我这个好像也是这个错误

sxml, 2023-03-24:
大佬们 怎么解决呢？

Package kmod-bluetooth is missing dependencies for the following libraries:
btbcm.ko
make[3]: *** [modules/other.mk:73: /workdir/openwrt/bin/targets/armvirt/64/packages/kmod-bluetooth_5.15.104-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#4.07#0.46#4.50
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2023-03-24:
这个l不是已经删除了么 你们自己在distclean 在menuconfig 一下 就好了，如果不想用删除版本 就像我上面图那样改，我自己是保留了的

commit ae12d2f7 (2023-03-24): rockchip: refresh kernel 5.15 config

Also fixes: #11019
closed 2023-03-24 by aiamadeus