coolsnowwolf/lede #11025: 关于MWAN3添加线路以后按第二次应用的时候会导致卡住，请问有办法解决这个问题吗？

longjindong, 2023-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [y] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
关于MWAN3添加线路以后按第二次应用的时候会导致卡住
A：
[image]

### (2) 路由器型号和固件版本

A：X86 最新编译版本

### (3) 详细日志

A：并无出现什么错误日志

hgh188, 2023-03-23:
一样的问题

WYC-2020, 2023-03-23:
https://github.com/coolsnowwolf/lede/issues/10910#issuecomment-1441696278  用这个人的那个方法试下吧

WYC-2020, 2023-03-23:
你也可以看看这个issue 反正讨论很激烈，我没遇到这个问题，我用的库 很多是新的 和l大的有区别，所以我也不知道更新那个库能解决这个问题

longjindong, 2023-03-24:
我尝试修改了防火墙INIT的文件，但是还是不行 也是一样，请问如何单独回滚内核 不回滚其他插件的修改呢

longjindong, 2023-03-30:
有人解决过这个问题吗，求助。

alexliu269, 2023-03-31:
网上能搜索的方法都试过了 不行 mwan大佬修复之前只能用回养老版了

chiyu521, 2023-11-25:
问题解决 
替换 MWAN3  
https://github.com/0xACE8/luci-app-mwan3
https://github.com/0xACE8/mwan3

coolsnowwolf, 2026-09-05:
该 issue 没有错误日志或最小配置，仅描述在 x86 上第二次点击应用时页面卡住。当前 mwan3/luci-app-mwan3 已同步上游 2.12.x 并包含接口重载、ACL 和状态页修复；截图无法判断是浏览器等待、netifd 重载还是用户自定义接口配置。请在现行版本提供 `logread -f`、浏览器网络请求及 `/etc/config/network`/`mwan3` 后再提交。当前无可验证的 LEDE 补丁，关闭该历史 issue。

closed 2026-09-05 by coolsnowwolf